nanoframework/Home #1401: `Math.Max(double)` and `Math.Min(double)` lose precision in the return value

CoryCharlton, 2023-11-07:
### Library/API/IoT binding

System.Math

### Target name(s)

Testing on ESP32_PSRAM_BLE_GenericGraphic_REV3 but this should impact any target that doesn't have `DP_FLOATINGPOINT == TRUE`

### Firmware version

Latest

### Description

The input values to Math.Max(double) and Math.Min(double) lose precision/change when they are returned.

### How to reproduce

Run the code example below.

### Expected behaviour

The exact same value that is sent in is returned.

### Screenshots

[image]

### Sample project or code

```c#
var doubleLow = 1234.5678d;
var doubleHigh = 8765.4321d;

var doubleMax = Math.Max(doubleLow, doubleHigh);
var doubleMin = Math.Min(doubleLow, doubleHigh);

Console.WriteLine($"Double - Inputs: {doubleLow}, {doubleHigh} - Max: {doubleMax} - Min: {doubleMin}");

var floatLow = 1234.5678f;
var floatHigh = 8765.4321f;

var floatMax = Math.Max(floatLow, floatHigh);
var floatMin = Math.Min(floatLow, floatHigh);

Console.WriteLine($"Float - Inputs: {floatLow}, {floatHigh} - Max: {floatMax} - Min: {floatMin}");
```

CoryCharlton, 2023-11-07:
The issue is that when `DP_FLOATINGPOINT != TRUE` the input `double`s are cast to `float` which understandably causes a loss in precision. Removing the conditional check here and always working with double is a trivial change, in fact I've already made it and it works on my test device, but I'm not sure I have enough context to be sure that is the correct change.

For instance what is `DP_FLOATINGPOINT` actually indicating? I see this:

```CMake
#################################################################################
# chooses if double-point float point support is provided by the platform
# (default is OFF which means that single-precision floating point is effective)
option(DP_FLOATINGPOINT "option to enable support for double-precision floating point")
```

What platform? nanoFramework? The RTOS? The device?
Is it indicating that double-precision floating points won't work or just that we are choosing not to support them?

Like I said I made the change to remove the condition and always work with the double and it runs fine on my ESP32 (ESP32-D0WDQ6-V3 (revision v3.0)) test device. 

If the intent of `DP_FLOATINGPOINT` is to say "Double precision floating points work but because they are not hardware optimized we are electing to work with single precision values" then I believe removing this check in a "pure" method like Min/Max is the correct option.

torbacz, 2023-12-28:
Already completed?
@CoryCharlton @josesimoes

CoryCharlton, 2023-12-28:
> Already completed? @CoryCharlton @josesimoes

No, my questions regarding the specific meaning of `DP_FLOATINGPOINT` were never answered so I'm not sure how to proceed.

The issue is the cast from `double` to `float` and removing it here is simple enough. The outstanding question is whether or not this is the correct change for all devices.

josesimoes, 2024-01-02:
`DP_FLOATINGPOINT` is to signal that the hardware platform has support for DOUBLE PRECISION floating point calculation. Meaning that there is full support for doubles.
If there's not, instead of pulling in the math libs from the CRT, we choose to not support doubles as default simply because float precision should cover the vast majority of the use cases for nanoFramework usage scenarios.

CoryCharlton, 2024-01-06:
> `DP_FLOATINGPOINT` is to signal that the hardware platform has support for DOUBLE PRECISION floating point calculation. Meaning that there is full support for doubles. If there's not, instead of [...]

So is `DP_FLOATINGPOINT` set incorrectly in this case? If I remove the `#if` and use doubles the code works fine and the values are as expected so I'm not clear why `DP_FLOATINGPOINT` is not defined as true in this case.

josesimoes, 2024-01-08:
To the best of my knowledge, ESP32 devices do NOT have hardware DP, that's why they are not supporting doubles, only floats.

Ellerbach, 2024-01-08:
Correct, no hardware support for double in ESP32 devices. It's emulated in software.

See here: https://docs.espressif.com/projects/esp-idf/en/v5.0.3/esp32s3/api-guides/performance/speed.html#improving-overall-speed

If you really need this, you can activate this:

Open the sdkconfig file you want to add the double point.
Add the following line to the file: CONFIG_COMPILER_DOUBLE_MATH_FUNCTIONS=y
Save the file and rebuild your project (don't forget to delete the one generated at the root of the directory).

Not sure if you need to adjust something else somewhere else.

josesimoes, 2024-07-31:
The reason for the "lost precision" with `double` type is explained on our Architecture docs about [Floating-point calculations](https://docs.nanoframework.net/content/architecture/floating-point-calculations.html).

The default build option for the precompiled firmware images we provide has DP (double precision) calculation turned OFF, thus the "backing type" for these calculations is `float` type.
I'll copy here an excerpt of the referenced doc "(...) the extra precision provided by the double type is seldom required on typical embedded application use cases."

Offering a precision of ~6-9 digits, with a size of 4 bytes and saving precious flash space is considered (by design) more than adequate for the vast majority of embedded world applications. This behavior is by design and well documented.

Adding to this, the fact that the build option `DP_FLOATINGPOINT` is provided, allowing builds with support for double precision FP which will enable the use of `double` type without any loss of precision, gives developers the choice of using it, if their application calls for such precision.

So, this is by design. The behavior you're seeing is the expected one. If `double` with full precision are wanted, need to build the firmware with _that_ build option.

Closing this now. Feel free to continue the conversation if more clarification is required.

closed 2024-07-31 by josesimoes