nanoframework/Home #1405: Calling Register method of the ProvisioningDeviceClient class failed for M5StickC/M5StickCPlus targets

romankiss, 2023-11-16:
### Library/API/IoT binding

nanoFramework.Azure.Devices.Client

### Visual Studio version

VS2022 Version 17.7.7

### .NET nanoFramework extension version

2022.3.0.65

### Target name(s)

M5StickC, M5StickCPlus

### Firmware version

1.9.0.849, 1.9.0.842, 1.9.0.823, ...

### Description

calling the following method to obtain an info of the assigned IoT Hub:

    var drr = provisioningClient.Register(adata, CancellationToken.None);

is always failed with the following error:

    Failed allocation for 83 blocks, 996 bytes.
    There's enough free memory, heap compaction scheduled.

    ++++ Exception System.InvalidCastException - CLR_E_INVALID_CAST (6) ++++
    ++++ Message: 
    ++++ nanoFramework.Azure.Devices.Provisioning.Client.ProvisioningDeviceClient::ClientMqttMsgReceived [IP: 00be] ++++
    ++++ nanoFramework.M2Mqtt.MqttClient::OnMqttMsgPublishReceived [IP: 0024] ++++
    ++++ nanoFramework.M2Mqtt.MqttClient::DispatchEventThread [IP: 00bb] ++++

**Note, that this method is working very well for targets such as ESP32_PICO and ESP32_S3**

Ellerbach, 2023-11-16:
@romankiss thanks for opening the issue. Which device are you using? As it seems to fail because of memory allocation.

romankiss, 2023-11-16:
@Ellerbach  thanks for prompt response. I am using the controllers from M5Stack such as AtomLite, AtomMatrix, M5StampS3, M5Capsule, AtomS3Lite, AtomS3, **M5StickC** and **M5StickCPlus**. 

Note, that the M5Stack dropped the price for M5StickC device (older version) such as a perfect device for student class for Azure IoT Hub exploring, so I have tried to use it but always failed. It looks like, the provisioning has been done properly (I can see it in my IoT Central app) but at the device side this response message failed.

romankiss, 2023-11-16:
M5StickC is based on the ESP32-PICO(4M) the same as is used in the working AtomLite device. The difference between those devices is that the M5StickC has a RGB screen (80x160). 

Yes, I have tried using the DeviceClient in the way what you mentioned, but without a success result. Note, that sometime after couple hours retrying process, the device has been connected and work properly, but when the device restarted again, the problem is returned back.

I will try to use M5StickC device without the screen, in other words, I will use a target ESP32_PICO such as a working target. One more time, this device [M5StickC](https://shop.m5stack.com/products/stick-c?variant=43982750875905) is a perfekt device (< $10) for HandsOnLab for Azure IoT Central, so it will be fine to find a properly solution for this issue.

p.s. what is the tip for enabling a Debug output from the library  nanoFramework.Azure.Devices.Client? I would like to see on my _Serial Monitor 2_ all these Debug.WriteLine checkpoints.
.

romankiss, 2023-11-16:
I just finished another test, where the **provisioningDeviceClient.Register** call has been commented and I used explicitly value of the _AssignedHub_ (created offline using my tool) and everything is working well with sensors, etc.

IMHO, something in the ProvisioningDeviceClient.Register is sensitive, when the message from the DPS server has been received, and it should be optimized.

alberk8, 2023-11-17:
@romankiss ,  Can you try to use this to measure the memory on the ESP32, before and after the **provisioningDeviceClient.Registe**r call ?

```
using nanoFramework.Hardware.Esp32;
using Memory = nanoFramework.Runtime.Native.GC;

 public static class Diag
 {
     public static void PrintMemory(string msg)
     {
         
         NativeMemory.GetMemoryInfo(NativeMemory.MemoryType.All, out uint totalSize, out uint totalFreeSize, out uint largestBlock);
         Console.WriteLine($"\n{msg}-> Total Mem {totalSize} Total Free {totalFreeSize} Largest Block {largestBlock}");
         Console.WriteLine($"nF Mem {Memory.Run(false)}\n ");
     }
 }
```

Ellerbach, 2023-11-17:
> p.s. what is the tip for enabling a Debug output from the library nanoFramework.Azure.Devices.Client? I would like to see on my Serial Monitor 2 all these Debug.WriteLine checkpoints.

Rather than using the nuget, you clone the project, reference the project in your application and run all in debug. But you're going to use even more memory!

Keep also in mind that if you are using the M5StickC nuget, it comes with quite a lot of things and that can make a difference as well. It's all in lazy loading (so the screen is not initialize if you're not using it, etc) but this can affect as well the amount of memory used.

And just before calling the Register function, you can try to explicitly call the GC to clean and compress the memory. See https://docs.nanoframework.net/api/nanoFramework.Runtime.Native.GC.html

```csharp
using Memory = nanoFramework.Runtime.Native.GC;

// Your code before
Native.GC.Run(true);
// Right after, call the Register function
var drr = provisioningClient.Register(adata, CancellationToken.None);
```

And I fully understand the rational with the price of the device. Now, this is a very limited one!

romankiss, 2023-11-17:
@Ellerbach thanks for suggestion using the GG, I already have it in the multiple places.

@alberk8 thanks for this tip, the following are my outputs from the Register code:

    using (var provisioningClient = ProvisioningDeviceClient.Create(_deviceProvisioningEndpoint, iotcScopeId, deviceId, deviceKey, azureCertDPS))
    {
        Debug.WriteLine($"DPS: the provisioningClient has been created");
        var adata = new ProvisioningRegistrationAdditionalData() { JsonData = PnpConvention.CreateDpsPayload(modelId) };
        nanoFramework.Runtime.Native.GC.Run(true);
        Diag.PrintMemory("Before Register");
        var drr = provisioningClient.Register(adata, new CancellationTokenSource(40000).Token);
        Debug.WriteLine($"Status={drr.Status}, Error={drr.ErrorMessage ?? "null"}");
        Diag.PrintMemory("After Register");
        if (drr.Status != ProvisioningRegistrationStatusType.Assigned)
            throw new Exception($"{drr.ErrorMessage}");
        iothubname = drr.AssignedHub;
        Debug.WriteLine($"DPS successed, IoTHub: {iothubname}");
    }

**Test1-bad result: Device M5StickC  (target M5StickC)**

    Before Register-> Total Mem 267480 Total Free 63100 Largest Block 17408
    nF Mem 14292
 
    Failed allocation for 83 blocks, 996 bytes.
    There's enough free memory, heap compaction scheduled.

        ++++ Exception System.Exception - CLR_E_WRONG_TYPE (6) ++++
        ++++ Message: 
        ++++ nanoFramework.Azure.Devices.Provisioning.RegistrationState::get_registrationId [IP: 0004] ++++
        ++++ nanoFramework.M2Mqtt.MqttClient::OnMqttMsgPublishReceived [IP: 0024] ++++
        ++++ nanoFramework.M2Mqtt.MqttClient::DispatchEventThread [IP: 00bb] ++++
    Status=2, Error=Unknown error, cancellation requested: True

    After Register-> Total Mem 267480 Total Free 77396 Largest Block 23552
    nF Mem 21336

**Test2-good result: Device M5StampS3  (target ESP32_S3)** 

    Before Register-> Total Mem 337600 Total Free 33328 Largest Block 16384
    nF Mem 26172
 
    The thread '<No Name>' (0x3) has exited with code 0 (0x0).
    The thread '<No Name>' (0x4) has exited with code 0 (0x0).
    The thread '<No Name>' (0x6) has exited with code 0 (0x0).
    The thread '<No Name>' (0x5) has exited with code 0 (0x0).
    Status=3, Error=null

    After Register-> Total Mem 337600 Total Free 47632 Largest Block 28672
    nF Mem 33612

**Test3 - No Register: Device M5StickC  (target M5StickC)**
 
    Before CreateAndNoRegister-> Total Mem 267480 Total Free 74916 Largest Block 40960
    nF Mem 22980
 
    DPS: the provisioningClient has been created
    The thread '<No Name>' (0x6) has exited with code 0 (0x0).
    The thread '<No Name>' (0x3) has exited with code 0 (0x0).
    The thread '<No Name>' (0x4) has exited with code 0 (0x0).
    The thread '<No Name>' (0x5) has exited with code 0 (0x0).

    After CreateAndNoRegister-> Total Mem 267480 Total Free 74404 Largest Block 23552
    nF Mem 22620

Ellerbach, 2023-11-17:
@romankiss, here is an idea you can try because the message says there is enough memory but not as 1 block:

Clone the Azure repo and before the try on this line: https://github.com/nanoframework/nanoFramework.Azure.Devices/blob/2c9d4cda617260d3d1ec6ef2c499561acb6c86e1/nanoFramework.Azure.Devices.Client/ProvisioningDeviceClient.cs#L257 add `Native.GC.Run(true);`

What it will do is force to clean and compact the memory so that it will have a clean path to run this function. If that works, then you'll only get a little penalty and we can even include this in the lib. This little penalty doesn't really matter compare to the rest of the networking operations which are long.

If you can confirm that this is solving your issue, you can even go for a PR, I4ll be happy to appprove it :-)

alberk8, 2023-11-18:
> @Ellerbach thanks for suggestion using the GG, I already have it in the multiple places.
>
> @alberk8 thanks for this tip, the following are my outputs from the Register code:
>
> ```
> using (var [...]

It looks like you actually runs out of managed memory when running the call.  If the method @Ellerbach  suggest does not work.  Then the only other option is to tweak the native memory allocation (this calls for custom firmware compilation).

romankiss, 2023-11-19:
@Ellerbach After couple days of testing, I can confirm, that your suggestion such as using the `Native.GC.Run(true);` is a simple/fast way for devices like M5StickC/M5StickCPlus to solve this issue.

The method _provisioningClient.Register succeded_ in the first time or sometime after a rebooting by watchdog timer. 

Note, that I didn't clone the Azure repo, just only the class _ProvisioningDeviceClient_ and its dependencies to the internal classes into my namespace. Also, I replaced a class _RegistrationOperationStatusSimple_ by _RegistrationOperationStatus_ for optimizing of memory usage.

Thanks for helping to resolve this issue. Now, I can use as a workaround my class _ProvisioningDeviceClient2_, but it will be nice to have it in the nanoFramework.Azure.Devices.Client library.

The following is a screen snippet of the successful Register call:

    Before Register-> Total Mem 267480 Total Free 63088 Largest Block 17408
    nF Mem 14196
 
    $dps/registrations/res/202/?$rid=1700402073&retry-after=3
    {"operationId":"5.d964ae308cb29aea.f2a6546d-6497-4da5-8982-88527c0ce00a","status":"assigning"}
    $dps/registrations/res/200/?$rid=1700402073
    {"operationId":"5.d964ae308cb29aea.f2a6546d-6497-4da5-8982-88527c0ce00a","status":"assigned","registrationState":{...}}

    After Register-> Total Mem 267480 Total Free 77384 Largest Block 24576
    nF Mem 21540

Ellerbach, 2023-11-20:
Great, happy you found a way!

> Now, I can use as a workaround my class ProvisioningDeviceClient2, but it will be nice to have it in the nanoFramework.Azure.Devices.Client library

I'm not really a huge fan of creating multiple nugets. I'm fine adding couple of GC calls in the function but definitely not splitting the lib and creating multiple ones. The code is indeed open source for those like you who have special needs.

romankiss, 2023-11-21:
@Ellerbach After more testing days, I have figured out, that using the `Native.GC.Run(true);` is not enough for reliable process of the DPS for devices like is the M5StickC/M5StickCPlus, etc. in other words for devices, where is a memory resource limited.

Note, that the _ProvisioningDeviceClient_ consumes a lot of memory and its mqtt subscriber _ClientMqttMsgReceived_ doesn't work reliable with its memory usage included a `Native.GC.Run(true);`, so I made some changes based on the following implementation patterns:
1. The subscriber should be more lightweight, deserializing only properties for DPS processing, such as _operationId_ and _status_. When the _status = assigned_, the DPS message (json formatted text) is save it in the ProvisioningDeviceClient for its postprocessing such as deserializing, etc., see my implementation:

        private void ClientMqttMsgReceived(object sender, MqttMsgPublishEventArgs e)
        {
            nanoFramework.Runtime.Native.GC.Run(true);
            try
            {
                string message = Encoding.UTF8.GetString(e.Message, 0, e.Message.Length);
                Debug.WriteLine(e.Topic);
                Debug.WriteLine(message);
                ProvisioningRegistrationStatusType status = _status;
                // The message after the publish looks like that:
                // $dps/registrations/res/202/?$rid={request_id}&retry-after=x
                // x is the retry-after value in seconds
                var opeStat = (RegistrationOperationStatusSimple)nanoFramework.Json.JsonConvert.DeserializeObject(message, typeof(RegistrationOperationStatusSimple));
                _operationId = opeStat.operationId;
                status = AssignStatus(opeStat.status);
                if (e.Topic.StartsWith($"$dps/registrations/res/202/?$rid={_requestId}"))
                {
                    _retry = Convert.ToInt32(e.Topic.Substring(e.Topic.LastIndexOf('=') + 1));
                    _retryInitTime = DateTime.UtcNow.AddSeconds(_retry);
                }
                else if ((e.Topic.StartsWith($"$dps/registrations/res/200/?$rid={_requestId}")) && (!_isMessageProcessed))
                {
                    // This is to avoid having multiple messages to process
                    _isMessageProcessed = true;
                    _message = message;
                    _statusLast = _status;
                }
                else if (!_isMessageProcessed)
                {
                    // This is an error and we will have an error response
                    Hashtable error = (Hashtable)nanoFramework.Json.JsonConvert.DeserializeObject(message, typeof(Hashtable));
                    _result = new(_registrationId, DateTime.UtcNow, string.Empty, string.Empty, ProvisioningRegistrationStatusType.Failed, string.Empty, DateTime.UtcNow, (int)error["errorCode"], (string)error["message"], string.Empty);
                    // Forcing the loop to stop
                    status = ProvisioningRegistrationStatusType.Assigned;
                }
        
                _status = status;
            }
            catch (Exception ex)
            {
                // Just don't do anything, let's give it another try
                Debug.WriteLine($"Exception in event: {ex}");
            }
        }

2. Creating a _DeviceRegistrationResult_ object for return back from the _Register_ call is going to make **after cleaning all resources** related to the MQTT connectivity:

          CleanAll();

          #region Creating a DeviceRegistrationResult 
          nanoFramework.Runtime.Native.GC.Run(true);
          Diag.PrintMemory("AfterCleanAll");
          if (!string.IsNullOrEmpty(_message))
          {
              var opeStat = (RegistrationOperationStatus)nanoFramework.Json.JsonConvert.DeserializeObject(_message, typeof(RegistrationOperationStatus));
              var reg = opeStat.registrationState;
              var status = AssignStatus(opeStat.status);
              if (_statusLast == ProvisioningRegistrationStatusType.Assigned)
              {
                  ProvisioningRegistrationSubstatusType sub = ProvisioningRegistrationSubstatusType.InitialAssignment;
                  if (reg.substatus == "deviceDataMigrated")
                  {
                      sub = ProvisioningRegistrationSubstatusType.DeviceDataMigrated;
                  }
                  else if (reg.substatus == "deviceDataReset")
[... 8 lines cut ...]
              }
          }
          #endregion

The following screen snippet shows usage of the memory before and after the Register call:

        DPSRegisterBefore-> Total Mem 267480 Total Free 63432 Largest Block 17408
        nF Mem 15480
         
        $dps/registrations/res/202/?$rid=1700580186&retry-after=3
        {"operationId":"5.d964ae308cb29aea.b4971313-2743-4ebc-a261-d6f17114549f","status":"assigning"}
        $dps/registrations/res/200/?$rid=1700580186
        {"operationId":"5.d964ae308cb29aea.b4971313-2743-4ebc-a261-d6f17114549f","status":"assigned","registrationState":{...}}
        
        AfterCleanAll-> Total Mem 267480 Total Free 77736 Largest Block 24576
        nF Mem 21972
        
        >>>Creating a DeviceRegistrationResult
        
        DPSRegisterAfter-> Total Mem 267480 Total Free 77736 Largest Block 24576
        nF Mem 22464

As you can see, the DPS json formatted message is deserialized to the DeviceRegistrationResult object using more memory (21972) than in the mqtt subscriber (15480).

After this changes, each Register call is successful without any memory problem.

Ellerbach, 2023-11-22:
> After this changes, each Register call is successful without any memory problem.

Thanks for the deep investigation. If you can put all this in a PR, that would be perfect. As it seems that those adjustments you have been doing is not changing the output but improve the memory usage. So by having those changes in, this will benefit all kind of devices!

romankiss, 2023-11-22:
@Ellerbach The following is the modified file _nanoFramework.Azure.Devices.Client/ProvisioningDeviceClient.cs_ Please, could you make those changes in the repo? I don't have setup my dev environment for GitHub and also, I have not experience for this PR. Note, that the file has been completely tested.

Thank you so much and Happy Thanksgiving.
Roman

**nanoFramework.Azure.Devices.Client/ProvisioningDeviceClient.cs:**

```
// Copyright (c) Microsoft. All rights reserved.
// Licensed under the MIT license. See LICENSE file in the project root for full license information.

using nanoFramework.M2Mqtt;
using nanoFramework.M2Mqtt.Messages;
using NFAppM5StickC_Testing;
using System;
using System.Collections;
using System.Diagnostics;
#if !FULLYMANAGED
using System.Security.Cryptography.X509Certificates;
#endif
using System.Text;
using System.Threading;

namespace nanoFramework.Azure.Devices.Provisioning.Client
{
    /// <summary>
    /// Allows devices to use the Device Provisioning Service.
    /// </summary>
    public class ProvisioningDeviceClient : IDisposable
    {
        const string DpsSubscription = "$dps/registrations/res/#";
#if FULLYMANAGED
        private IMqttClient _mqttc;
#else
        private MqttClient _mqttc;
#endif
        private readonly string _deviceEndPoint;
        private long _requestId;
        private string _registrationId;
        private int _retry = -1;
        private DateTime _retryInitTime;
        private string _operationId;
        private ProvisioningRegistrationStatusType _status = ProvisioningRegistrationStatusType.Unassigned;
        private DeviceRegistrationResult _result = null;
        private bool _isMessageProcessed = false;
        private bool _isDisposed = false;
        private string _message = null;
        private ProvisioningRegistrationStatusType _statusLast = ProvisioningRegistrationStatusType.Unassigned;
 
        /// <summary>
        /// Creates an instance of the Device Provisioning Client.
        /// </summary>
        /// <param name="globalDeviceEndpoint">The GlobalDeviceEndpoint for the Device Provisioning Service.</param>
        /// <param name="idScope">The IDScope for the Device Provisioning Service.</param>
        /// <param name="registrationId">The registration ID</param>
        /// <param name="securityProvider">The security provider instance.</param>
        /// <param name="azureCert">The Azure root certificate, leave it null if you have it stored in the device.</param>
#if FULLYMANAGED
        /// <param name="mqtt">The MQTT client instance.</param>
#endif
        /// <returns>An instance of the ProvisioningDeviceClient</returns>
        public static ProvisioningDeviceClient Create(
            string globalDeviceEndpoint,
            string idScope, string registrationId,
            string securityProvider,
#if FULLYMANAGED
            byte[] azureCert = null
#else
            X509Certificate azureCert = null
#endif
[... 291 lines cut ...]
                return ProvisioningRegistrationStatusType.Disabled;
            }

            // That should never happen but in case
            return ProvisioningRegistrationStatusType.Failed;
        }

        /// <inheritdoc/>
        public void Dispose()
        {
            if (_isDisposed)
            {
                return;
            }

            _isDisposed = true;
            if (_mqttc != null)
            {
                CleanAll();
            }
        }
    }
}

```

Ellerbach, 2023-11-23:
@romankiss I've made the PR. If you want to give it another try, let me know. Otherwise, I'll merge it. I've been doing few cosmetic adjustments compare to your code but all up, it's the exact same. So it should work!
Have a great Thanksgiving!

closed 2023-11-23 by Ellerbach

romankiss, 2023-11-23:
@Ellerbach Thank you so much, I have just finished testing and I can confirm, that the updated library is working very well on the device M5StickC and also on the other ones, such as Adafruit QT Py ESP32-S3, Waveshare ESP32-S3-Zero,  Waveshare ESP32-S3-Pico, AtomS3Lite, AtomLite, M5StampS3 and Luatos Core-ESP32S3-A10.

The following snippet is a memory usage of the DPS Register call for device M5StickC:

```
[24:28.228]DPSRegisterBefore-> Total Mem 267480 Total Free 46264 Largest Block 17408
nF Mem 15756
 
The thread '<No Name>' (0x3) has exited with code 0 (0x0).
The thread '<No Name>' (0x4) has exited with code 0 (0x0).
The thread '<No Name>' (0x6) has exited with code 0 (0x0).
The thread '<No Name>' (0x5) has exited with code 0 (0x0).

[24:34.813]DPSRegisterAfter-> Total Mem 267480 Total Free 60540 Largest Block 17408
nF Mem 22740
```

Ellerbach, 2023-11-23:
> I can confirm, that the updated library is working very well on the device M5StickC and also on the other ones

Thank YOU for digging into all this and finding a great optimization and collaboration on solving this issue!  Have a great Thanksgiving :-)