nanoframework/Home #1406: Tests do not report results if a thread is suspended

CoryCharlton, 2023-11-16:
### Library/API/IoT binding

nanoFramework.TestFramework

### Description

Tests are run and the debug output shows success/failure but the test adapter never completes and reports results if a thread is still running in a suspended state.

### How to reproduce

See code example.

### Expected behaviour

The tests to complete after all methods have returned OR for the results captured to be reported after a timeout.

### Sample project or code

```c#
using nanoFramework.TestFramework;
using System;
using System.Threading;

namespace NFUnitTest1
{
    [TestClass]
    public class Test1
    {
        private static readonly ManualResetEvent ResetEvent = new(false);

        [TestMethod]
        public void TestMethod1()
        {
            Assert.IsTrue(true, nameof(TestMethod1));
        }

        [TestMethod]
        public void TestMethod2()
        {
            Assert.IsTrue(true, nameof(TestMethod2));
        }

        [TestMethod]
        public void TestMethod3()
        {
            new Thread(() =>
            {
                Console.WriteLine($"Thread {Thread.CurrentThread.ManagedThreadId} is running...");
                // Either one of these statements will reproduce the issue. If a timeout value is supplied (ie: 30 seconds) the tests complete as expected.
                Thread.Sleep(Timeout.Infinite);
                //ResetEvent.WaitOne();
            }).Start();

            Assert.IsTrue(true, nameof(TestMethod3));
        }
    }
}
```

### Aditional information

Here's the debug output from my tests:

```
Using automatically detected runsettings file(s). To learn more visit https://aka.ms/vs-runsettings.
Using automatically detected runsettings file(s). To learn more visit https://aka.ms/vs-runsettings.
========== Starting test run ==========
Test run detected DLL(s) which would use different framework and platform versions. Following DLL(s) do not match current settings, which are .NETFramework,Version=v4.8 framework and X64 platform.
NFUnitTest.dll would use Framework .NETnanoFramework,Version=v1.0 and Platform X64.
Go to https://aka.ms/tp/vstest/multitargetingdoc?view=vs-2019 for more details on managing these settings.
[nanoTestAdapter]: Settings parsed
[nanoTestAdapter]: Test sources enumerated
[nanoTestAdapter]: Checking device on port COM3.
[nanoTestAdapter]: Device connect result is True. Attempt 0/5
[nanoTestAdapter]: Erase deployment block storage. Attempt 0/5.
[nanoTestAdapter]: Erase result is True.
[nanoTestAdapter]: Device status verified as being in initialized state. Requesting to resume execution. Attempt 0/5.
[nanoTestAdapter]: Device has completed initialization.
[nanoTestAdapter]: Device is initialized and ready!
[nanoTestAdapter]: Computing deployment blob.
[nanoTestAdapter]: Added 4 assemblies to deploy.
[nanoTestAdapter]: Adding mscorlib v1.15.6.0 (31892 bytes) to deployment bundle
[nanoTestAdapter]: Adding nanoFramework.TestFramework v2.1.87.0 (8772 bytes) to deployment bundle
[nanoTestAdapter]: Adding nanoFramework.UnitTestLauncher v0.0.0.0 (1556 bytes) to deployment bundle
[nanoTestAdapter]: Adding NFUnitTest v1.0.0.0 (976 bytes) to deployment bundle
[nanoTestAdapter]: Deploying 4 assemblies to device... Total size in bytes is 43196.
[nanoTestAdapter]: Resolving.

[nanoTestAdapter]: 
Total: (4732 RAM - 43196 ROM - 23673 METADATA)

[nanoTestAdapter]:    AssemblyRef    =       20 bytes (       5 elements)

[nanoTestAdapter]:    TypeRef        =      260 bytes (      65 elements)

[nanoTestAdapter]:    FieldRef       =        4 bytes (       1 elements)

[nanoTestAdapter]:    MethodRef      =      276 bytes (      69 elements)

[nanoTestAdapter]:    TypeDef        =     1296 bytes (     162 elements)

[nanoTestAdapter]:    FieldDef       =      216 bytes (     105 elements)

[nanoTestAdapter]:    MethodDef      =     1872 bytes (     935 elements)

[nanoTestAdapter]:    StaticFields   =      228 bytes (      19 elements)

[nanoTestAdapter]: 

[nanoTestAdapter]:    DebuggingInfo  =      936 bytes

[nanoTestAdapter]: 

[nanoTestAdapter]:    Attributes      =      104 bytes (      13 elements)

[nanoTestAdapter]:    TypeSpec        =        4 bytes (       1 elements)

[nanoTestAdapter]:    Resources Files =        0 bytes (       0 elements)

[nanoTestAdapter]:    Resources       =        0 bytes (       0 elements)

[nanoTestAdapter]:    Resources Data  =        0 bytes

[nanoTestAdapter]:    Strings         =     5443 bytes

[nanoTestAdapter]:    Signatures      =     2737 bytes

[nanoTestAdapter]:    ByteCode        =    13571 bytes

[nanoTestAdapter]: 

[nanoTestAdapter]: Ready.

[nanoTestAdapter]: Test passed,NFUnitTest1.Test1.TestMethod1.0,8490

[nanoTestAdapter]: Test passed,NFUnitTest1.Test1.TestMethod2.0,10650

[nanoTestAdapter]: Test passed,NFUnitTest1.Test1.TestMethod3.0,14170

[nanoTestAdapter]: Thread 3 is running...

Aborting test run: test run timeout of 120000 milliseconds exceeded.
========== Test run aborted: 0 Tests (0 Passed, 0 Failed, 0 Skipped) run in < 1 ms ==========
```

CoryCharlton, 2023-11-16:
Took a quick peek and it seems that the issue is the "Done." message is never sent (either [here](https://github.com/nanoframework/nf-interpreter/blob/0e1df346f49408dab1666dd4a231fad5741e33b1/targets/netcore/nanoFramework.nanoCLR/CLRStartup.cpp#L595) or [here](https://github.com/nanoframework/nf-interpreter/blob/0e1df346f49408dab1666dd4a231fad5741e33b1/targets/win32/nanoCLR/CLRStartup.cpp#L627)).

What I'm not clear on is why a suspended thread prevents this but waking a thread (ex: wait timeout expires) does not.

CoryCharlton, 2023-11-16:
It also just occurred to me that maybe the `UnitTestLauncher` should be the one spitting out a "magic" message when the tests are complete as it knows that it has executed them all.

CoryCharlton, 2023-11-16:
This works around the issue when tests are run on the device:

```c#
new Thread(() =>
{
    Thread.Sleep(TimeSpan.FromSeconds(10));
    Debug.WriteLine("Done.");
}).Start();
```

But not when run in the emulator (tried `Console` as well).

josesimoes, 2023-11-16:
*That* `Done` is being output by the execution engine when the execution ends and the CLR "reboots".
That's the only safe way to get the notion that execution has actually completed.

I understand your current issue, but I think it has to be solved other way... having the UnitTestLauncher taking care of that can be misleading. 

Take for example a test that is running a long running thread to test something there. If instead of relying on the execution engine do signal that everything has executed, we have UnitTestLauncher "deciding" that the tests are done, how do we instruct the UnitTestLauncher that a long running task? And for "how long" should it wait?

You have to make sure that the Threads do end.

CoryCharlton, 2023-11-16:
Thanks for the insight. I have a better understanding of what is occurring now and see that this is the expected behavior and can't be "solved" via the test framework.

I'll look into the possibility of parsing test results up to the point of the timeout but the best option here is going to be for the unit test author to ensure that all threads are completed.

closed 2023-11-16 by CoryCharlton