nanoframework/Home #1407: SpanBytes invalid output in debbuger

torbacz, 2023-11-22:
### Tool

Visual Studio extension

### Description

When using "Watch" or "Immediate" window SpanByte indexer is returning invalid data (always for first index?). 

[image]

### How to reproduce

```csharp
  var bytes = new byte[] { 10, 20 };
  var spanBytes = new SpanByte(bytes);

  Console.WriteLine($"Array: [0] {bytes[0]} {bytes[1]}");
  Console.WriteLine($"Span: [0] {spanBytes[0]} {spanBytes[1]}");
```
Place breakpoint on last line and check "watch" window for spanBytes[1] and spanBytes[0]. Both returns "10", but [1] should be 20.

### Expected behaviour

SpanBytes should reflect the same values in both console and watch window

CoryCharlton, 2023-11-22:
I spent some time looking into this. Unfortunately I'm not sure where to find the relevant code so I've been focused on reproducing.

This appears to impact an class with an indexer that doesn't derive from `Array`. For instance the issue is reproducible with `string`. Checking `test[0]`, `test[1]`, or `test[2]` in the immediate/quick watch/etc. will display `a` (the value at index 0):

```
var testString = "abc";
Console.WriteLine($"string [0] {testString[0]}, [1] {testString[1]}, [2] {testString[2]}");
```

This simple test class will always display 0 as well:

```
public class TestIndexer
{
    public int this[int index] => index;
}
```

And this class will fail to retrieve a value at all (presumably becase 0 to being passed to the indexer):

```
public class TestCharIndexer
{
    public char this[char index] => index;
}
```

Long story short the issue definitely appears to be related with how the indexer is being parsed (or not parsed).

josesimoes, 2024-01-05:
Seems that our debugger is having a hard time figuring out what it the indexer...
At this time it's not clear what is missing, still requires some investigation.