nanoframework/Home #1409: ESP32 GpioPulseCounter -> wrong timings

Alex-111, 2023-11-25:
### Target name(s)

ESP32

### Firmware version

1.9.0.67

### Description

The GpioPulseCounter has a wrong "RelativeTime" property result. It is wrong by a factor of 1000.

### How to reproduce

 counter = new GpioPulseCounter(21)
 {
     Polarity = GpioPulsePolarity.Both,
     FilterPulses = 100
 };
 var cnt = counter.Read();
 var timeString = cnt.RelativeTime.TotalSeconds.ToString();

timeString is wrong by a factor of 1000!

Ellerbach, 2023-11-25:
Thanks for the code, can you please write what the expected time and what you get? A factor of 1000 can be in different ways :-)

Alex-111, 2023-11-25:
E.g. I get 10000 instead of 10.

Ellerbach, 2023-12-23:
@josesimoes I've been investigating, and here are my thoughts:

* The value for the `TimeSpan` is stored as a number of ticks. See the line here: https://github.com/nanoframework/nf-interpreter/blob/fa53ac7ea065abae43e2987ba17153a487675820/targets/ESP32/_nanoCLR/nanoFramework.Hardware.ESP32/nanoFramework_hardware_esp32_native_System_Device_Gpio_GpioPulseCounter.cpp#L278
* It's a pure `TimeSpan`, see here: https://github.com/nanoframework/nanoFramework.Hardware.Esp32/blob/04abff3b7115063b3fa10846f14291fcde0cf0c5/nanoFramework.Hardware.Esp32/Gpio/GpioPulseCount.cs#L24
* The `relativeTime` is a microsecond encoding, see: https://github.com/nanoframework/nf-interpreter/blob/fa53ac7ea065abae43e2987ba17153a487675820/targets/ESP32/_nanoCLR/nanoFramework.Hardware.ESP32/nanoFramework_hardware_esp32_native_System_Device_Gpio_GpioPulseCounter.cpp#L245 and the IDF doc: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/esp_timer.html#obtaining-current-time
* And finally, the `TIME_CONVERSION_TICKUNITS` is defined here: https://github.com/nanoframework/nf-interpreter/blob/fa53ac7ea065abae43e2987ba17153a487675820/src/HAL/Include/nanoHAL_Time.h#L22 to 10_000

So, my conclusion:
* As it seems we are a 1_000 times too much now, is there anything that has changed in all this time calculation recently (or not that recently or maybe it was always off that way but I think it used to worked when implemented)?
* As we are converting micro seconds to ticks, what then should be the right formula? It seems that then it should be: `*val *= TIME_CONVERSION__TICKUNITS / 1000;`

If that seems ok, I'll go for a PR, if you see anything, and something rings a bell, let's investigate a bit more.

Alex-111, 2024-03-25:
Thanks for the detailed analysis, please could you go for the PR, or is something blocking it?