nanoframework/Home #1414: Dev Container Build Error for ESP32 C3

alberk8, 2023-12-03:
### Target name(s)

ESP32_C3_REV3

### Was working before? On which version?

N/A

### Description

Build ESP32 C3 REV3 with Dev Container

### How to reproduce

1. Clone the nf-Interpreter
2. Open in dev container in VS Code
3. Select Target ESP32_C3_REV3
4. Click Build 

### Expected behaviour

no error

### Aditional information

```
[cmake] -- Reserved RAM for allocation by IDF: 100k
[cmake] -- Reserved SPI RAM for allocation by IDF: 0k
[cmake] -- ESP32 IDF v4.4.6 source from: /sources/esp-idf
[cmake] -- 
[cmake] -- IDF_PATH is '/sources/esp-idf'
[cmake] -- Configuring incomplete, errors occurred!
[cmake] CMake Error at targets/ESP32/CMakeLists.txt:135 (message):
[cmake]   
[cmake] 
[cmake]   *******************************************************************
[cmake] 
[cmake]   '/sources/esp-idf/components/fatfs/src/ffconf.h' file is missing from IDF
[cmake] 
[cmake]   If this was deleted on a previous build using SDCard, just revert the
[cmake]   change and restore the file.
[cmake] 
[cmake]   *******************************************************************
[cmake] 
[cmake] 
[cmake] 
[proc] The command: /usr/bin/cmake/bin/cmake -DTARGET_NAME=ESP32_C3_REV3 -DSDK_CONFIG_FILE=sdkconfig.default_rev3.esp32c3 -DNF_BUILD_RTM=OFF -DNF_FEATURE_DEBUGGER=ON -DNF_FEATURE_RTC=ON -DNF_FEATURE_HAS_SDCARD=OFF -DAPI_System.IO.FileSystem=ON -DAPI_Windows.Storage=OFF -DAPI_nanoFramework.Device.OneWire=OFF -DAPI_nanoFramework.Hardware.Esp32.Rmt=ON -DESP32_RESERVED_RAM_FOR_IDF_ALLOCATION=100 -DCMAKE_TOOLCHAIN_FILE:FILEPATH=/workspaces/nf-interpreter/CMake/toolchain.riscv32-esp-elf.cmake -DNF_TARGET_HAS_NANOBOOTER=OFF -DRTOS=ESP32 -DTARGET_SERIES=ESP32_C3 -DTARGET_BOARD=ESP32_C3 -DNF_FEATURE_HAS_CONFIG_BLOCK=ON -DSUPPORT_ANY_BASE_CONVERSION=ON -DAPI_System.Net=ON -DAPI_System.Math=ON -DAPI_System.Device.Adc=ON -DAPI_System.Device.Dac=OFF -DAPI_System.Device.Gpio=ON -DAPI_System.Device.I2c=ON -DAPI_System.Device.I2c.Slave=ON -DAPI_System.Device.I2s=ON -DAPI_System.Device.Spi=ON -DAPI_System.Device.Pwm=ON -DAPI_System.Device.Wifi=ON -DAPI_System.IO.Ports=ON -DAPI_Hardware.Esp32=ON -DAPI_nanoFramework.System.IO.Hashing=ON -DAPI_nanoFramework.System.Security.Cryptography=ON -DAPI_nanoFramework.ResourceManager=ON -DAPI_nanoFramework.System.Collections=ON -DAPI_nanoFramework.System.Text=ON -DTOOL_HEX2DFU_PREFIX=/usr/local/bin/hex2dfu -DTOOL_SRECORD_PREFIX=/usr/bin -DCHIBIOS_SOURCE_FOLDER=/sources/ChibiOs -DFREERTOS_SOURCE_FOLDER=/sources/FreeRTOS -DCHIBIOS_CONTRIB_SOURCE=/sources/ChibiOs-Contrib -DCHIBIOS_HAL_SOURCE=/sources/ChibiOs -DSTM32_CUBE_PACKAGE_SOURCE=/sources/STM32CubeF7 -DLWIP_SOURCE=/sources/lwip -DMBEDTLS_SOURCE=/sources/mbedtls -DFATFS_SOURCE=/sources/fatfs -DSPIFFS_SOURCE=/sources/spiffs -DESP32_IDF_PATH=/sources/esp-idf -DTI_SL_CC32xx_SDK_SOURCE=/sources/SimpleLinkCC32 -DTI_SL_CC13xx_26xx_SDK_SOURCE=/sources/SimpleLinkCC13 -DTI_XDCTOOLS_SOURCE=/usr/local/bin/titools -DTI_SYSCONFIG_SOURCE=/sources/TI_SysConfig -DAZURERTOS_SOURCE_FOLDER=/sources/AzureRTOS -DNETXDUO_SOURCE_FOLDER=/sources/NetxDuo -DCMAKE_BUILD_TYPE=Debug -DBUILD_VERSION=0.9.99.999 -DBUILD_VERBOSE=OFF -DNF_WP_TRACE_ERRORS=OFF -DNF_WP_TRACE_HEADERS=OFF -DNF_WP_TRACE_STATE=OFF -DNF_WP_TRACE_NODATA=OFF -DNF_WP_TRACE_ALL=OFF -DNF_WP_IMPLEMENTS_CRC32=OFF -DNF_PLATFORM_NO_CLR_TRACE=OFF -DNF_CLR_NO_IL_INLINE=OFF -DNF_FEATURE_WATCHDOG=ON -DSWO_OUTPUT=OFF -DRADIO_FREQUENCY=CHANGE_ME_TO_A_VALID_VALUE_868_OR_915 -Wdev -Wdeprecated --no-warn-unused-cli -S/workspaces/nf-interpreter -B/workspaces/nf-interpreter/build -G Ninja exited with code: 1
```

alberk8, 2023-12-03:
I find that if comment out line 135 of `targets/ESP32/CMakeLists.txt` and this line in [ffconf.h](https://github.com/nanoframework/nf-interpreter/blob/2365bf41e1dcb2afab2bb3cc35c1f21e51a55d3e/targets/ESP32/ESP32_C3/ffconf.h#L15).  The build will go through.

Ellerbach, 2023-12-23:
As it's written, you have to go to the dev container and revert the file. Open a terminal and run:
[image]

```shell
cd /sources/esp-idf
git checkout HEAD -- .
```

You'll be good to go! No need to comment anything else. It's related to the fact some of the ESP devices requires the file, some others not...

closed 2023-12-23 by Ellerbach

alberk8, 2023-12-24:
This issue only occur with the C3 target,  with the other ESP32 target I just clone, make the necessary changes, select target then build.

Ellerbach, 2023-12-25:
> If this was deleted on a previous build using SDCard

OK, as I wrote, it depends on the device and the error message seems clear to me. If you are building outside a container, you'll get the same issue. Now, message may include the instructions maybe.

Ellerbach, 2023-12-25:
See https://github.com/nanoframework/nf-interpreter/pull/2853, let me know if that's more clear.

alberk8, 2023-12-26:
@Ellerbach, Thank you.  That is crystal clear.

josesimoes, 2024-01-05:
This happens with any build for any series that builds after a previous one with SD card enabled.
The remedy, as the message tries to explain, it's simply reverting the changes on that file.