nanoframework/Home #1418: .NET nanoFramework Firmware Flasher v2.5.49 doesn't work

romankiss, 2023-12-21:
### Tool

Visual Studio extension

### Description

I have just updated a **nanoff** tool from 2.5.46 version and I am facing the following problem:

          PS C:\WINDOWS\system32> nanoff --listtargets
          .NET nanoFramework Firmware Flasher v2.5.49+da497901db
          Copyright (C) 2019 .NET Foundation and nanoFramework project contributors
          
          
          nanoff : Unhandled exception. System.TypeInitializationException: The type initializer for 
          'nanoFramework.Tools.FirmwareFlasher.FirmwarePackage' threw an exception.
          At line:1 char:1
          + nanoff --listtargets
          + ~~~~~~~~~~~~~~~~~~~~
              + CategoryInfo          : NotSpecified: (Unhandled excep...w an exception.:String) [], RemoteExcep 
             tion
              + FullyQualifiedErrorId : NativeCommandError
           
           ---> System.IO.FileNotFoundException: The configuration file 'appsettings.json' was not found and is 
          not optional. The expected physical path was 'C:\WINDOWS\system32\appsettings.json'.
             at 
          Microsoft.Extensions.Configuration.FileConfigurationProvider.HandleException(ExceptionDispatchInfo 
          info)
             at Microsoft.Extensions.Configuration.FileConfigurationProvider.Load(Boolean reload)
             at Microsoft.Extensions.Configuration.FileConfigurationProvider.Load()
             at Microsoft.Extensions.Configuration.ConfigurationRoot..ctor(IList`1 providers)
             at Microsoft.Extensions.Configuration.ConfigurationBuilder.Build()
             at nanoFramework.Tools.FirmwareFlasher.FirmwarePackage..cctor() in 
          D:\a\1\s\nanoFirmwareFlasher.Library\FirmwarePackage.cs:line 120
             --- End of inner exception stack trace ---
             at nanoFramework.Tools.FirmwareFlasher.FirmwarePackage.GetTargetList(Boolean communityTargets, 
          Boolean preview, Nullable`1 platform, VerbosityLevel verbosity) in 
          D:\a\1\s\nanoFirmwareFlasher.Library\FirmwarePackage.cs:line 169
             at nanoFramework.Tools.FirmwareFlasher.Program.RunOptionsAndReturnExitCodeAsync(Options o) in 
          D:\a\1\s\nanoFirmwareFlasher.Tool\Program.cs:line 267
             at CommandLine.ParserResultExtensions.WithParsedAsync[T](ParserResult`1 result, Func`2 action)
             at 
          nanoFramework.Tools.FirmwareFlasher.Extensions.CommandLineExtensions.WithNotParsedAsync[T](Task`1 
          task, Func`2 errorFunc) in D:\a\1\s\nanoFirmwareFlasher.Tool\Extensions\CommandLineExtensions.cs:line 
          31
             at nanoFramework.Tools.FirmwareFlasher.Program.Main(String[] args) in 
          D:\a\1\s\nanoFirmwareFlasher.Tool\Program.cs:line 92
             at nanoFramework.Tools.FirmwareFlasher.Program.<Main>(String[] args)

It looks like this version required an appsettings.json file. 

Note, that the access to the target is working:

        PS C:\WINDOWS\system32> nanoff --listports
        .NET nanoFramework Firmware Flasher v2.5.49+da497901db
        Copyright (C) 2019 .NET Foundation and nanoFramework project contributors
        
        
        Available COM ports:
          COM44

alberk8, 2023-12-22:
@Ellerbach ,  Is there a spelling error [here](https://github.com/nanoframework/nanoFirmwareFlasher/blob/da497901dba0d894554503363ebd56741664a1da/nanoFirmwareFlasher.Library/appsettings.json#L2C1-L2C1)?

josesimoes, 2023-12-22:
@alberk8 confirmed 😅

josesimoes, 2023-12-22:
@romankiss `nanoff --platform esp32` will throw an error as this combination is not actually doing anything...

romankiss, 2023-12-22:
@josesimoes  you are right, it is my mistake, the correct command is: 

    nanoff --listtargets --platform esp32 

so, anyway it doesn't work for version 2.5.49

Thanks

josesimoes, 2023-12-22:
Do you have this installed at the default .NET tools location or at a specific folder?

romankiss, 2023-12-22:
Yes, it is at the default location. Note, that the updates worked very well over years without any problem, just now the latest update such as a version 2.5.49 has this issue. I made a "rollback" to the version 2.5.46 to continue my work.

josesimoes, 2023-12-22:
Things change and sometimes there are hiccups. 
Don't worry that this will be fixed and will resume working for many more years. 😉

closed 2023-12-22 by josesimoes

romankiss, 2023-12-22:
@josesimoes thanks for fixing this issue. I have updated nanoff to the latest version such as 2.5.52. The process of the _Flashing firmware succeeded but it took ~5 minutes for target ESP32_S3. Note, that the version 2.5.46 this flashing took <10 seconds.

Ellerbach, 2023-12-23:
> I have updated nanoff to the latest version such as 2.5.52. The process of the _Flashing firmware_ succeeded, but it took ~5 minutes for target ESP32_S3. Note, that the version 2.5.46 this flashing took <10 seconds.

This may be due to the fact the binary needs first to be downloaded. If already downloaded, the cache is used. So if you reflash, it should definitely be faster!

romankiss, 2023-12-23:
@Ellerbach Thanks for tip. I have deleted a cache for my target ESP32_S3 (ESP32_S3-1.9.0.178) and 

        PS C:\WINDOWS\system32> nanoff --target ESP32_S3 --serialport COM3 --update
        .NET nanoFramework Firmware Flasher v2.5.52+8258d78e70
        Copyright (C) 2019 .NET Foundation and nanoFramework project contributors
        
        
        Reading details from chip...OK
        
        Connected to:
        ESP32-S3 (ESP32-S3 (revision v0.1))
        Features WiFi, BLE
        Flash size 8MB GD25Q64 from GIGADEVICE (manufacturer 0x200 device 0x16407)
        PSRAM: undetermined
        Crystal 40MHz
        MAC F4:12:FA:9E:18:80
        
        Downloading firmware package...OK
        Extracting ESP32_S3-1.9.0.178.zip...OK
        
        Updating to 1.9.0.178
        
        Flashing firmware...OK 

so, I can see the process of the downloading. The complete time included flashing took ~5 minutes, 
Then I made a reflashing of firmware:

        PS C:\WINDOWS\system32> nanoff --target ESP32_S3 --serialport COM3 --update
        .NET nanoFramework Firmware Flasher v2.5.52+8258d78e70
        Copyright (C) 2019 .NET Foundation and nanoFramework project contributors
        
        
        Reading details from chip...OK
        
        Connected to:
        ESP32-S3 (ESP32-S3 (revision v0.1))
        Features WiFi, BLE
        Flash size 8MB GD25Q64 from GIGADEVICE (manufacturer 0x200 device 0x16407)
        PSRAM: undetermined
        Crystal 40MHz
        MAC F4:12:FA:9E:18:80
        
        Extracting ESP32_S3-1.9.0.178.zip...OK
        
        Updating to 1.9.0.178
        
        Flashing firmware...OK    

As you can see, there is no downloading process and the completion time is the same, such as ~5 minutes, so it looks like the problem is somewhere else.

alberk8, 2023-12-24:
Try this to get more details

`nanoff --target ESP32_S3 --serialport COM3 --update -v d`

romankiss, 2023-12-24:
@alberk8 Thanks for tip, the following is my result:

        PS C:\WINDOWS\system32> nanoff --target ESP32_S3 --serialport COM3 --update -v d
        .NET nanoFramework Firmware Flasher v2.5.52+8258d78e70
        Copyright (C) 2019 .NET Foundation and nanoFramework project contributors
        
        
        Using COM3 @ 1500000 baud to connect to ESP32.
        Reading details from chip...OK
        
        Connected to:
        ESP32-S3 (ESP32-S3 (revision v0.1))
        Features WiFi, BLE
        Flash size 8MB GD25Q64 from GIGADEVICE (manufacturer 0x200 device 0x16407)
        PSRAM: undetermined
        Crystal 40MHz
        MAC F4:12:FA:9E:18:80
        
        Trying to find ESP32_S3 in stable repository...OK
        Extracting ESP32_S3-1.9.0.178.zip...OK
        
        Updating to 1.9.0.178
        
        Flashing firmware...Read 2097152 bytes at 0x490000 in 234.6 seconds (71.5 kbit/s)...
        
        Uploading stub...                                                                                             
        Running stub...                                                                                               
        Stub running...                                                                                               
        Changing baud rate to 1500000                                                                                 
        Changed.                                                                                                      
        Configuring flash size...                                                                                     
        Flash will be erased from 0x00000000 to 0x00004fff...                                                         
        Flash will be erased from 0x00010000 to 0x00104fff...                                                         
        Flash will be erased from 0x00008000 to 0x00008fff...                                                         
        Flash will be erased from 0x00490000 to 0x0068ffff...                                                         
        Compressed 19920 bytes to 12651...                                                                            
        Writing at 0x00000000... (100 %)                                                                              
        Wrote 19920 bytes (12651 compressed) at 0x00000000 in 0.3 seconds (effective 553.6 kbit/s)...                 
        Hash of data verified.                                                                                        
        Compressed 1003328 bytes to 677302...                                                                         
        Writing at 0x00010000... (2 %)                                                                                
        Writing at 0x0001a8c8... (4 %)                                                                                
        Writing at 0x000241b1... (7 %)                                                                                
        Writing at 0x00031fc5... (9 %)                                                                                
        Writing at 0x00037519... (11 %)                                                                               
        Writing at 0x0003c8cb... (14 %)                                                                               
        Writing at 0x00041c3a... (16 %)                                                                               
        Writing at 0x00047274... (19 %)                                                                               
        Writing at 0x0004c63f... (21 %)                                                                               
        Writing at 0x00051add... (23 %)                                                                               
        Writing at 0x0005731b... (26 %)                                                                               
        Writing at 0x0005cb13... (28 %)                                                                               
        Writing at 0x00061695... (30 %)                                                                               
        Writing at 0x00066a70... (33 %)                                                                               
        Writing at 0x0006b90a... (35 %)                                                                               
        Writing at 0x000710a4... (38 %)                                                                               
        Writing at 0x000769f7... (40 %)                                                                               
        Writing at 0x0007c2d9... (42 %)                                                                               
        Writing at 0x000812f1... (45 %)                                                                               
        Writing at 0x000862f3... (47 %)                                                                               
        Writing at 0x0008bf2e... (50 %)                                                                               
        Writing at 0x00091844... (52 %)                                                                               
        Writing at 0x00096cd3... (54 %)                                                                               
[... cut ...]
        Wrote 262144 bytes (770 compressed) at 0x003c0000 in 1.6 seconds (effective 1296.4 kbit/s)...                 
        Hash of data verified.                                                                                        
                                                                                                                      
        Leaving...                                                                                                    
        Hard resetting via RTS pin...                                                                                 
        
        OK

romankiss, 2023-12-24:
@alberk8 @josesimoes I found more errors at the version 2.5.52, see the following, when I try to flash a previously firmware version, such as 1.9.0.169:

        PS C:\Users\rkiss> nanoff --target ESP32_S3 --serialport COM3 --update -v d --fwversion 1.9.0.169
        .NET nanoFramework Firmware Flasher v2.5.52+8258d78e70
        Copyright (C) 2019 .NET Foundation and nanoFramework project contributors
        
        
        Using COM3 @ 1500000 baud to connect to ESP32.
        Reading details from chip...OK
        
        Connected to:
        ESP32-S3 (ESP32-S3 (revision v0.1))
        Features WiFi, BLE
        Flash size 8MB GD25Q64 from GIGADEVICE (manufacturer 0x200 device 0x16407)
        PSRAM: undetermined
        Crystal 40MHz
        MAC F4:12:FA:9E:18:80
        
        Trying to find ESP32_S3 in stable repository...OK
        Downloading firmware package...OK
        Extracting ESP32_S3-1.9.0.169.zip...OK
        
        Updating to 1.9.0.169
        
        Flashing firmware...Error E4000: Error executing esptool command. (Could not find file 'C:\Users\rkiss\.nanoFramework\fw_cache\ESP32_S3\partitions_nanoclr_8mb.csv'.)

It looks like starting a fwversion 1.9.0.178, we have in the cache *.csv files, which they must be deleted for nanoff versions <= 2.5.46.