nanoframework/Home #1420: Create SpiDevice on SPI2 through an ArgumentException on ESP32_S3(module ESP32-S3R16V)

VergilGao, 2023-12-27:
### Library/API/IoT binding

System.Device.Spi

### Visual Studio version

VS2022(17.8.3)

### .NET nanoFramework extension version

2022.3.0.67

### Target name(s)

ESP32_S3

### Firmware version

1.9.0.178

### Device capabilities

ESP32-S3 (ESP32-S3 (revision v0.2))
Features WiFi, BLE
Flash size 16MB GD25Q128 from GIGADEVICE (manufacturer 0x200 device 0x16408)
PSRAM: undetermined
Crystal 40MHz

### Description

I have a ESP32_S3 board, the FSPI( 0 in Arduino ) connect to Octal SPI PSRAM so i have to use HSPI( 1 in Arduino ) 
here is my init code in Arduino:
```
#include <SPI.h>
#include "src/mcp25xx/mcp25xxfd.h"

#define MCP_25XX_CS (int8_t)4
#define MCP_25XX_SCK (int8_t)5
#define MCP_25XX_MISO (int8_t)6
#define MCP_25XX_MOSI (int8_t)7

#define MCP_25XX_SPI_FREQUENCY (uint32_t)40UL * 1000 * 1000

SPIClass spi(HSPI); 
Mcp25xxFd can(spi, MCP_25XX_SPI_FREQUENCY, MCP_25XX_CS); 

void setup()
{
    Serial.begin(115200);
    delay(1000);
    Serial.println("debug start!");
    spi.begin(MCP_25XX_SCK, MCP_25XX_MISO, MCP_25XX_MOSI);
    delay(1000);
    auto errorCode = can.begin();

    Serial.println(errorCode, HEX);
}
```
It works well, so i tried the same configuration in nanoframework:
```
const int ClockFrequency = 40_000_000;

Configuration.SetPinFunction(Gpio.IO07, DeviceFunction.SPI2_MOSI);
Configuration.SetPinFunction(Gpio.IO06, DeviceFunction.SPI2_MISO);
Configuration.SetPinFunction(Gpio.IO05, DeviceFunction.SPI2_CLOCK);

var spiConnectionSettings = new SpiConnectionSettings(2, Gpio.IO04)
{
    ClockFrequency = ClockFrequency,
    Mode = SpiMode.Mode0,
    ChipSelectLineActiveState = PinValue.Low,
    DataFlow = DataFlow.MsbFirst,
    Configuration = SpiBusConfiguration.FullDuplex
};

var spiDevice = SpiDevice.Create(spiConnectionSettings); // throw
```
then it through an ArgumentException.

If I use SPI1, I cant connect to my board after a reboot.

Ellerbach, 2024-01-02:
@VergilGao please use SPI1 instead of 2rt how it's going? Don't bother about the fact that something else may be used, it's transparent.

VergilGao, 2024-01-02:
> @VergilGao please use SPI1 instead of 2rt how it's going? Don't bother about the fact that something else may be used, it's transparent.

if use SPI1, I couldn't attach to the debug anymore.

Ellerbach, 2024-01-02:
> if use SPI1, I couldn't attach to the debug anymore.

Debug is using Serial, not SPI. Please use SPI1 and report back.

josesimoes, 2024-01-02:
Than it's something with the code that's causing a hard fault or a reboot which makes de device unresponsive...

VergilGao, 2024-01-02:
> > if use SPI1, I couldn't attach to the debug anymore.
> 
> Debug is using Serial, not SPI. Please use SPI1 and report back.

OK， i will try tomorrow. and paste the output here.

VergilGao, 2024-01-03:
First, this code run well, and I can attach the debug after a reboot.
```
Thread.Sleep(1000);

Debug.WriteLine("Hello world!");

Thread.Sleep(Timeout.Infinite);
```

then, i run
```
Thread.Sleep(1000);

const int ClockFrequency = 40_000_000;

Configuration.SetPinFunction(Gpio.IO07, DeviceFunction.SPI1_MOSI);
Configuration.SetPinFunction(Gpio.IO06, DeviceFunction.SPI1_MISO);
Configuration.SetPinFunction(Gpio.IO05, DeviceFunction.SPI1_CLOCK);

var spiConnectionSettings = new SpiConnectionSettings(1, Gpio.IO04)
{
    ClockFrequency = ClockFrequency,
    Mode = SpiMode.Mode0,
    ChipSelectLineActiveState = PinValue.Low,
    DataFlow = DataFlow.MsbFirst,
    Configuration = SpiBusConfiguration.FullDuplex
};

var spiDevice = SpiDevice.Create(spiConnectionSettings);

//var can = new Mcp25xxFd(spiDevice);

//var errorCode = can.Begin();

//Debug.WriteLine(errorCode.ToString());

Debug.WriteLine(spiDevice.ConnectionSettings.BusId.ToString());

Thread.Sleep(Timeout.Infinite);
```

OK, I got `1` in the debug view.

then, I stop, and start the debug. it shows:
```
** .NET nanoFramework extension v2022.3.0.67 loaded **
GitHub repo: https://github.com/nanoframework/Home
Report issues: https://github.com/nanoframework/Home/issues
Browse samples: https://github.com/nanoframework/samples
Join our Discord community: https://discord.gg/gCyBu8T
Join our Hackster.io platform: https://www.hackster.io/nanoframework
Follow us on Twitter: https://twitter.com/nanoframework
Follow our YouTube channel: https://www.youtube.com/c/nanoFramework
Star our GitHub repos: https://github.com/nanoframework/Home
Add a short review or rate the VS extension: https://marketplace.visualstudio.com/items?itemName=nanoframework.nanoFramework-VS2022-Extension

10:03:09.896 [PortSerial device watcher started @ Thread 31 [ProcessID: 17460]]
10:03:09.896 [PortTcpIp network watcher started @ Thread 32 [ProcessID: 17460]]
10:03:09.901 [NanoDevices: new device arrival COM5]
10:03:09.913 [NanoDevices: candidate nano device COM5]
10:03:09.979 [NanoDevices: checking device  COM5 @ 921600]
10:03:10.646 [NanoDevices: valid device ESP32_S3 @ COM5]
10:03:10.647 [NanoDevices: Serial device enumeration completed. Found 1 devices]
10:03:21.740 [Starting deployment transaction from v2022.3.0.67+395fd66257]
10:03:21.742 [Starting debug engine on nanoDevice]
10:03:21.751 [Connecting to debugger engine...OK]
10:03:22.921 [Assemblies to deploy:]
10:03:22.921 [Assembly: AirChips v1.0.0.0 (3512 bytes)]
10:03:22.923 [Assembly: System.Net v1.10.64.0 (20888 bytes)]
10:03:22.923 [Assembly: System.Device.Gpio v1.1.38.0 (5868 bytes)]
[... 131 lines cut ...]
10:03:54.574 [Attempting to connect the debugger engine (4/5)]
10:03:57.581 [Attempting to connect the debugger engine (5/5)]
10:04:03.089 [Attempting to connect the debugger engine (1/5)]
10:04:03.089 [Assigning debug engine of target device]
10:04:04.595 [Attempting to connect the debugger engine (2/5)]
10:04:06.602 [Attempting to connect the debugger engine (3/5)]
10:04:09.109 [Attempting to connect the debugger engine (4/5)]
10:04:12.117 [Attempting to connect the debugger engine (5/5)]
10:04:18.125 [Attempting to connect the debugger engine (1/5)]
10:04:18.125 [Assigning debug engine of target device]
10:04:19.632 [Attempting to connect the debugger engine (2/5)]
10:04:21.639 [Attempting to connect the debugger engine (3/5)]
10:04:24.146 [Attempting to connect the debugger engine (4/5)]
10:04:27.153 [Attempting to connect the debugger engine (5/5)]
10:04:33.661 [*** ERROR: all attempts to initialize the device state have failed! ***]
10:04:33.662 [Exception starting CLR on device: Could not reconnect to the debugging target after rebooting it..]
```
and the debug view:
```
Attaching to nanoDevice...
Waiting for nanoDevice to initialize...
The nanoDevice target and the debugger engine failed to initialize because of unspecified device errors.
程序“[1] .NET nanoFramework application: 托管”已退出，返回值为 0 (0x0)。
The debugger engine thread has terminated unexpectedly with error 'Could not reconnect to the debugging target after rebooting it.'.
```

alberk8, 2024-01-03:
@VergilGao ,  

GPIO 6 to GPIO 11 are exposed in some ESP32 development boards. However, these pins are connected to the integrated SPI flash on the ESP-WROOM-32 chip and are not recommended for other use.

VergilGao, 2024-01-03:
@alberk8  
I download the firmware then run the `hello world` code at the first run.

Then according to the document below, I modified the pin definition of SPI1
[image]

```
Thread.Sleep(1000);

const int ClockFrequency = 40_000_000;

Configuration.SetPinFunction(Gpio.IO13, DeviceFunction.SPI1_MOSI);
Configuration.SetPinFunction(Gpio.IO12, DeviceFunction.SPI1_MISO);
Configuration.SetPinFunction(Gpio.IO14, DeviceFunction.SPI1_CLOCK);

var spiConnectionSettings = new SpiConnectionSettings(1, Gpio.IO15)
{
    ClockFrequency = ClockFrequency,
    Mode = SpiMode.Mode0,
    ChipSelectLineActiveState = PinValue.Low,
    DataFlow = DataFlow.MsbFirst,
    Configuration = SpiBusConfiguration.FullDuplex
};

var spiDevice = SpiDevice.Create(spiConnectionSettings);

//var can = new Mcp25xxFd(spiDevice);

//var errorCode = can.Begin();

//Debug.WriteLine(errorCode.ToString());

Debug.WriteLine(spiDevice.ConnectionSettings.BusId.ToString());

Thread.Sleep(Timeout.Infinite);
```
then， it show `1` at the second run. and broken at the thrid.
```
** .NET nanoFramework extension v2022.3.0.67 loaded **
GitHub repo: https://github.com/nanoframework/Home
Report issues: https://github.com/nanoframework/Home/issues
Browse samples: https://github.com/nanoframework/samples
Join our Discord community: https://discord.gg/gCyBu8T
Join our Hackster.io platform: https://www.hackster.io/nanoframework
Follow us on Twitter: https://twitter.com/nanoframework
Follow our YouTube channel: https://www.youtube.com/c/nanoFramework
Star our GitHub repos: https://github.com/nanoframework/Home
Add a short review or rate the VS extension: https://marketplace.visualstudio.com/items?itemName=nanoframework.nanoFramework-VS2022-Extension

13:03:13.648 [PortTcpIp network watcher started @ Thread 32 [ProcessID: 18800]]
13:03:13.647 [PortSerial device watcher started @ Thread 31 [ProcessID: 18800]]
13:03:13.652 [NanoDevices: new device arrival COM5]
13:03:13.664 [NanoDevices: candidate nano device COM5]
13:03:13.960 [NanoDevices: checking device  COM5 @ 921600]
13:03:15.088 [NanoDevices: checking device  COM5 @ 460800]
13:03:16.167 [NanoDevices: checking device  COM5 @ 115200]
13:03:19.228 [NanoDevices: checking device  COM5 *** 2nd attempt ***]
13:03:19.275 [NanoDevices: checking device  COM5 @ 921600]
13:03:21.299 [NanoDevices: checking device  COM5 @ 460800]
13:03:21.855 [NanoDevices: valid device ESP32_S3 @ COM5]
13:03:21.855 [NanoDevices: Serial device enumeration completed. Found 1 devices]
13:03:29.511 [Starting deployment transaction from v2022.3.0.67+395fd66257]
13:03:29.514 [Starting debug engine on nanoDevice]
13:03:29.531 [Connecting to debugger engine...OK]
13:03:30.884 [Assemblies to deploy:]
13:03:30.884 [Assembly: AirChips v1.0.0.0 (3368 bytes)]
13:03:30.893 [Assembly: System.Net v1.10.64.0 (20888 bytes)]
13:03:30.893 [Assembly: System.Device.Gpio v1.1.38.0 (5868 bytes)]
13:03:30.893 [Assembly: System.Threading v1.1.32.63105 (3884 bytes)]
13:03:30.894 [Assembly: System.Device.Spi v1.3.43.0 (2952 bytes)]
[... 391 lines cut ...]
13:06:11.900 [Attempting to connect the debugger engine (3/5)]
13:06:14.407 [Attempting to connect the debugger engine (4/5)]
13:06:17.413 [Attempting to connect the debugger engine (5/5)]
13:06:22.421 [Attempting to connect the debugger engine (1/5)]
13:06:22.421 [Assigning debug engine of target device]
13:06:23.928 [Attempting to connect the debugger engine (2/5)]
13:06:25.936 [Attempting to connect the debugger engine (3/5)]
13:06:28.443 [Attempting to connect the debugger engine (4/5)]
13:06:31.449 [Attempting to connect the debugger engine (5/5)]
13:06:36.957 [Attempting to connect the debugger engine (1/5)]
13:06:36.957 [Assigning debug engine of target device]
13:06:38.464 [Attempting to connect the debugger engine (2/5)]
13:06:40.471 [Attempting to connect the debugger engine (3/5)]
13:06:42.978 [Attempting to connect the debugger engine (4/5)]
13:06:45.992 [Attempting to connect the debugger engine (5/5)]
13:06:52.000 [Attempting to connect the debugger engine (1/5)]
13:06:52.000 [Assigning debug engine of target device]
13:06:53.507 [Attempting to connect the debugger engine (2/5)]
13:06:55.513 [Attempting to connect the debugger engine (3/5)]
13:06:58.020 [Attempting to connect the debugger engine (4/5)]
13:07:01.027 [Attempting to connect the debugger engine (5/5)]
13:07:07.536 [*** ERROR: all attempts to initialize the device state have failed! ***]
13:07:07.536 [Exception starting CLR on device: Could not reconnect to the debugging target after rebooting it..]

```

alberk8, 2024-01-03:
I have run your sample code above and there it runs without disconnecting from the debugger.  Try remove the  `[Assembly: AirChips v1.0.0.0 (3516 bytes)]`

closed 2024-02-11 by VergilGao