nanoframework/Home #1422: Addition files in Firmware package

alberk8, 2023-12-28:
### Target name(s)

ESP32_BLE_REV0

### Firmware version

1.9.0.178

### Description

Flushing the firmware a second time is not successful with the following error.

```
Extracting ESP32_BLE_REV0-1.9.0.178.zip...Unhandled exception. System.IO.IOException: The file 'C:\Users\user\.nanoFramework\fw_cache\ESP32_BLE_REV0\partitions_nanoclr_16mb.csv' already exists.
   at Microsoft.Win32.SafeHandles.SafeFileHandle.CreateFile(String fullPath, FileMode mode, FileAccess access, FileShare share, FileOptions options)
   at Microsoft.Win32.SafeHandles.SafeFileHandle.Open(String fullPath, FileMode mode, FileAccess access, FileShare share, FileOptions options, Int64 preallocationSize)
   at System.IO.Strategies.OSFileStreamStrategy..ctor(String path, FileMode mode, FileAccess access, FileShare share, FileOptions options, Int64 preallocationSize)
   at System.IO.Strategies.FileStreamHelpers.ChooseStrategyCore(String path, FileMode mode, FileAccess access, FileShare share, FileOptions options, Int64 preallocationSize)
   at System.IO.Strategies.FileStreamHelpers.ChooseStrategy(FileStream fileStream, String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, FileOptions options, Int64 preallocationSize)
   at System.IO.FileStream..ctor(String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, Boolean useAsync)
   at System.IO.Compression.ZipFileExtensions.ExtractToFile(ZipArchiveEntry source, String destinationFileName, Boolean overwrite)
   at System.IO.Compression.ZipFileExtensions.ExtractRelativeToDirectory(ZipArchiveEntry source, String destinationDirectoryName, Boolean overwrite)
   at System.IO.Compression.ZipFileExtensions.ExtractToDirectory(ZipArchive source, String destinationDirectoryName, Boolean overwriteFiles)
   at System.IO.Compression.ZipFile.ExtractToDirectory(String sourceArchiveFileName, String destinationDirectoryName, Encoding entryNameEncoding, Boolean overwriteFiles)
   at nanoFramework.Tools.FirmwareFlasher.FirmwarePackage.DownloadAndExtractAsync() in D:\a\1\s\nanoFirmwareFlasher.Library\FirmwarePackage.cs:line 408
   at nanoFramework.Tools.FirmwareFlasher.Esp32Firmware.DownloadAndExtractAsync(Esp32DeviceInfo deviceInfo) in D:\a\1\s\nanoFirmwareFlasher.Library\Esp32Firmware.cs:line 76
   at nanoFramework.Tools.FirmwareFlasher.Esp32Operations.UpdateFirmwareAsync(EspTool espTool, Esp32DeviceInfo esp32Device, String targetName, Boolean updateFw, String fwVersion, Boolean preview, String applicationPath, String deploymentAddress, String clrFile, Boolean fitCheck, VerbosityLevel verbosity, Nullable`1 partitionTableSize) in D:\a\1\s\nanoFirmwareFlasher.Library\Esp32Operations.cs:line 353
   at nanoFramework.Tools.FirmwareFlasher.Program.RunOptionsAndReturnExitCodeAsync(Options o) in D:\a\1\s\nanoFirmwareFlasher.Tool\Program.cs:line 688
   at CommandLine.ParserResultExtensions.WithParsedAsync[T](ParserResult`1 result, Func`2 action)
   at nanoFramework.Tools.FirmwareFlasher.Extensions.CommandLineExtensions.WithNotParsedAsync[T](Task`1 task, Func`2 errorFunc) in D:\a\1\s\nanoFirmwareFlasher.Tool\Extensions\CommandLineExtensions.cs:line 31
   at nanoFramework.Tools.FirmwareFlasher.Program.Main(String[] args) in D:\a\1\s\nanoFirmwareFlasher.Tool\Program.cs:line 92
   at nanoFramework.Tools.FirmwareFlasher.Program.<Main>(String[] args) 
```

### How to reproduce

1. Delete the c:/users/username/.nanoframework/fw_cache
2. nanoff --target ESP_BLE_REV0 --update --serialport COM6 
3. repeat step 2

### Expected behaviour

Flash firmware without error.   The screenshot below shows that there are additional files in the firmware zip file.

### Screenshots

[image]

josesimoes, 2024-01-02:
@alberk8 those CSV files are there on purpose. This was adeded recently to improve the update workflow of ESP32 devives.
I can't reproduce the above using the latest nanoff version... care to check if you have the latest one installed?

alberk8, 2024-01-03:
I have upgraded to latest nanoff version V2.5.54 and it stop at Flashing firmware for a while before actually flashing the firmware. 

So the new behaviour is to preserve the config partition and write back during the actual flushing. 

`nanoff --target ESP32_REV0 --update --serialport com6 -v d`

`Flashing firmware...Read 262144 bytes at 0x3c0000 in 23.6 seconds (89.0 kbit/s)...`

josesimoes, 2024-01-03:
Yes. It now preserves the config partition. As for the delay, that's up to the esptool. Call are made with the same parameters, nothing else. I guess that's could be the flash erase step.

OK to close this, correct?

alberk8, 2024-01-03:
For those who don't want to preserve the config partition,  add the --masserase to delete everything and the flashing is faster.

`nanoff --target ESP32_REV0 --update --serialport com6 --masserase -v d`

closed 2024-01-03 by alberk8