nanoframework/Home #1423: Nanoff unable to upload managed application

torbacz, 2024-01-01:
### Tool

nanoff

### Description

Unable to deploy managed application file to device via nanoff.

The last version that it was working was 2.5.38. From version 2.5.43 there is an error. Probably the issue was created by moving to "managers" here 
https://github.com/nanoframework/nanoFirmwareFlasher/pull/242/files#diff-1ea0e59f56df5cf2e680f93d56fba49ea3fc942012cc6088737ab272f412e32b

### How to reproduce

1. nanoff  --target ESP32_REV3 --update --serialport COM9 --deploy --image "C:\Sources\private\FMT\Device\FMT.Device.ESP32\bin\Release\test.bin"

Command fails with 
`Error E4000: Error executing esptool command. (Value cannot be null. (Parameter 'path1'))`

torbacz, 2024-01-01:
I've found a root cause 
https://github.com/nanoframework/nanoFirmwareFlasher/pull/254/files#diff-0105837069c3b0ce14c09864cabb5e65b9d0174870e5ef5042cad7b397a42ba4R453

string partitionCsvFile = Path.Combine(firmware.LocationPath, $"partitions_nanoclr_{Esp32DeviceInfo.GetFlashSizeAsString(esp32Device.FlashSize).ToLowerInvariant()}.csv");
Path.Combiane is failing because firmware.LocationPath is null. Replacing it with string.Empty works, but then program cannot find partitions_nanoclr_4mb.csv file. Not quite sure what should be there.
@josesimoes Any ideas?

josesimoes, 2024-01-02:
@torbacz `firmware.LocationPath` can't be null or empty otherwise there is no firmware file to update and the code shouldn't even be there...

josesimoes, 2024-01-02:
Question about the above: what you're really after is UPDATING **AND** DEPLOYING an image?

torbacz, 2024-01-02:
Imagine if you hade like 20 new boards to flash. My current workflow is:
- nanoff --target ESP32_REV3 --update --serialport COM9
- open VisualStudio and deploy managed application via it

It takes a lot of time (switching between tools, resetting VistualStudio, unplugging USB in between)
Then I found that nanoff can upload managed application (thanks for that BTW ;) )

Now i'm trying call only one command with nanoff:
- nanoff --target ESP32_REV3 --update --serialport COM9 --deploy --image "C:\Sources\private\FMT\Device\FMT.Device.ESP32\bin\Release\test.bin"

Which is failing due to firmware.LocationPath being null.

Could be that I've tested nanoff with command (without update):
nanoff --target ESP32_REV3 --serialport COM9 --deploy --image "C:\Sources\private\FMT\Device\FMT.Device.ESP32\bin\Release\test.bin"

Please keep in mind that the same commands were working with version 2.5.38 before introducing massErase.

josesimoes, 2024-01-02:
I get it now and I was able to reproduce it.
This was "masked" as the deploy option used to be handled separately from the update. For sometime now it's part of it, but it had some edge case that was recently uncovered by the improved update process now trying to preserve the config partition.

closed 2024-01-02 by josesimoes

torbacz, 2024-01-02:
@josesimoes Thanks!