nanoframework/Home #1424: SerialPort issues while using ReadLine and the DataReceived event

Ellerbach, 2024-01-01:
### Library/API/IoT binding

System.IO.Ports

### Target name(s)

ESP32_PSRAM_REV0 but problem seems to exist with all the ESP32

### Firmware version

1.9.0.178

### Description

Reading thru the event is not properly working when reading with ReadLine

### How to reproduce

Flash an ESP32, setup pins to use COM2. And then, use a TTL or equivalent for sending/receiving on the serial port.
Use the simple program like this:

```cshap
using nanoFramework.Hardware.Esp32;
using System;
using System.Diagnostics;
using System.IO.Ports;
using System.Threading;

namespace TestSerial
{
    public class Program
    {
        private static SerialPort _port;

        public static void Main()
        {
            Debug.WriteLine("Hello from nanoFramework!");
            Configuration.SetPinFunction(15, DeviceFunction.COM2_RX);
            Configuration.SetPinFunction(2, DeviceFunction.COM2_TX);

            _port = new SerialPort($"COM2", 115_200);
            _port.ReadTimeout = 15_000;
            _port.WriteTimeout = 1_000;
            _port.DataReceived += SerialDataReceived;
            _port.Open();

            Thread.Sleep(Timeout.Infinite);
        }

        private static void SerialDataReceived(object sender, SerialDataReceivedEventArgs e)
        {
            if (e.EventType == SerialData.Chars)
            {
                if ((int)_port.BytesToRead > 0)
                {
                    try
                    {
                        string url;
                        Console.WriteLine($"bytes to read: {_port.BytesToRead}");
                        url = _port.ReadLine(); //.Trim('\n');
                        Console.WriteLine(url);
                        if (string.IsNullOrEmpty(url))
                        {
                            return;
                        }

                        bool res = false;
                        res = url.Contains("combo");

                        string ret = res ? "OK" : "KO";
                        _port.WriteLine(ret);

                    }
                    catch (Exception ex)
                    {
                        Console.WriteLine($"Exception in serial: {ex}");
                    }
                }
            }
        }
    }
}
```

You will not get the exact amount of available bytes and reading the ReadLine will give rubbish. Ultimately, an exception is raised after some time and also this does reboot the board for no reason.

To have it crashing much faster, just remove the `if ((int)_port.BytesToRead > 0)` and things crashes in few iterations

```text
bytes to read: 20
combo?rd=0&bl=0&ch=0    20419 bytes

bytes to read: 3
h=0bo?rd=0&bl=0&ch=0combo?rd=0&bl=0&ch=0    0 elements)

bytes to read: 1
h=0combo?rd=0&bl=0&ch=0
combo?rd=0&bl=0&ch=0
wlkshfqlskdfh����������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������?h=0combo?rd=0&bl=0&ch=0
combo?rd=0&bl=0&ch=0
wlkshfqlskdfh��������������������������������������������������������������������������������������������������������������������������������������
    ++++ Exception System.ArgumentOutOfRangeException - CLR_E_OUT_OF_RANGE (3) ++++
    ++++ Message: 
    ++++ System.String::IndexOf [IP: 0000] ++++
    ++++ System.String::Contains [IP: 0010] ++++
    ++++ System.IO.Ports.SerialPort::OnSerialDataReceivedInternal [IP: 0029] ++++
    ++++ System.IO.Ports.SerialDeviceEventListener::OnEvent [IP: 002e] ++++
    ++++ nanoFramework.Runtime.Events.EventSink::ProcessEvent [IP: 0023] ++++
    ++++ nanoFramework.Runtime.Events.EventSink::EventDispatchCallback [IP: 0014] ++++
Exception thrown: 'System.ArgumentOutOfRangeException' in mscorlib.dll
Exception in serial: System.ArgumentOutOfRangeException: Exception was thrown: System.ArgumentOutOfRangeException
```

DarkNoir, 2024-02-12:
This code is wrong.
ReadLine is a Blocking call that will return a portion of the buffer up to the defined NewLine chars. You cannot use it in the DataRecieved event. ReadLine is used in a blocking pattern only and usually with ReadTimout set to Timeout.Infinate.
`
                    while (_IsOnline)
                    {
                            string data = _SerialPort.ReadLine();
                            ReceiveData(data);
                    }
` 

However there is an issue with ReadLine as it ignores the defined SerialPort.NewLine value and looks like it uses null term instead "\0" . This causes it to return multiple lines, as many as there is up to whatever line termination it's using instead of the specified one. 

Sometimes, SerialPort.ReadTime out is set to ultimately throw an exception if nothing is received.

DarkNoir, 2024-02-12:
The workaround for the issue as I reported is take the data returned by ReadLine and String.Split it by whatever NewLine designator you are trying to use. Then use the result as intended.

closed 2024-03-22 by josesimoes