nanoframework/Home #1425: The connected target does not have support for nanoFramework.Device.Bluetooth

wolfgangmauer, 2024-01-05:
### Library/API/IoT binding

nanoFramework.Device.Bluetooth

### Visual Studio version

VS2022

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF 3572900
  Target:   ESP32_REV3
  Platform: ESP32

Firmware build Info:
  Date:        Jan  3 2024
  Type:        MinSizeRel build, chip rev. 3, without support for PSRAM
  CLR Version: 1.9.0.970
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  NFApp2, 1.0.0.0
  System.Device.Gpio, 1.1.38.0
  mscorlib, 1.15.6.0
  nanoFramework.Runtime.Events, 1.11.15.0

Native Assemblies:
  mscorlib v100.5.0.19, checksum 0x445C7AF9
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.10.0, checksum 0x6A20A689
  nanoFramework.Hardware.Esp32.Rmt v100.0.4.0, checksum 0x608C5658
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.System.IO.Hashing v100.0.0.1, checksum 0xEBD8ED20
  nanoFramework.System.Security.Cryptography v100.0.0.1, checksum 0xC71CFC75
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.2, checksum 0x545A6C79
  System.Math v100.0.5.5, checksum 0x9F9E2A7E
  System.Net v100.1.5.0, checksum 0x5BAB8CB3
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.2, checksum 0xFA806D33
  System.Device.I2c.Slave v1.0.0.0, checksum 0x4238164B
  System.Device.I2s v100.0.0.1, checksum 0x478490FE
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Runtime.Serialization v100.0.0.0, checksum 0x0A066871
  System.Device.Wifi v100.0.6.4, checksum 0x00A058C6
  Windows.Storage v100.0.2.0, checksum 0x954A4192

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffe46bc  0x0001b000
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x1F0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x1F0000 (1984kB)   Deployment

Deployment Map
Empty

### Description

I have some [ESP32-WROOM-32](https://www.amazon.de/dp/B0CLCNQK8W)
If i try a BLE sample i get
22:32:53.938 [NanoDevices: new device arrival COM5]
22:32:53.963 [NanoDevices: candidate nano device COM5]
22:32:53.979 [NanoDevices: checking device  COM5 @ 921600]
22:32:54.559 [NanoDevices: valid device ESP32_REV3 @ COM5]
22:32:54.559 [NanoDevices: Serial device enumeration completed. Found 1 devices]
22:33:02.285 [Starting deployment transaction from v2022.3.0.67+395fd66257]
22:33:02.285 [Starting debug engine on nanoDevice]
22:33:02.285 [Connecting to debugger engine...OK]
22:33:03.790 [The connected target does not have support for nanoFramework.Device.Bluetooth.]

alberk8, 2024-01-06:
@wolfgangmauer , You can try the ESP32_BLE_REV3 or ESP32_BLE_REV0 firmware for BLE support.

josesimoes, 2024-01-06:
@wolfgangmauer moving this to discussions. As the requested in the issue initial context, these are to be used to track work and confirmed issues. 
For support and questions please use discussions or, for more interactive support, our [Discord community](https://discordapp.com/invite/gCyBu8T).