nanoframework/Home #1427: Can't install any M5Stack NuGet package v1.1.192

romankiss, 2024-01-07:
### Library/API/IoT binding

M5Stack Board Support Packages v1.1.192

### Description

The latest version of the NuGet package v1.1.192 (also v1.1.190) is not available to download it.

### Screenshots

[image]

[image]

josesimoes, 2024-01-08:
There is currently an issue with our signing certificate from the .NET Foundation which is preventing the nuget packages from being successfully validated by NuGet. This has been reported and we're waiting to hear back from them about a fix.

romankiss, 2024-01-08:
@josesimoes Thanks for info, it happen at version 1.1.190 (3 weeks ago).

romankiss, 2024-01-26:
@josesimoes presently, there is the newest version of the M5Stack NuGet package such as **v1.1.199**, but still we can't make a successful up to date all assemblies during the runtime (linking) time, where some assemblies are updated in the source repository, but there are missing their NuGet packages, for instance in my case:

        Assembly: NFAppM5StickC_Testing (1.0.0.0) needs assembly 'Iot.Device.Sht3x' (1.2.0.0)
        Assembly: NFAppM5StickC_Testing (1.0.0.0) needs assembly 'Iot.Device.Mpu6886' (1.2.0.0)
        Assembly: NFAppM5StickC_Testing (1.0.0.0) needs assembly 'nanoFramework.M5StickC' (1.1.199.0)
        Assembly: Iot.Device.Rtc (1.2.0.0) needs assembly 'UnitsNet.Temperature' (5.42.0.0)
        Assembly: Iot.Device.Sht3x (1.2.0.0) needs assembly 'UnitsNet.Temperature' (5.42.0.0)
        Assembly: Iot.Device.Sht3x (1.2.0.0) needs assembly 'UnitsNet.RelativeHumidity' (5.42.0.0)
        Assembly: nanoFramework.M5StickC (1.1.199.0) needs assembly 'Iot.Device.Mpu6886' (1.2.0.0)
        Assembly: nanoFramework.M5StickC (1.1.199.0) needs assembly 'Iot.Device.Rtc' (1.2.0.0)
        Assembly: Iot.Device.Mpu6886 (1.2.0.0) needs assembly 'UnitsNet.Temperature' (5.42.0.0)

Thanks
Roman

josesimoes, 2024-01-26:
Things have been stabilizing since the nuget publishing issue. But some packages may still be behind...
I've just started a release on those IoT ones, then will trigger the M5stack ones again. 
Hopefully in a couple of iterations we can get this settled.

romankiss, 2024-01-26:
Thanks, I can see the progress, for my case, there is missing one more update for assembly _Iot.Device.Rtc_:

      Assembly: NFAppM5StickC_Testing (1.0.0.0) needs assembly 'nanoFramework.M5StickC' (1.1.199.0)
      Assembly: Iot.Device.Rtc (1.2.0.0) needs assembly 'UnitsNet.Temperature' (5.42.0.0)
      Assembly: nanoFramework.M5StickC (1.1.199.0) needs assembly 'Iot.Device.Rtc' (1.2.0.0)

Note, that the above result requires to make a rollback of the **nanoFramework.System.Net** assembly to the version 1.10.68 because have been just updated to the newest version such as _1.10.70_.

romankiss, 2024-01-27:
Thanks @josesimoes I can confirm that the newest firmware _M5StickC 1.9.0.1043_ for M5Stack M5StickC device is working with all the latest NuGet packages for my application connected to Azure IoT Central, in other words, the IoT device is running up-to-date.

Thanks again.
Roman

networkfusion, 2024-01-27:
Closing as fixed after latest comment.

closed 2024-01-27 by networkfusion