nanoframework/Home #1428: WIFI Adapter scanning fails with CLR_E_INVALID_OPERATION after FW update to 1.9.0.976

jan1409, 2024-01-07:
### Target name(s)

ESP32_REV3

### Firmware version

1.9.0.976

### Was working before? On which version?

1.9.0.970

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF 3572900
  Target:   ESP32_REV3
  Platform: ESP32

Firmware build Info:
  Date:        Jan  6 2024
  Type:        MinSizeRel build, chip rev. 3, without support for PSRAM
  CLR Version: 1.9.0.976
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  DistanceSensor1, 1.0.0.0
  System.IO.Streams, 1.1.52.0
  UnitsNet.ElectricResistance, 5.42.0.0
  nanoFramework.System.Text, 1.2.54.0
  nanoFramework.Json, 2.2.103.0
  UnitsNet.Ratio, 5.42.0.0
  Iot.Device.Bmxx80, 1.2.0.0
  nanoFramework.Hardware.Esp32, 1.6.12.0
  System.Math, 1.5.43.0
  Iot.Device.Common.WeatherHelper, 1.2.0.0
  nanoFramework.System.Collections, 1.5.31.0
  nanoFramework.Runtime.Events, 1.11.15.0
  System.Device.I2c, 1.1.16.0
  UnitsNet.Pressure, 5.42.0.0
  UnitsNet.Duration, 5.42.0.0
  System.Buffers.Binary.BinaryPrimitives, 1.2.0.0
  System.Device.Gpio, 1.1.38.0
  nanoFramework.M2Mqtt.Core, 0.0.0.0
  System.Net, 1.10.64.0
  System.Device.Wifi, 1.5.71.0
  nanoFramework.M2Mqtt, 5.1.107.0
  System.Threading, 1.1.32.63105
  nanoFramework.Runtime.Native, 1.6.12.0
  UnitsNet.Illuminance, 5.42.0.0
  UnitsNet.Length, 5.42.0.0
  UnitsNet.RelativeHumidity, 5.42.0.0
  Iot.Device.Bh1750fvi, 1.2.0.0
  System.Device.Model, 1.2.0.0
  UnitsNet.Temperature, 5.42.0.0
  mscorlib, 1.15.6.0
  System.IO.Ports, 1.1.78.0
  Iot.Device.Ld2410, 1.0.0.0

Native Assemblies:
  mscorlib v100.5.0.19, checksum 0x445C7AF9
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.10.0, checksum 0x6A20A689
  nanoFramework.Hardware.Esp32.Rmt v100.0.4.0, checksum 0x608C5658
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.System.IO.Hashing v100.0.0.1, checksum 0xEBD8ED20
  nanoFramework.System.Security.Cryptography v100.0.0.1, checksum 0xC71CFC75
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.2, checksum 0x545A6C79
  System.Math v100.0.5.5, checksum 0x9F9E2A7E
  System.Net v100.1.5.0, checksum 0x5BAB8CB3
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.2, checksum 0xFA806D33
  System.Device.I2c.Slave v1.0.0.0, checksum 0x4238164B
  System.Device.I2s v100.0.0.1, checksum 0x478490FE
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Runtime.Serialization v100.0.0.0, checksum 0x0A066871
  System.Device.Wifi v100.0.6.4, checksum 0x00A058C6
  Windows.Storage v100.0.2.0, checksum 0x954A4192

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffe46bc  0x0001b000
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x1F0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x1F0000 (1984kB)   Deployment

Deployment Map
Empty

### Description

After updating to Firmware version 1.9.0.976 the connection to Wireless network is throwing an error `CLR_E_INVALID_OPERATION` when trying to run the scan-command. 

I have tested to run the sample [Wifi scan](https://github.com/nanoframework/Samples/tree/main/samples/Wifi/ScanWiFi)  on a just updated device where it was failing - on the same device with 1.9.0.970 the code is working. 

I assume there is an issue in the current firmware?! 

### How to reproduce

1. Download the sample ScanWiFi App [here](https://github.com/nanoframework/Samples/tree/main/samples/Wifi/ScanWiFi)
2. Execute the code on a device with Firmware 1.9.0.976 and ESP32 built with ESP-IDF 3572900 (at least)

### Expected behaviour

Expect to connect to a Wireless network. 

### Aditional information

```
var wifiAdapter = WifiAdapter.FindAllAdapters()[0];
try
{
    Debug.WriteLine("starting Wi-Fi scan");
    wifiAdapter.ScanAsync(); //<-- issue will be raised here
}
catch (Exception ex)
{
    Debug.WriteLine($"Failure starting a scan operation: {ex}");
}
```

alberk8, 2024-01-08:
@jan1409 ,  Can try to flash the firmware with the --masserrase parameter 

`nanoff --target ESP32_REV3 --update --serialport COM3 --masserase -v d`

jan1409, 2024-01-08:
Hi @alberk8 , 
thanks very much for this hint! 
I ran the nanoff command and afterwards it was working Fine!. 
But I ran into the same behavior again when I was activating the Access point mode  - reboot - Disable Access point mode - reboot (which should set back to normal Wireless mode). 
However, this might be even my coding issue which I have to investigate further. 

Thanks for the help!

closed 2024-01-08 by jan1409

alberk8, 2024-01-08:
@jan1409,  The error you are  getting is due to how the underlaying Esp32 WIFI works.   When you do a Connect the ssid and password is save in the config and when it is rebooted it will use that to connect to WIFI even if the SSID or Password is wrong.  

The ScanAsync will error out when there is a on going connection attempt in the background.  If you have the ScanAsync in a for loop with Thread.Sleep(1_000) then one of the attempts will succeed.

jan1409, 2024-01-09:
@alberk8 , cool thanks. You are right this is working. Thanks very much for the help and sorry for the false alarm ...