nanoframework/Home #1429: Float and Double Console Output Value is not accurate

alberk8, 2024-01-09:
### Target name(s)

ESP32_REV0

### Firmware version

1.9.0.976

### Was working before? On which version?

N/A

### Description

Console.WriteLine of float or double does not produce the correct output value.  For example 33.1f should show 33.1 and not 33.099985 as the input value does not have trailing digits.

### How to reproduce

**Code**
```
 public static void Main()
 {
     Debug.WriteLine("Hello from nanoFramework!");

     float f1 = 33.1f;
     float f2 = 33.2f; 
     float f3 = 33.3f;
     float f4 = 33.4f;
     float f5 = 33.5f;
     float f6 = 33.6f;
     float f7 = 33.7f;
     float f8 = 33.8f;
     float f9 = 33.9f;
     float f10 = 34f;

     Debug.WriteLine($"Float1: {f1}");
     Debug.WriteLine($"Float2: {f2}");
     Debug.WriteLine($"Float3: {f3}");
     Debug.WriteLine($"Float4: {f4}");
     Debug.WriteLine($"Float5: {f5}");
     Debug.WriteLine($"Float6: {f6}");
     Debug.WriteLine($"Float7: {f7}");
     Debug.WriteLine($"Float8: {f8}");
     Debug.WriteLine($"Float9: {f9}");
     Debug.WriteLine($"Float10: {f10}");
     
     if (f8 + f8 == 67.6f)
     {
         Debug.WriteLine($"FloatX1: {f8 + f8}");
     }
     else
     {
         Debug.WriteLine($"FloatX2: {f8 + f8}");
     }

     double d1 = 33.1d;
     double d2 = 33.2d;
     double d3 = 33.3d;
     double d4 = 33.4d;
     double d5 = 33.5d;
     double d6 = 33.6d;
     double d7 = 33.7d;
     double d8 = 33.8d;
     double d9 = 33.9d;
     double d10 = 34d;

     Debug.WriteLine($"double1: {d1}");
     Debug.WriteLine($"double2: {d2}");
     Debug.WriteLine($"double3: {d3}");
     Debug.WriteLine($"double4: {d4}");
     Debug.WriteLine($"double5: {d5}");
     Debug.WriteLine($"double6: {d6}");
     Debug.WriteLine($"double7: {d7}");
     Debug.WriteLine($"double8: {d8}");
     Debug.WriteLine($"double9: {d9}");
     Debug.WriteLine($"double10: {d10}");

     Thread.Sleep(Timeout.Infinite);

 }
```

**Debug Output**
```
Hello from nanoFramework!
Float1: 33.0999985
Float2: 33.2
Float3: 33.2999992
Float4: 33.4000015
Float5: 33.5
Float6: 33.5999985
Float7: 33.7
Float8: 33.7999992
Float9: 33.9000015
Float10: 34
FloatX1: 67.5999985
double1: 33.1
double2: 33.2
double3: 33.299999999999996
double4: 33.399999999999999
double5: 33.5
double6: 33.6
double7: 33.7
double8: 33.799999999999999
double9: 33.899999999999997
double10: 33.999999999999999
```

### Expected behaviour

The output should be consistent.  For example, the float and double are one decimal place and the output should should one decimal place.

TerryFogg, 2024-01-09:
I’m guessing 33.1 is not a number that can be stored accurately in binary. What do you get running this on a pc?

alberk8, 2024-01-09:
@TerryFogg on PC it is all correct https://dotnetfiddle.net/ydVqA7

Ellerbach, 2024-01-09:
> @TerryFogg on PC it is all correct https://dotnetfiddle.net/ydVqA7

real full .NET is too smart.
33.1 can't be stored properly in a float with IEEE 754!
If you want, you can use a less smart language like Python ;-) 

```python
num = 33.1
float_num = float(num)
print(float_num)  # Output: 33.099998474121094

// And even 33.2

num = 33.2
float_num = float(num)
print(float_num)  # Output: 33.20000076293945

// And go on and on!
```

Ellerbach, 2024-01-10:
> Rounding for display purpose?

Yes, of course: `Console.Writeline($"My value: {val:0.0}");` or a `val.ToString("0.0")` should just work fine. We do have the formatters for floats and other elements.

Ellerbach, 2024-01-29:
So, after finally some time for investigations, if you want to adjust the number of digits, you can use the following:

```csharp
            Debug.WriteLine($"Float1: {f1.ToString("F1")}");
            Debug.WriteLine($"Float2: {f2:F1}");
```

Both will give you 1 digit. You can adjust the number of digits. Some others from here https://learn.microsoft.com/en-us/dotnet/standard/base-types/standard-numeric-format-stringscan be supported.

My suggestiong about "0.0" or even "#.#" seems that is not working in .NET nanoFramework.
So, closing this issue as it's not really an issue, it's just the way the values are stored.

closed 2024-01-29 by Ellerbach