nanoframework/Home #1430: Implement the WPS conection method for ESP32

AdrianSoundy, 2024-01-10:
### Description

Add support for the WPS wifi connection method for ESP32.

### How to solve the problem

There would need to be C# api that the application can call which will start the WPS connection process. This could be an async call to start process and then monitor events on C# side or call waits until whole process is complete in an async way.

On the native side you would need to start WPS and monitor the WPS events until complete or error.

FDE-LC2L, 2024-01-13:
I would really appreciate this feature. This would simplify the use of some of my IoTs.

Thank you in advance to the nanoFramework team for studying this request.

FDE-LC2L, 2024-01-13:
I would really appreciate this feature. This would simplify the use of some of my IoT.

Thank you in advance to the nanoFramework team for studying this request.