nanoframework/Home #1432: CS0433 error: conflict when using nanoFramework.Iot.Device.AtModem and nanoFramework.WebServer

NimaHp, 2024-01-20:
### Library/API/IoT binding

nanoFramework.WebServer

### Visual Studio version

vs2022 (17.8.5)

### .NET nanoFramework extension version

2022.3.0.67

### Target name(s)

ALL

### Description

using nanoFramework.Iot.Device.AtModem (1.0.65) and nanoFramework.WebServer (1.2.14) nuget packages both latest version. Need to use HttpStatusCode but the compiler gives error: CS0433	The type 'HttpStatusCode' exists in both 'Iot.Device.AtModem, Version=1.0.0.0, Culture=neutral, PublicKeyToken=c07d481e9758c731' and 'System.Net.Http, Version=1.5.113.0, Culture=neutral, PublicKeyToken=c07d481e9758c731'

### How to reproduce

1. Install nanoFramework.Iot.Device.AtModem and nanoFramework.WebServer.
2. use anything from System.Net like HttpStatusCode in the code.

alberk8, 2024-01-20:
There is an [issue](https://discord.com/channels/478725473862549535/481780524864503828/1193869905242624031) with new nuget not being published.  It is awaiting for fixing.  You have to use the firmware that is a least one week back.

networkfusion, 2024-01-27:
The nuget has now been updated.

closed 2024-01-27 by networkfusion

NimaHp, 2024-01-31:
update nuget packages to the latest but the error  persists.
[image]

networkfusion, 2024-01-31:
> update nuget packages to the latest but the error persists. ![Error Screenshot](https://private-user-images.githubusercontent.com/12736825/301275434-5db8bef3-8356-4ba7-a74b-8417adb63c44.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY3MjgxNDMsIm5iZiI6MTcwNjcyNzg0MywicGF0aCI6Ii8xMjczNjgyNS8zMDEyNzU0MzQtNWRiOGJlZjMtODM1Ni00YmE3LWE3NGItODQxN2FkYjYzYzQ0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTMxVDE5MDQwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNlZjIwOGMzYjZmNGRhMDY2NWJjZmMzNTg0ZGQ1Y2UzZjI0YWYyZWY2MTYzZjM0NmQ5Y2M2OTY0MDgwNWJhYjUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.rYtR3e3FXoaSCBBjHxJTI1AoptFPEkosH7Q0Ac_GnWc)

It seems that the `HttpStatusCode` is being used from the `System.Net` lib (as a global enum)` rather than the intended one from the `AtModem` lib. Try using something like `Iot.Device.AtModem.HttpStatusCode.BadRequest` instead.