nanoframework/Home #1436: Missing native assembly nanoFramework.Graphics in the target AtomS3

romankiss, 2024-01-26:
### Target name(s)

AtomS3

### Firmware version

1.9.0.1037

### Was working before? On which version?

no 

### Device capabilities

        System Information
        HAL build info: nanoCLR running @ ESP32_S3 built with ESP-IDF 3572900
          Target:   AtomS3
          Platform: ESP32
        
        Firmware build Info:
          Date:        Jan 25 2024
          Type:        MinSizeRel build, support for PSRAM
          CLR Version: 1.9.0.1037
          Compiler:    GNU ARM GCC v8.4.0
        
        OEM Product codes (vendor, model, SKU): 0, 0, 0
        
        Serial Numbers (module, system):
          00000000000000000000000000000000
          0000000000000000
        
        Target capabilities:
          Has nanoBooter: NO
          IFU capable: NO
          Has proprietary bootloader: YES
        
        AppDomains:
        
        Assemblies:
        
        Native Assemblies:
          mscorlib v100.5.0.19, checksum 0x445C7AF9
          nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
          nanoFramework.Hardware.Esp32 v100.0.10.0, checksum 0x6A20A689
          nanoFramework.Hardware.Esp32.Rmt v100.0.4.0, checksum 0x608C5658
          nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
          nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
          nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
          nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
          nanoFramework.System.IO.Hashing v100.0.0.1, checksum 0xEBD8ED20
          nanoFramework.System.Security.Cryptography v100.0.0.1, checksum 0xC71CFC75
          nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
          EventSink v1.0.0.0, checksum 0xF32F4C3E
          System.IO.FileSystem v1.0.0.2, checksum 0x545A6C79
          System.Math v100.0.5.5, checksum 0x9F9E2A7E
          System.Net v100.2.0.0, checksum 0xA6875261
          System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
          System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
          System.Device.I2c v100.0.0.2, checksum 0xFA806D33
          System.Device.I2s v100.0.0.1, checksum 0x478490FE
          System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
          System.IO.Ports v100.1.6.1, checksum 0xB798CE30
          System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
          System.Runtime.Serialization v100.0.0.0, checksum 0x0A066871
          System.Device.Wifi v100.0.6.4, checksum 0x00A058C6
          Windows.Storage v100.0.2.0, checksum 0x954A4192
        
        
        
        ++++++++++++++++++++++++++++++++
        ++        Memory Map          ++
        ++++++++++++++++++++++++++++++++
          Type     Start       Size
        ++++++++++++++++++++++++++++++++
          RAM   0x3fcaa830  0x0003e000
          FLASH 0x00000000  0x00800000
        
        
        
        +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
        ++                   Flash Sector Map                        ++
        +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
          Region     Start      Blocks   Bytes/Block    Usage
        +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
              0    0x00010000       1      0x1A0000     nanoCLR
              1    0x001B0000       1      0x2E0000     Deployment
              2    0x00490000       1      0x200000     Configuration
        
        
        +++++++++++++++++++++++++++++++++++++++++++++++++++
        ++              Storage Usage Map                ++
        +++++++++++++++++++++++++++++++++++++++++++++++++++
          Start        Size (kB)           Usage
        +++++++++++++++++++++++++++++++++++++++++++++++++++
          0x00490000    0x200000 (2048kB)   Configuration
          0x00010000    0x1A0000 (1664kB)   nanoCLR
          0x001B0000    0x2E0000 (2944kB)   Deployment
        
        
        
        Deployment Map
        Empty

### Description

I would like to use an internal display of the **M5Stack AtomS3** device using the generic graphics driver supported by NuGet package nanoFramework.Graphics 1.2.4, but the firmware target **AtomS3** is missing its native assembly such as the nanoFramework.Graphics.

Note, that currently for this M5Stack AtomS3 device I am using a firmware target _ESP32_S3_ and external display without any problem to handle an Azure IoT Central Application.

romankiss, 2024-01-26:
@Ellerbach thank you for your info, I have assumed that this target _AtomS3_ has been created for the _M5Stack AtomS3_ device like other ones, for example M5StickC, M5StickCPlus, etc., in other words, I have expected all native assemblies included in the target AtomS3 for device M5Stack AtomS3.

Ellerbach, 2024-01-27:
Not for that one, it's a generic S3 image as so far, no screen driver. But if you're adding a screen driver, we can definitely build an image (at least the generic driver with a managed driver)

romankiss, 2024-01-29:
Thanks @Ellerbach, it will be nice to have also a S3 image with a screen driver. The following devices run a generic S3 image (no BLE & Screen) very well as an IoT Device to Azure IoT Central App:

- [AtomS3](https://shop.m5stack.com/products/atoms3-dev-kit-w-0-85-inch-screen)
- [M5Dial](https://shop.m5stack.com/products/m5stack-dial-esp32-s3-smart-rotary-knob-w-1-28-round-touch-screen)
- [CoreS3](https://shop.m5stack.com/products/m5stack-cores3-esp32s3-lotdevelopment-kit)
- [ESP32-S3-LCD-1.28](https://www.waveshare.com/esp32-s3-lcd-1.28.htm)
- [ESP32-S3-Touch-LCD-1.28](https://www.waveshare.com/product/esp32-s3-touch-lcd-1.28.htm)
- [ESP32-S3-GEEK](https://www.waveshare.com/product/esp32-s3-geek.htm)

Ellerbach, 2024-01-29:
@romankiss, thanks for pointing out on all those devices. I'm not opposed to add a generic driver image. It still requires a bit of work as it needs to be tested. And I don't have any of this device. If you have build sucessfully an image and if it runs without any issue, I'll be happy to review a PR to add this device and add the missing elements in the PR for the build system. Happy to assist as well in our Discord channel whcih will be easier for discussion than this issue.

Ellerbach, 2024-03-26:
@romankiss did you have a chance to create your own image for those devices? Anything that you can share back if that's working?

romankiss, 2024-03-26:
@Ellerbach , thanks for asking me. I didn't have a chance to create own image with a support for _nanoFramework.Graphics_ at the **ESP32_S3** target. I am not familiarly for this level of creating a custom image. 

The needs having a support for display at the ESP32_S3 target is increasing by new product from M5Stack such as [M5Stack Din Meter](https://shop.m5stack.com/products/m5stack-din-meter-w-m5stamps3) based on the [M5StampS3](https://shop.m5stack.com/products/m5stamp-esp32s3-module). 

Note, that the M5StampS3 product is working very well as a device of the Azure IoT Central app with a firmware ESP32_S3. M5Stack using this product for building some standart features for IoT Devices, see another their product such as [M5CapsuleS3](https://shop.m5stack.com/products/m5stack-capsule-kit-w-m5stamps3).

AndreiGoryankin, 2024-04-18:
@Ellerbach , I am having a very similar issue with the NuGet package nanoFramework.Graphics 1.2.4 for two more cases:

1) STM32 board **NUCLEO-F746ZG** NanoFramework target **ST_NUCLEO144_F746ZG**
2) ESP32 board **ESP32-DevKitC-VE** NanoFramework target **ESP32_PSRAM_REV0**

All I need is to create a bitmap and work with it, and then transfer the result via the serial port.  **No need to display anything.**
Something like this:

       Bitmap originalBitmap = new Bitmap(128, 128);

Compile works. but deploy fails with this message:

      The connected target does not have support for the following assembly(ies):  'nanoFramework.Graphics'

My list of native assemblies is almost the same as above.

Any advice on what can I do to resolve this?  Or, possibly, give me a few board options that will work for my task, but don't have a display on them?

josesimoes, 2024-04-18:
@AndreiGoryankin let's follow this on Discord, OK? No need to cross post.

antonsterkhov, 2024-04-21:
> @romankiss, thanks for pointing out on all those devices. I'm not opposed to add a generic driver image. It still requires a bit of work as it needs to be tested. And I don't have any of this [...]

I could test if you put together an assembly that supports the screen.

AdrianSoundy, 2024-06-08:
The ESP32_S3_BLE image now has the generic display driver.
There is no need to have a specific firmware for ATomS3 as the generic ESP32_S3_BLE can be used.

Does this resolve the issue ? @romankiss 

Really the AtomS3 firmware should be removed.

romankiss, 2024-06-08:
@AdrianSoundy , for now, when we don't have a _nanoFramework.AtomS3_ nuget assembly, the firmware AtomS3 should be removed.

Yes, the generic firmware **ESP32_S3_BLE** (screen, BLE) can be used for M5Stack products such as AtomS3, AtomS3Lite, AtomS3U, M5StampS3, M5Capsule, M5DinMeter and M5Dial, they are running well with my test IoT Application connected to the Azure IoT Central.

Also, I already successfully tested my IoT App with this firmware on the Waveshare products such as [ESP32-S3 Zero](https://www.waveshare.com/esp32-s3-zero.htm) and [ESP32-S3-GEEK.](https://www.waveshare.com/esp32-s3-geek.htm)

josesimoes, 2024-07-31:
@romankiss I suppose this one can be closed now, right?

closed 2024-07-31 by josesimoes