nanoframework/Home #1437: Regular Expression wrongly matching lower case alphabet

alberk8, 2024-01-27:
### Library/API/IoT binding

System.Text.RegularExpressions

### Visual Studio version

VS2022 17.8.5

### .NET nanoFramework extension version

2022.3.0.67

### Target name(s)

ESP32_REV0

### Firmware version

1.9.0.1043

### Description

Regular Expression is matching lower case Alphabets when the pattern is only for upper case

### How to reproduce

Use the code enclosed.  The return is TRUE.

### Expected behaviour

The regular expression should return FALSE.  This is confirm on .Net 7 and 8.

### Sample project or code

Regex AlphanumericRegex = new Regex("^[A-Z0-9 $%*+./:-]*$", RegexOptions.Compiled);
string s = "This is nano";
Console.WriteLine("Regex: " + AlphanumericRegex.IsMatch(s));

Ellerbach, 2024-01-27:
The regular expression is not really valid. It should be `^[A-Z0-9\ $%+.\/:-]$` (normally not necessary to escape space but give it a try)
Can you try again?
Also please note that the engine is quite limited

alberk8, 2024-01-28:
Yes the additional escape works.  I also need to use the verbatim identifier @ for the Regex pattern.

Ellerbach, 2024-01-29:
> Yes the additional escape works. I also need to use the verbatim identifier @ for the Regex pattern.

Great, so we can close this issue :-) Again, the RegEx lib is a bit sensitive and only handles simple and well-formed cases! Please improve the documentation if needed!

closed 2024-01-29 by Ellerbach