nanoframework/Home #1438: SpiDevice is not disposing on ESP32-C3 and blocks the execution

Ellerbach, 2024-01-28:
### Library/API/IoT binding

System.Device.Spi

### Target name(s)

XIAO_ESP32C3

### Firmware version

latest

### Description

SpiDevice dispose blocks the execution and is never disposing

### How to reproduce

            // Frequency is 38KHz in the protocol
            float t_carrier = 1 / 38.0f;
            // Reality is that there is a 2us difference in the output as there is always a 2us bit on on SPI using MOSI
            float t_ushort = t_carrier - 2e-3f;
            // Calulate the outpout frenquency. Here = 16/(1/38 -2^-3) = 658KHz
            int freq = (int)(16.0f * 1000 / t_ushort);

            Configuration.SetPinFunction(4, DeviceFunction.SPI1_CLOCK);
            Configuration.SetPinFunction(5, DeviceFunction.SPI1_MISO);
            Configuration.SetPinFunction(6, DeviceFunction.SPI1_MOSI);

            var settings = new SpiConnectionSettings(1, -1)
            {
                ClockFrequency = freq,
                Mode = SpiMode.Mode3,
                ChipSelectLineActiveState = PinValue.High,
            };

            _spi = new SpiDevice(settings);

            Thread.Sleep(1000);
            _spi.Dispose();
// This is then never ever executed!
_spi = new SpiDevice(settings);

Thread.Sleep(1000);

_spi.Dispose();

Thread.Sleep(Timeout.Infinite);

### Expected behaviour

SpiDevice should properly dispose

Ellerbach, 2024-01-29:
I've been trying on different ESP32-C3 models and they all have the same issue. There is no problem for a normal ESP32. So it seems related specifically to ESP32-C3.

gwnpeter, 2024-03-04:
i have some test code,may be the same problem ,on xiao_esp32_c3

josesimoes, 2024-04-24:
This is fixed with nanoframework/nf-interpreter#2915.
Can you please confirm?

josesimoes, 2024-06-12:
Pinging to the followers of this issue. Appreciate if anyone can retest this and confirm that is indeed fixed. Thanks! 😃

josesimoes, 2024-06-26:
Thanks for testing this.
Closing it as fixed. Can be reopened if needed.

closed 2024-06-26 by josesimoes