nanoframework/Home #1439: Error E9000 when specifying "--version"

matsujirushi, 2024-01-28:
### Tool

nanoff

### Description

```
C:\>nanoff --version
nanoff 2.5.56+94d47f695e
Error E9000: Invalid or missing arguments.

C:\>
```

Ellerbach, 2024-01-29:
You can also just sue `nanoff` without anything:

[image]

That say, yes, that should be fixed but not critical, so allocating a very low priority.

matsujirushi, 2024-01-29:
@josesimoes 
Okay.
Please assign me this issue. 😉

matsujirushi, 2024-01-29:
Note:
Built in version reporting
https://github.com/commandlineparser/commandline/wiki/Generating-Help-and-Usage-information#built-in-version-reporting

matsujirushi, 2024-01-31:
* `--help` return HelpRequestedError class. ([here](https://github.com/commandlineparser/commandline/blob/1e3607b97af6141743edb3c434c06d5b492f6fb3/src/CommandLine/Core/PreprocessorGuards.cs#L27))
* `--version` return VersionRequestedError class. ([here](https://github.com/commandlineparser/commandline/blob/1e3607b97af6141743edb3c434c06d5b492f6fb3/src/CommandLine/Core/PreprocessorGuards.cs#L36))

**Links:**
* [Design discussion: how to handle --help option](https://github.com/commandlineparser/commandline/issues/600#issuecomment-599987440)

closed 2024-01-31 by josesimoes