nanoframework/Home #1444: [BLE Discvovery] ServiceUuids is always empty

tarasverq, 2024-02-11:
### Library/API/IoT binding

nanoFramework.Device.Bluetooth

### Visual Studio version

VS2022

### Target name(s)

ESP32_BLE_REV3

### Firmware version

1.9.0.1095

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF 3572900
  Target:   ESP32_BLE_REV3
  Platform: ESP32

Firmware build Info:
  Date:        Feb 10 2024
  Type:        MinSizeRel build, chip rev. 3, support for PSRAM, support for BLE
  CLR Version: 1.9.0.1095
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  Central1, 1.0.0.0
  nanoFramework.System.Collections, 1.5.31.0
  nanoFramework.System.Text, 1.2.54.0
  nanoFramework.Runtime.Native, 1.6.12.0
  nanoFramework.Device.Bluetooth, 1.1.60.0
  nanoFramework.Runtime.Events, 1.11.15.0
  mscorlib, 1.15.6.0

Native Assemblies:
  mscorlib v100.5.0.19, checksum 0x445C7AF9
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.10.0, checksum 0x6A20A689
  nanoFramework.Hardware.Esp32.Rmt v100.0.4.0, checksum 0x608C5658
  nanoFramework.Device.Bluetooth v100.0.5.0, checksum 0xD163A9D5
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.System.IO.Hashing v100.0.0.1, checksum 0xEBD8ED20
  nanoFramework.System.Security.Cryptography v100.0.0.2, checksum 0xF4AEFE6C
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.2, checksum 0x545A6C79
  System.Math v100.0.5.5, checksum 0x9F9E2A7E
  System.Net v100.2.0.1, checksum 0xD82C1452
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.2, checksum 0xFA806D33
  System.Device.I2s v100.0.0.1, checksum 0x478490FE
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Runtime.Serialization v100.0.0.0, checksum 0x0A066871
  System.Device.Wifi v100.0.6.4, checksum 0x00A058C6
  Windows.Storage v100.0.3.0, checksum 0xF0C37E1B

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffe46bc  0x00013800
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x1F0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x1F0000 (1984kB)   Deployment

Deployment Map
Empty

### Description

I am trying to use BLE library, but I face the problem:
`ServiceUuids` is always empty. Even with example solution `Central1`.

Example output:
```
Central: Simple Bluetooth LE watcher
=== Starting Bluetooth advert watcher ====
Will run for 2 minutes and close down

=== Advert received ==== 01/01/1970 00:09:39
Address:7C2D0ACBB9 RSSI:-98
Local name:
Manufacturers Data:1
-- Company:117 Length:24
4241806007C2DACBB927C2DACBB81DFD10000
Service UUIDS:0

=== Advert received ==== 01/01/1970 00:09:39
Address:4EF4E99D0383 RSSI:-89
Local name:
Manufacturers Data:0
Service UUIDS:0

=== Advert received ==== 01/01/1970 00:09:39
Address:2FC75EF7709E RSSI:-49
Local name:
Manufacturers Data:1
-- Company:6 Length:27
192022B0F2C4EADE4029333CEB31B1455148C1E046A4860EF
Service UUIDS:0

=== Advert received ==== 01/01/1970 00:09:40
Address:5D80CF02618E RSSI:-85
Local name:
Manufacturers Data:1
-- Company:76 Length:7
105361CF768CA
Service UUIDS:0

=== Advert received ==== 01/01/1970 00:09:40
Address:BC7E8B30063B RSSI:-88
Local name:
Manufacturers Data:1
-- Company:117 Length:24
42418066BC7E8B3063BBE7E8B3063A1597F2D8D00
Service UUIDS:0

=== Advert received ==== 01/01/1970 00:09:40
Address:4EF4E99D0383 RSSI:-88
Local name:
Manufacturers Data:1
-- Company:224 Length:6
5892CA9B4FBB
Service UUIDS:0

=== Advert received ==== 01/01/1970 00:09:40
Address:5D80CF02618E RSSI:-78
Local name:
Manufacturers Data:0
Service UUIDS:0

=== Advert received ==== 01/01/1970 00:09:40
Address:74CB4A27F7AC RSSI:-89
Local name:
Manufacturers Data:1
-- Company:117 Length:24
21861B17FA5D26D572712A9BD4DC7A2FB93EE94810CFF2
Service UUIDS:0

=== Advert received ==== 01/01/1970 00:09:40
Address:74CB4A27F7AC RSSI:-89
Local name:
Manufacturers Data:0
Service UUIDS:0

=== Advert received ==== 01/01/1970 00:09:43
Address:C96CFAE98478 RSSI:-74
Local name:
Manufacturers Data:1
-- Company:76 Length:4
12201
Service UUIDS:0

=== Advert received ==== 01/01/1970 00:10:25
Address:E5D9BEFD4A88 RSSI:-71
Local name:
Manufacturers Data:1
-- Company:76 Length:4
12202
Service UUIDS:0

=== Advert received ==== 01/01/1970 00:10:25
Address:5C41173EF280 RSSI:-74
Local name:
Manufacturers Data:1
-- Company:117 Length:24
21861B13F24375971ECDEA631E2015D3A91320E1AD2953
Service UUIDS:0

=== Advert received ==== 01/01/1970 00:10:26
Address:5C41173EF280 RSSI:-74
Local name:
Manufacturers Data:0
Service UUIDS:0

=== Advert received ==== 01/01/1970 00:10:26
Address:4422AF26F185 RSSI:-77
Local name:
Manufacturers Data:1
-- Company:117 Length:24
21861B13F24F74D1158961448E33D3A71C3CC543026BB86
Service UUIDS:0

=== Advert received ==== 01/01/1970 00:10:27
Address:4422AF26F185 RSSI:-77
Local name:
Manufacturers Data:0
Service UUIDS:0

=== Advert received ==== 01/01/1970 00:10:50
Address:C4F092B068E5 RSSI:-46
Local name:
Manufacturers Data:0
Service UUIDS:0

=== Advert received ==== 01/01/1970 00:10:50
Address:C4F092B068E5 RSSI:-46
Local name:Galaxy Buds2 Pro LE
Manufacturers Data:0
Service UUIDS:0

=== Advert received ==== 01/01/1970 00:10:55
Address:2B55A696C66F RSSI:-70
Local name:
Manufacturers Data:0
Service UUIDS:0

=== Advert received ==== 01/01/1970 00:10:55
Address:2B55A696C66F RSSI:-72
Local name:BT-ZUBU_BLE
Manufacturers Data:1
-- Company:12592 Length:8
3233343536373839
Service UUIDS:0

=== Stopping Bluetooth advert watcher ====
```

All the listed devices has *zero* Service UUIDS. I checked it with android app named BLE Scanned, and I found that at least five devices has some discoverable services. 
Is it a bug? Or am I doing something wrong?

Nanoff device info:
```
Connected to:
ESP32 (ESP32-D0WD-V3 (revision v3.1))
Features WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Flash size 4MB unknown from TENX (manufacturer 0x94 device 0x16406)
PSRAM: undetermined
Crystal 40MHz
MAC 08:D1:F9:D7:73:A4
```

### Sample project or code

Just Cenral1 example

tarasverq, 2024-02-11:
Found that with fw version `1.8.1.233 ` and `mscorlib v100.5.0.19` and  `nanoFramework.Device.Bluetooth v100.0.5.0` it works okay.

networkfusion, 2024-05-09:
Please re-check and update this issue. If fixed, please let us know.

AdrianSoundy, 2024-06-08:
As no more information we can assume this issue is already fixed, so closing

closed 2024-06-08 by AdrianSoundy