nanoframework/Home #1453: Device not in `nanoff --listdevices` result

regis1512, 2024-03-24:
### Tool

nanoff

### Description

My device from "Freenove Super Starter Kit for ESP32-S3" is not found in the list of devices

### How to reproduce

1. I execute 
`nanoff --target ESP32_S3_BLE --serialport COM5 --update --reset --masserase`
2. I get this result : 

> .NET nanoFramework Firmware Flasher v2.5.59+e3ff02e768
> Copyright (C) 2019 .NET Foundation and nanoFramework project contributors
> 
> 
> Reading details from chip...OK
> 
> Connected to:
> ESP32-S3 (ESP32-S3 (revision v0.1))
> Features WiFi, BLE
> Flash size 8MB GD25Q64 from GIGADEVICE (manufacturer 0x200 device 0x16407)
> PSRAM: undetermined
> Crystal 40MHz
> MAC 34:85:18:6D:35:40
> 
> Extracting ESP32_S3_BLE-1.9.1.21.zip...OK
> 
> Updating to 1.9.1.21
> 
> Erasing flash...OK
> Flashing firmware...OK

3. I execute 
`nanoff --listdevices`
4. I get this result

> .NET nanoFramework Firmware Flasher v2.5.59+e3ff02e768
> Copyright (C) 2019 .NET Foundation and nanoFramework project contributors
> 
> 
> No devices found

### Expected behaviour

I expect that the device is in the list
(I suppose it is why the device do no appear in the devices list in Visual Studio)

### Aditional context

I use : 
- Visual Studio : 2022
- .NET nanoFramework Extension : 2022.3.0.73
- target name : ESP32_S3_BLE
- firmware version : 1.9.1.21
- nanooff : 2.5.59

regis1512, 2024-03-24:
I have follow the instructions in https://docs.nanoframework.net/content/getting-started-guides/troubleshooting-device-connection.html#finding-ports-on-windows

I have got ; 

[image]

and : 

[image]

So now, i do not know if the problem come from nanoff or from the nano framework image which is not compatible with the board or something else.

alberk8, 2024-03-25:
@regis1512 in the first screen capture, does the message run till the end.  On the same image is this the with a different baud rate?

regis1512, 2024-03-25:
The first screen capture is Putty with a baud rate of 115200 (like indicated at https://docs.nanoframework.net/content/getting-started-guides/troubleshooting-device-connection.html#special-esp32). The message run till the end : when i do the screen capture there is no more information which appear in Putty since many seconds.

The second screen capture is Putty with the baud rate of 921600 (like indicated at https://docs.nanoframework.net/content/getting-started-guides/troubleshooting-device-connection.html#finding-if-nanoframework-booted). The message run till the end too.

regis1512, 2024-03-25:
I have unplugged and plugged the cable many times.
I have tried other cables too.
Thanks.

alberk8, 2024-03-25:
The firmware that you are using expects that your Board has **TWO**  Usb C port.

josesimoes, 2024-03-25:
This is most likely the common situation of having it connected on the "wrong" USB connector.
I suppose that board of yours has a second USB connector? Please try connecting it there.
Also it would be helpful to include in the issue a link to the product page and/or schematic.

josesimoes, 2024-03-25:
Moving this to a discussion as we don't use issues to answer questions and this is not a bug report, as noted in the issue notes. 😉