nanoframework/Home #1458: System.IO.StreamWriter does not use the configured NewLine character

patagonaa, 2024-04-02:
### Library/API/IoT binding

nanoFramework.System.IO.Streams

### Description

When using a StreamWriter to write text to a stream, the newline characters are always `\r\n` instead of the newline character that was set using `NewLine`.

### Sample project or code

```csharp
using var ms = new MemoryStream();
var sw = new StreamWriter(ms) { NewLine = "\n" };
sw.WriteLine("Test");
sw.Flush();

var outBytes = ms.ToArray();
var output = Encoding.UTF8.GetString(outBytes, 0, outBytes.Length);
// output is "Test\r\n" instead of "Test\n"
```

Ellerbach, 2024-04-02:
Seems that's indeed an issue: https://github.com/nanoframework/System.IO.Streams/blob/e89f687c13268f80c08f818ed19bd32ccc125e2c/System.IO.Streams/StreamWriter.cs#L163 it's uses the internal constant rather than the parent NewLine.
We ❤️ PR, so if you're willing to fix the issue, that would be amazing! .NET nanoFramework is a community driven effort, so any effort makes the difference. Happy to assis on our Discord channel if help needed.

patagonaa, 2024-04-02:
I'll see what I can do.
I'll probably make a second PR for StreamWriter as well, because I found a pretty annoying performance issue as well while testing.

closed 2024-04-03 by josesimoes