nanoframework/Home #1464: Failing Unit Tests in nanoframework.System.Net.Http

patagonaa, 2024-04-08:
### Library/API/IoT binding

nanoFramework.System.Net.Http

### Description

There are some Unit Tests failing in nanoframework.System.Net.Http.
Namely:
- `HttpUtilityTest.UrlEncodeTest`
- Miscellaneous tests in `StreamContentTest`

These are not new issues related to nanoframework/System.Net.Http#416, but have been there before.
Broken unit tests should be verified (if they point to an actual issue), then fixed or deleted.

mizady, 2024-12-22:
I'm gonna go ahead and go through these and clean up the tests a bit and verify which ones are working properly and not.  It looks like the UriUnitTests are also in rough shape as well.

josesimoes, 2024-12-23:
@mizady awesome that you're willing to tackle this one! Thanks. 😃 
Also, as we now have the wonderful hardware tests, HTTP is a great candidate to run those and be able to run real network tests on devices. That would be bonus score.😉

mizady, 2024-12-30:
To note, the linked pull request should not close this issue. Though i believe it is required to solve this one.  I will be posting a draft pull request soon that will have the url encoding/decoding tests and fixes, and the uri tests that seemed to be passing initially but upon closer inspection were not.

josesimoes, 2024-12-30:
@mizady noted. It's a matter of replacing the verb that links to this issue to not use one of the "key" ones 😉 (done!).
.