nanoframework/Home #1467: Failing build dockeer.all v2.45

valoni, 2024-04-14:
### Library/API/IoT binding

N/A

### Description

failed to build and deploy docker all version v2.45

### Expected behaviour

https://github.com/nanoframework/nf-interpreter/actions/runs/8142685541

### Screenshots

[image]
[image]

alberk8, 2024-04-16:
> The size limit of a docker image is 10GB. I think we must be exceeding that now. We must see if we can optimize the install to remove anything un-necessery or look at layering.

How about deleting the older version(leaving only 5 version) of the images?.  After all, the default script is to get the latest version and anybody that needs a specific version older than that should be able to build on their own.

networkfusion, 2024-04-16:
> How about deleting the older version(leaving only 5 version) of the images?. After all, the default script is to get the latest version and anybody that needs a specific version older than that [...]

This would be about the size of the image, after adding all toolchain items like IDF, ChibiOS, etc. It is not a problem with storage of the image itself (as far as I can tell).

closed 2024-04-18 by josesimoes