nanoframework/Home #1469: Build Project Error

fujisheng, 2024-04-16:
### Tool

Visual Studio extension

### Description

[image]

egokb, 2024-04-17:
https://github.com/egokb/EMK.git

fujisheng, 2024-04-17:
I think I found the reason. I used a multidimensional array in my code. Thank you for your reply.

closed 2024-04-17 by fujisheng

josesimoes, 2024-04-17:
That could be it. Only jagged arrays are supported. See the FAQ entry [here](https://docs.nanoframework.net/content/architecture/simplifications-and-trade-offs.html#multidimensional-arrays)