nanoframework/Home #1470: `HostBuilder.Properties` and `HostBuilderContext.Properties` should be dictionaries not arrays

CoryCharlton, 2024-04-16:
### Library/API/IoT binding

nanoFramework.Hosting

### Description

`HostBuilder.Properties` and `HostBuilderContext.Properties` are currently arrays. This makes them challenging to work with as the values need to be set at a single time and the order of the items must be known.

.NET full implements these as dictionaries which makes more sense (https://github.com/dotnet/runtime/blob/69280cd085ca65e8cecee4d94888784f37271386/src/libraries/Microsoft.Extensions.Hosting/src/HostBuilder.cs#L52).

josesimoes, 2024-04-16:
Pinging @bytewizer to ask if there was a design motivation to use Arrays instead of dictionaries.

josesimoes, 2024-04-17:
IIRC the namespace being under nanoframework name means that we are deviating from the .NET API... 🤔

CoryCharlton, 2024-04-17:
> IIRC the namespace being under nanoframework name means that we are deviating from the .NET API... 🤔

Sure that makes sense in cases where we are deviating but this doesn't appear to be one as it's pretty similar. 

We made a namespace change in this [PR](https://github.com/nanoframework/nanoFramework.DependencyInjection/pull/33) for [nanoFramework.DependencyInjection](https://github.com/nanoframework/nanoFramework.DependencyInjection) which is part of my motivation for suggesting we do the same here.

josesimoes, 2024-07-31:
Another attempt to get feedback from @bytewizer about this... 😉 

@CoryCharlton if there is no feedback (as I don't see any issues with these changes) please go ahead.

josesimoes, 2024-11-18:
Wondering if there was any progress on this... 😉

josesimoes, 2025-02-05:
@CoryCharlton without any feedback from the library owner, please go ahead with the suggested change.

closed 2025-03-13 by josesimoes