nanoframework/Home #1471: Wifi support disappeared after upgrade to ESP32-C3-REV3 1.9.1.52

JohnMasen, 2024-04-17:
### Target name(s)

ESP32-C3 Devkit

### Firmware version

ESP32-C3-REV 1.9.1.47

### Was working before? On which version?

Yes, can't remember the firmware version, found my project was using nanoFramework.Hardware.Esp32 1.6.3 before this upgrade

### Device capabilities

WIFI feature becomes unavailable after upgrade.

### Description

1. Network config in VS extension reports "Connected nanoDevice doesn't have network capabilities. Can't open Network Configuration dialog."
[image]

3. ```System.Device.Wifi.WifiAdapter.FindAllAdapters()``` and ```NetworkInterface.GetAllNetworkInterfaces()``` will throw exception
[image]
[image]

### How to reproduce

run command ```nanoff --platform esp32 --serialport COM5 --update```

[image]

### Expected behaviour

The WIFI feature should work as before

### Aditional information

VS packages.config compare (before and after upgrading)
Before:(the firmware matches these packages works)
```
<?xml version="1.0" encoding="utf-8"?>
<packages>
  <package id="nanoFramework.Azure.Devices.Client" version="1.1.125" targetFramework="netnano1.0" />
  <package id="nanoFramework.CoreLibrary" version="1.14.2" targetFramework="netnano1.0" />
  <package id="nanoFramework.DependencyInjection" version="1.0.22" targetFramework="netnano1.0" />
  <package id="nanoFramework.Hardware.Esp32" version="1.6.3" targetFramework="netnano1.0" />
  <package id="nanoFramework.Hosting" version="1.0.25" targetFramework="netnano1.0" />
  <package id="nanoFramework.Iot.Device.DhcpServer" version="1.2.174" targetFramework="netnano1.0" />
  <package id="nanoFramework.Json" version="2.2.84" targetFramework="netnano1.0" />
  <package id="nanoFramework.Logging" version="1.1.63" targetFramework="netnano1.0" />
  <package id="nanoFramework.Logging.Stream" version="1.1.63" targetFramework="netnano1.0" />
  <package id="nanoFramework.M2Mqtt" version="5.1.79" targetFramework="netnano1.0" />
  <package id="nanoFramework.ResourceManager" version="1.2.13" targetFramework="netnano1.0" />
  <package id="nanoFramework.Runtime.Events" version="1.11.6" targetFramework="netnano1.0" />
  <package id="nanoFramework.Runtime.Native" version="1.6.6" targetFramework="netnano1.0" />
  <package id="nanoFramework.System.Collections" version="1.5.18" targetFramework="netnano1.0" />
  <package id="nanoFramework.System.Device.Wifi" version="1.5.54" targetFramework="netnano1.0" />
  <package id="nanoFramework.System.IO.FileSystem" version="1.1.23" targetFramework="netnano1.0" />
  <package id="nanoFramework.System.IO.Ports" version="1.1.60" targetFramework="netnano1.0" />
  <package id="nanoFramework.System.IO.Streams" version="1.1.38" targetFramework="netnano1.0" />
  <package id="nanoFramework.System.Math" version="1.5.29" targetFramework="netnano1.0" />
  <package id="nanoFramework.System.Net" version="1.10.52" targetFramework="netnano1.0" />
  <package id="nanoFramework.System.Net.Http" version="1.5.84" targetFramework="netnano1.0" />
  <package id="nanoFramework.System.Net.Http.Server" version="1.5.84" targetFramework="netnano1.0" />
  <package id="nanoframework.System.Net.Sockets.TcpClient" version="1.1.52" targetFramework="netnano1.0" />
  <package id="nanoFramework.System.Text" version="1.2.37" targetFramework="netnano1.0" />
  <package id="nanoFramework.System.Threading" version="1.1.19" targetFramework="netnano1.0" />
  <package id="nanoFramework.WebServer" version="1.1.65" targetFramework="netnano1.0" />
  <package id="nanoFramework.Windows.Storage" version="1.5.33" targetFramework="netnano1.0" />
  <package id="nanoFramework.Windows.Storage.Streams" version="1.14.24" targetFramework="netnano1.0" />
</packages>
```
After:(the firmware matches these packages has NO WIFI)
```
<?xml version="1.0" encoding="utf-8"?>
<packages>
  <package id="nanoFramework.Azure.Devices.Client" version="1.2.38" targetFramework="netnano1.0" />
  <package id="nanoFramework.CoreLibrary" version="1.15.5" targetFramework="netnano1.0" />
  <package id="nanoFramework.DependencyInjection" version="1.1.3" targetFramework="netnano1.0" />
  <package id="nanoFramework.Hardware.Esp32" version="1.6.12" targetFramework="netnano1.0" />
  <package id="nanoFramework.Hosting" version="1.0.40" targetFramework="netnano1.0" />
  <package id="nanoFramework.Iot.Device.DhcpServer" version="1.2.507" targetFramework="netnano1.0" />
  <package id="nanoFramework.Json" version="2.2.117" targetFramework="netnano1.0" />
  <package id="nanoFramework.Logging" version="1.1.94" targetFramework="netnano1.0" />
  <package id="nanoFramework.Logging.Stream" version="1.1.94" targetFramework="netnano1.0" />
  <package id="nanoFramework.M2Mqtt" version="5.1.123" targetFramework="netnano1.0" />
  <package id="nanoFramework.ResourceManager" version="1.2.19" targetFramework="netnano1.0" />
  <package id="nanoFramework.Runtime.Events" version="1.11.15" targetFramework="netnano1.0" />
  <package id="nanoFramework.Runtime.Native" version="1.6.12" targetFramework="netnano1.0" />
  <package id="nanoFramework.System.Collections" version="1.5.31" targetFramework="netnano1.0" />
  <package id="nanoFramework.System.Device.Wifi" version="1.5.81" targetFramework="netnano1.0" />
  <package id="nanoFramework.System.IO.FileSystem" version="1.1.41" targetFramework="netnano1.0" />
  <package id="nanoFramework.System.IO.Ports" version="1.1.84" targetFramework="netnano1.0" />
  <package id="nanoFramework.System.IO.Streams" version="1.1.59" targetFramework="netnano1.0" />
  <package id="nanoFramework.System.Math" version="1.5.43" targetFramework="netnano1.0" />
  <package id="nanoFramework.System.Net" version="1.10.77" targetFramework="netnano1.0" />
  <package id="nanoFramework.System.Net.Http" version="1.5.134" targetFramework="netnano1.0" />
  <package id="nanoFramework.System.Net.Http.Server" version="1.5.134" targetFramework="netnano1.0" />
  <package id="nanoframework.System.Net.Sockets.TcpClient" version="1.1.74" targetFramework="netnano1.0" />
  <package id="nanoFramework.System.Runtime" version="1.0.6" targetFramework="netnano1.0" />
  <package id="nanoFramework.System.Text" version="1.2.54" targetFramework="netnano1.0" />
  <package id="nanoFramework.System.Threading" version="1.1.32" targetFramework="netnano1.0" />
  <package id="nanoFramework.WebServer" version="1.2.40" targetFramework="netnano1.0" />
  <package id="nanoFramework.Windows.Storage" version="1.5.49" targetFramework="netnano1.0" />
  <package id="nanoFramework.Windows.Storage.Streams" version="1.14.32" targetFramework="netnano1.0" />
</packages>
```

josesimoes, 2024-04-17:
All ESP32 targets have Wifi (and network capabilities) enabled by default, so this has to be something else.
If you are able to deploy an app which references `nanoFramework.System.Net` that's an indication that network is enabled. Otherwise VS won't even deploy it.

Can you post here the output of device capabilites please?

yureng, 2024-04-18:
I'm having the same issue. 
after trying to：nanoff --target ESP32_REV3 --serialport COM5 --masserase --update,  the network is now available ！