nanoframework/Home #1474: Add support for IL0373 epaper driver

ThijsBorst, 2024-04-20:
### Description

I was hoping to utilize the nanoframework with my ThinkInk TriColor epaper display, but unfortunately there is no support yet. Could this be added?

### How to solve the problem

Add a driver for IL0373 based epaper devices. For example: https://www.adafruit.com/product/4778

Ellerbach, 2024-04-21:
.NET nanoFramework is a community project. So for bindings like screens, sensors, it's only based out of community work. So, you can check the existing drivers and with the datasheet, see how you can best add your needed driver: https://github.com/nanoframework/nanoFramework.IoT.Device/tree/develop/devices/ePaper

And we love PR, so we will be happy to review and help you in the process with feedback.
So closing this issue as we are not tracing the requests for bindings.

closed 2024-04-21 by Ellerbach

ThijsBorst, 2024-04-21:
Ok, got it... Busy on a driver then...