nanoframework/Home #1477: Support ESP32_S3 Octal SPIRAM

marcantoinebouchard, 2024-04-24:
### Description

Octal spiram is not supported for esp32s3 boards.

### How to solve the problem

Create a specific image for octal spiram support.

AdrianSoundy, 2024-04-24:
This has already been done in the 5.13 IDF pr.

networkfusion, 2024-05-09:
This PR had now been merged. Please confirm that it is fixed.

marcantoinebouchard, 2024-05-09:
> This PR had now been merged. Please confirm that it is fixed.

It fixes the problem for Oct PSRAM, but Quad PSRAM modules now only uses the internal memory.
I tested on 2 modules:
esp32s3 N16R8 (8MB psram oct): NativeMemory.GetMemoryInfo(NativeMemory.MemoryType.All returns 8.3MB
esp32s3 N8R2 (2MB psram quad): NativeMemory.GetMemoryInfo(NativeMemory.MemoryType.All returns 0.3MB (only the internal memory)

AdrianSoundy, 2024-05-09:
Thanks for testing. I'll see if I can get the quad memory working as well. I'll be doing an update in next day or 2.
@marcantoinebouchard

marcantoinebouchard, 2024-05-09:
@AdrianSoundy micropython uses 2 differents images (https://micropython.org/download/ESP32_GENERIC_S3). It looks like the oct and quad support are a build config, so nanoframework might need 2 different images as well

AdrianSoundy, 2024-05-15:
We can't support QUAD and OCTAL spiram in the firmware as IDF doesn't allow it to be configured.
To resolve this issue a new firmware called ESP32_S3_ALL will be created

This is for the bigger memory S3 which are octal psram.
As there is plenty of memory have also add BLE & Generic display so we only need one firmware instead of lots of different firmware for octal ram. The existing ESP32_S3 and ESP32_BLE will have support for QUAD psram.

closed 2024-05-15 by AdrianSoundy