nanoframework/Home #1479: ESP32 - nanoFramework.Json.DeserializationException

varogonz95, 2024-04-25:
### Library/API/IoT binding

nanoFramework.Json

### Visual Studio version

VS2022 (17.9.5)

### .NET nanoFramework extension version

2022.3.0.78

### Target name(s)

ESP32_REV3

### Firmware version

1.9.1.75

### Device capabilities

(Deployment area empty at the time of this capture)
```console
System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF 38eeba2
  Target:   ESP32_REV3
  Platform: ESP32

Firmware build Info:
  Date:        Apr 19 2024
  Type:        MinSizeRel build, chip rev. 3, without support for PSRAM
  CLR Version: 1.9.1.65
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:

Native Assemblies:
  mscorlib v100.5.0.19, checksum 0x445C7AF9
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.10.0, checksum 0x6A20A689
  nanoFramework.Hardware.Esp32.Rmt v100.0.4.0, checksum 0x608C5658
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.System.IO.Hashing v100.0.0.1, checksum 0xEBD8ED20
  nanoFramework.System.Security.Cryptography v100.0.0.2, checksum 0xF4AEFE6C
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.1.0.0, checksum 0x62EBC911
  System.Math v100.0.5.5, checksum 0x9F9E2A7E
  System.Net v100.2.0.1, checksum 0xD82C1452
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.2, checksum 0xFA806D33
  System.Device.I2c.Slave v1.0.0.0, checksum 0x4238164B
  System.Device.I2s v100.0.0.1, checksum 0x478490FE
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Runtime.Serialization v100.0.0.0, checksum 0x0A066871
  System.Device.Wifi v100.0.6.4, checksum 0x00A058C6
  Windows.Storage v100.0.3.0, checksum 0xF0C37E1B

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffe46bc  0x0001b000
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x1F0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x1F0000 (1984kB)   Deployment

Deployment Map
Empty
```

### Description

I'm getting a `nanoFramework.Json.DeserializationException` when trying to deserialize a response in correct JSON string format to an object.
Relevant code:
```cs
var json = JsonConvert.SerializeObject(request);
var data = new StringContent(json, Encoding.UTF8);

using var message = _httpClient.Post(RegisterDeviceUri, data);
message.EnsureSuccessStatusCode();

var jsonResponse = message.Content.ReadAsString(); // I also tried .ReadAsStream, no luck
response = (Response)JsonConvert.DeserializeObject(jsonResponse, typeof(Response));
```
This is the JSON response:
```json
{
"id":"xxxxxxxx",
"connectionState":0,
"status":0,
"authenticationType":0,
"symmetricPrimaryKey":"xxxxxxxxxxxxxxxxxxxx",
"symmetricSecondaryKey":"xxxxxxxxxxxxxxxxx",
"x509PrimaryThumbprint":null,
"x509SecondaryThumbprint":null
}
```
And this is the class im trying to deserialize to:
```cs
internal class RegisteredDeviceResponse
{
    public string Id { get; set; }
    public int ConnectionState { get; set; }
    public int Status { get; set; }
    public int AuthenticationType { get; set; }
    public string SymmetricPrimaryKey { get; set; }
    public string SymmetricSecondaryKey { get; set; }
    public string X509PrimaryThumbprint { get; set; }
    public string X509SecondaryThumbprint { get; set; }
}
```
The exception thrown:
```console
    ++++ Exception nanoFramework.Json.DeserializationException - 0x00000000 (1) ++++
    ++++ Message: 
    ++++ nanoFramework.Json.JsonConvert::GetNextTokenInternal [IP: 01e2] ++++
    ++++ nanoFramework.Json.JsonConvert::GetNextToken [IP: 0005] ++++
    ++++ nanoFramework.Json.JsonConvert::ParseValue [IP: 0006] ++++
    ++++ nanoFramework.Json.JsonConvert::ParseObject [IP: 0039] ++++
    ++++ nanoFramework.Json.JsonConvert::Deserialize [IP: 001b] ++++
    ++++ nanoFramework.Json.JsonConvert::Deserialize [IP: 0011] ++++
    ++++ nanoFramework.Json.JsonConvert::DeserializeObject [IP: 001a] ++++
    ++++ FloraBytes.Device.HttpClients.PgotchiHttpClient.PgotchiHttpClient::RegisterDevice [IP: 005e] ++++
    ++++ FloraBytes.Device.Application::RegisterDevice [IP: 0012] ++++
    ++++ FloraBytes.Device.Application::Start [IP: 0025] ++++
    ++++ FloraBytes.Device.Program::Main [IP: 0027] ++++
Exception thrown: 'nanoFramework.Json.DeserializationException' in nanoFramework.Json.dll
```

networkfusion, 2024-04-25:
Please try https://github.com/nanoframework/nanoFramework.Json?tab=readme-ov-file#case-insensitive-members-resolving

varogonz95, 2024-04-25:
Nope. Still doesn't work. 
I added `Settings.CaseSensitive = false;` at the top of `Main()` and keeps failing.
Additionally, I changed the properties of the `Response` class to camelCase, same output :(

varogonz95, 2024-04-25:
My code looks like this:
```cs
// Program.cs
public partial class Program
{
    public static void Main()
    {
        Settings.CaseSensitive = false;

        ConfigurePins();

        var app = (Application)
            ConfigureServices()
            .BuildServiceProvider()
            .GetRequiredService(typeof(Application));

        app.Start();
    }
}
```
```cs
// MyServiceHttpClient.cs
public Response PostRequest(Request request)
{
    Response response = null;
            
    try
    {
        var json = JsonConvert.SerializeObject(request); // <-- Serialization works though
        var data = new StringContent(json, Encoding.UTF8);
                
        using var message = _httpClient.Post(ServiceUri, data);
        message.EnsureSuccessStatusCode();

        var jsonResponse = message.Content.ReadAsString();
        Debug.WriteLine("HTTP Response");
        Debug.WriteLine("Content: \n" + jsonResponse);

        response = (Response)JsonConvert.DeserializeObject(jsonResponse, typeof(Response));
    }
    catch (Exception ex)
    {
        Debug.WriteLine($"An error occured on {nameof(RegisterDevice)}.");
        Debug.WriteLine(ex.Message);
        throw;
    }

    return response;
}
```
```cs
// Response.cs
internal class Response
{
    public string id { get; set; }
    public int connectionState { get; set; }
    public int status { get; set; }
    public int authenticationType { get; set; }
    public string symmetricPrimaryKey { get; set; }
    public string symmetricSecondaryKey { get; set; }
    public string x509PrimaryThumbprint { get; set; }
    public string x509SecondaryThumbprint { get; set; }
}
```

CoryCharlton, 2024-04-25:
This code works fine for me:

```c#
public class Program
{
    public static void Main()
    {
        Settings.CaseSensitive = false;

        var jsonResponse = "{\r\n\"id\":\"xxxxxxxx\",\r\n\"connectionState\":0,\r\n\"status\":0,\r\n\"authenticationType\":0,\r\n\"symmetricPrimaryKey\":\"xxxxxxxxxxxxxxxxxxxx\",\r\n\"symmetricSecondaryKey\":\"xxxxxxxxxxxxxxxxx\",\r\n\"x509PrimaryThumbprint\":null,\r\n\"x509SecondaryThumbprint\":null\r\n}";
        var response = (Response) JsonConvert.DeserializeObject(jsonResponse, typeof(Response));

        Debugger.Break();
        Thread.Sleep(Timeout.Infinite);
    }

    internal class Response
    {
        public string id { get; set; }
        public int connectionState { get; set; }
        public int status { get; set; }
        public int authenticationType { get; set; }
        public string symmetricPrimaryKey { get; set; }
        public string symmetricSecondaryKey { get; set; }
        public string x509PrimaryThumbprint { get; set; }
        public string x509SecondaryThumbprint { get; set; }
    }
}
```

So I'd double check your response string to confirm that it's valid JSON and the types are correct. 

https://jsonlint.com/

networkfusion, 2024-04-26:
Yes, very likely there are issues with `jsonResponse` containing chars at the beginning or the end. Without the actual string it is hard to determine, but it might need to used the `.Trim()` methods.

varogonz95, 2024-04-26:
Thanks for your comments, I'll double-check as soon as I get home 😊

MikaelMogren, 2024-04-26:
If this is related to the problems I'm having with reading the serial port. Downgrading firmware to 1.9.1.7 might help.

josesimoes, 2024-04-26:
@MikaelMogren should you be casting the deserialization to `RegisteredDeviceResponse` instead of `Response`?

varogonz95, 2024-04-26:
Update:
Turns out the json response was having an encoded character. The content for `SymmetricSecondayKey` is `lotsOfNumbersAndLetters+lotsOfNumbersAndLetters=` but the response is encoding the `+` char to `\u002B`.
I tried hitting the endpoint with Postman and the response is correct, so Im not sure if that's a bug from `JsonConvert`.
In the meantime, I'll come up with a workaround to replace any encoded chars

varogonz95, 2024-04-26:
> @MikaelMogren should you be casting the deserialization to `RegisteredDeviceResponse` instead of `Response`?

`Response` is `RegisteredDeviceResponse`, I just didnt want to expose the class name. Woops 😂

torbacz, 2024-04-26:
The JSON serialization is limited in nanoFramework. We are trying to keep the implementation to minimum required functionalities - just to find sweet spot between optimalization, deployment size and functionalities. 
Nevertheless, there are few points of customization, where you can implement your own code:
- MemberResolver - should help you with property naming, can be changed here https://github.com/nanoframework/nanoFramework.Json/blob/main/nanoFramework.Json/Configuration/Settings.cs
- Converters - that one might help you with custom types handling, can be changed here https://github.com/nanoframework/nanoFramework.Json/blob/main/nanoFramework.Json/Configuration/ConvertersMapping.cs

varogonz95, 2024-04-27:
I'm closing this as I found a workaround in my service.
I'm serializing the response with the `UnsafeRelaxedJsonEscaping` encoder for JsonSerializerOptions. This way the payload will be sent as-is, without sanitization and my ESP32 will be happy not dealing with encoding hassle.
Thanks all for your comments!

closed 2024-04-27 by varogonz95