nanoframework/Home #1480: Serial port broken on XIAO ESP32C3 after upgrade from 1.9.1.7

MikaelMogren, 2024-04-25:
### Target name(s)

XIAO_ESP32C3

### Firmware version

1.9.1.21

### Was working before? On which version?

1.9.1.7

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32_C3 built with ESP-IDF 38eeba2
  Target:   XIAO_ESP32C3
  Platform: ESP32

Firmware build Info:
  Date:        Mar 23 2024
  Type:        MinSizeRel build, chip rev. >= 3, without support for PSRAM
  CLR Version: 1.9.1.21
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:

Native Assemblies:
  mscorlib v100.5.0.19, checksum 0x445C7AF9
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.10.0, checksum 0x6A20A689
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.System.IO.Hashing v100.0.0.1, checksum 0xEBD8ED20
  nanoFramework.System.Security.Cryptography v100.0.0.2, checksum 0xF4AEFE6C
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.2, checksum 0x545A6C79
  System.Math v100.0.5.5, checksum 0x9F9E2A7E
  System.Net v100.2.0.1, checksum 0xD82C1452
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.2, checksum 0xFA806D33
  System.Device.I2c.Slave v1.0.0.0, checksum 0x4238164B
  System.Device.I2s v100.0.0.1, checksum 0x478490FE
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Runtime.Serialization v100.0.0.0, checksum 0x0A066871
  System.Device.Wifi v100.0.6.4, checksum 0x00A058C6
  Windows.Storage v100.0.3.0, checksum 0xF0C37E1B

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
[... 87 lines cut ...]
    }

    public class P1SerialPort
    {
        public SerialPort Create()
        {
            // XIAO_ESP32C3
            Configuration.SetPinFunction(20, DeviceFunction.COM2_RX);
            Configuration.SetPinFunction(21, DeviceFunction.COM2_TX);

            return new SerialPort("COM2")
            {
                StopBits = StopBits.One,
                BaudRate = 115200,
                DataBits = 8,
                Parity = Parity.None,
                ReadBufferSize = 900,
                InvertSignalLevels = true,
                ReceivedBytesThreshold = 500,
                Handshake = Handshake.None
            };
        }
    }
}
`

josesimoes, 2024-04-26:
Possibly related with nanoframework/nf-interpreter#2906

MikaelMogren, 2024-04-30:
Seeing a lot of new line handling in pull request #2906. 
Maybe worth mentioning. I'm receiving CR+LF as line feed terminator. Hex 0D-0A.

josesimoes, 2024-06-19:
@mikmog care to test again. Serial port should be working fine now.

AdrianSoundy, 2024-06-19:
The problem here is the InnverseSignalsLevels is not being actioned.
The native driver sets the inverse then clears setting when port is opened.

mikmog, 2024-06-20:
Midsummer eve festivities coming up here.  I’ll give it a try anyway in the next couple of days

mikmog, 2024-06-22:
Sorry. The problem still exist in 1.9.1.265. Based on Adrian's comment I also tried setting inverse signals after the port was opened. But that was not allowed 😕

------------------------------------------------------
`System Information
HAL build info: nanoCLR running @ ESP32_C3 built with ESP-IDF e7771c7
  Target:   XIAO_ESP32C3
  Platform: ESP32

Firmware build Info:
  Date:        Jun 22 2024
  Type:        MinSizeRel build, chip rev. >= 3, without support for PSRAM
  CLR Version: 1.9.1.265
  Compiler:    GNU ARM GCC v12.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:

Native Assemblies:
  mscorlib v100.5.0.19, checksum 0x445C7AF9
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.10.0, checksum 0x6A20A689
  nanoFramework.Hardware.Esp32.Rmt v100.0.4.0, checksum 0x608C5658
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.System.IO.Hashing v100.0.0.1, checksum 0xEBD8ED20
  nanoFramework.System.Security.Cryptography v100.0.0.2, checksum 0xF4AEFE6C
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.1.0.0, checksum 0xCC556D24
  System.Math v100.0.5.5, checksum 0x9F9E2A7E
  System.Net v100.2.0.1, checksum 0xD82C1452
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.2, checksum 0xFA806D33
  System.Device.I2c.Slave v1.0.0.0, checksum 0x4238164B
  System.Device.I2s v100.0.0.1, checksum 0x478490FE
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Runtime.Serialization v100.0.0.0, checksum 0x0A066871
  System.Device.Wifi v100.0.6.4, checksum 0x00A058C6
  Windows.Storage v100.0.3.0, checksum 0xF0C37E1B

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3fcc036c  0x0001c000
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x1F0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x1F0000 (1984kB)   Deployment`

AdrianSoundy, 2024-06-22:
Thanks for the feedback. I have looked at version 1.9.1.21 and can't see how it would have changed the behavior.
I'll do some further testing and get back to you.

mikmog, 2024-06-22:
Probably way off. But newer firmware might require a different value on the pull up. I'll do some tinkering and see if anything changes

mikmog, 2024-06-22:
The value of the pull up or which pin (3.3/5v) is not the cause.

From my point of view as a consumer at C# -level it feels like the assignment of `InvertSignalLevels = true` is not respected

Just to be sure, I tripled check and 
1.9.1.7 => works,
1.9.1.21 => gibberish

closed 2024-08-04 by AdrianSoundy