nanoframework/Home #1482: Problems with 96x16 OLED display

1kvin, 2024-04-27:
### Target name(s)

ESP32_C3_REV3

### Firmware version

ESP32_C3_REV3-1.9.1.75

### Description

I'm seeing problems with the display. The standard example does not work, I decided to try to figure out the problem and used a modified initialization from [here](https://github.com/greiman/SSD1306Ascii/blob/master/src/SSD1306init.h#L156C31-L156C48).

Here is the solution that works for me:

```
using var device = new Ssd1306(
      I2cDevice.Create(
        new I2cConnectionSettings(
            1,
            Ssd1306.DefaultI2cAddress,
            I2cBusSpeed.FastMode)),
    Ssd13xx.DisplayResolution.OLED96x16
    );

var init = new byte[]
   {
    0x00,       // is command
    0xae,       // turn display off
    0xd5, 0x80, // set display clock divide ratio/oscillator,  set ratio = 0x80
    0xa8, 0x0F, // set multiplex ratio 0x00-0x1f        
    0xd3, 0x00, // set display offset 0x00-0x3f, no offset = 0x00
    0x40, // set display start line 0x40-0x7F
    0x8d, 0x14, // set charge pump,  enable  = 0x14  disable = 0x10
    0x20, 0x00, // 0x20 set memory address mode,  0x0 = horizontal addressing mode
    //0xa0 | 0xA1, // set segment re-map
    0xc8,       // set com output scan direction
    0xda, 0x02, // set COM pins HW configuration
    0x81, 0xaf, // set contrast control for BANK0, extVcc = 0x9F,  internal = 0xcf
    0xd9, 0xf1, // set pre-charge period  to 0xf1,  if extVcc then set to 0x22
    0xdb,       // set VCOMH deselect level
    0x2e,       // Deactivate scroll
    0xa4,       // set display ON
    0xa6,       // set normal display
    0xaf        // turn display on 0xaf
   };

device.SendData(new SpanByte(init));
device.SendCommand(new SetSegmentReMap(true));
```
Maybe it's a copy paste victim

torbacz, 2024-06-18:
I can confirm that there is a problem. The initialization code from above doesn't work for me (also the OOTB solution doesn't work)

But, now when nanoFramework got graphics drivers, maybe it's time to mark this one as "obsolete" and stick only with one implementation?

@josesimoes Any thoughts on this?

josesimoes, 2024-06-18:
Maybe we should "collect" the various version of the init code that are being used out there.
I don't think there will be a one size fits all...

torbacz, 2024-06-20:
I just tested the same OLED display with generic config (I'm using OLED128x64)- now it works. Probably missed something last time. 

Anyway, I compared both configs from 96x16 version. There are just a few differences. I assume we can ignore the scroll setting. But there are two more. Which one is more generic? 
[image]

Ellerbach, 2024-06-20:
Each register has flags. So, it's "just" about looking at the documentation to understand which flags are used. Also, some copies may have added extra flags or bugs and require to set some flags that are ignored on genuine versions. I would not bother too much about all this.

networkfusion, 2024-12-16:
@1kvin is this still an issue, or was it solved with the above comments?

1kvin, 2024-12-23:
still works poorly on 96x16 and good on 128x64
I don't know why  :/