nanoframework/Home #1485: `DateTime.Parse` does not handle string that conform to RFC 3339.

CoryCharlton, 2024-05-15:
### Library/API/IoT binding

nanoFramework.CoreLibrary

### Description

[RFC 3339](https://datatracker.ietf.org/doc/html/rfc3339) does not specify the number of digits for the fractional second portion but a common practice is to use three digits so the fractional second portion relates to whole milliseconds:

`yyyy-MM-dd'T'HH:mm:ss.fffK` (or `yyyy-MM-dd'T'HH:mm:ss.fffZ` in NF since 'K' isn't supported and all `DateTime` are assumed to be UTC)

Ex: `2024-05-15T04:20:05.069Z`

Unfortunately the `DateTime.Parse` implementation in NF is currently only supporting the ISO 8601 "round trip time" format which requires that the fractional second portion contain 7 digits (`yyyy-MM-ddTHH:mm:ss.fffffff`) [code](https://github.com/nanoframework/nf-interpreter/blob/36dbc2d60350f63f759a35e92f7b1e2538d72c21/src/CLR/CorLib/corlib_native_System_Convert.cpp#L549).

This causes the example string above to parse with '0' milliseconds and '69' "extra" ticks. The issue is [here](https://github.com/nanoframework/nf-interpreter/blob/36dbc2d60350f63f759a35e92f7b1e2538d72c21/src/CLR/CorLib/corlib_native_System_Convert.cpp#L1077) because the code assumes the string represents a 7 digit number '0000069' when the correct 7 digit number is '0690000'

### How to reproduce

See sample code below.

### Sample project or code

```C#
var dateTime1 = new DateTime(2024, 5, 15, 4, 20, 5, 69);
var dateTimeString = dateTime1.ToString("yyyy-MM-dd'T'HH:mm:ss.fffZ");
var dateTime2 = DateTime.Parse("2024-05-15T04:20:05.069Z");
var dateTime3 = DateTime.Parse("2024-05-15T04:20:05.0690000Z");

Console.WriteLine($"dateTime1: {dateTime1.Ticks} - {dateTime1.Hour}:{dateTime1.Minute}:{dateTime1.Second}.{dateTime1.Millisecond}");
Console.WriteLine($"dateTime2: {dateTime2.Ticks} - {dateTime2.Hour}:{dateTime2.Minute}:{dateTime2.Second}.{dateTime2.Millisecond}");
Console.WriteLine($"dateTime3: {dateTime3.Ticks} - {dateTime3.Hour}:{dateTime3.Minute}:{dateTime3.Second}.{dateTime3.Millisecond}");
Console.WriteLine($"dateTimeString: {dateTimeString}");
```

Output:

```
dateTime1: 638513436050690000 - 4:20:5.69
dateTime2: 638513436050000069 - 4:20:5.0
dateTime3: 638513436050690000 - 4:20:5.69
dateTimeString: 2024-05-15T04:20:05.069Z
```

### Aditional information

I looked at the code a bit and I don't have a good solution at the moment.

networkfusion, 2024-05-16:
My common workaround is `.ToString("o")`

CoryCharlton, 2024-05-16:
> My common workaround is `.ToString("o")`
> 
> e.g. you can use `Debug.WriteLine($"as ISO date: {DateTime.FromUnixTimeSeconds(shadow.timestamp).ToString("o")}");`

My issue is that the strings I need to parse are coming from an API I don't have control over.

My current workaround is some IndexOf/Substring hacking to find the fractional seconds part and pad/trim it to 7 digits.

josesimoes, 2024-11-18:
Wondering if there was any progress on this... 😉