nanoframework/Home #1486: I2cDevice is throwing an ArgumentException for busId=2

romankiss, 2024-05-17:
### Target name(s)

ESP32_PICO, ESP32_S3

### Firmware version

1.9.1.123+, 1.9.1.90+

### Was working before? On which version?

1.9.1.93, 1.9.1.69

### Description

  The following line thrown an exception when the **busId = 2**:

     i2c_device = new I2cDevice(new I2cConnectionSettings(2, M5AtomicMotion.DefaultI2cAddress));

    ++++ Exception System.ArgumentException - 0xfd000000 (1) ++++
    ++++ Message: 
    ++++ System.Device.I2c.I2cDevice::NativeInit [IP: 0000] ++++
    ++++ System.Device.I2c.I2cDevice::.ctor [IP: 001c] ++++
    ++++ NFAppAtomLite_Testing.Program::Main [IP: 0041] ++++
    Exception thrown: 'System.ArgumentException' in System.Device.I2c.dll
    An unhandled exception of type 'System.ArgumentException' occurred in System.Device.I2c.dll

Note, that using a busId=1 is working well.

closed 2024-05-17 by AdrianSoundy

romankiss, 2024-05-17:
@josesimoes and @AdrianSoundy Thanks, I will test it tomorrow.

romankiss, 2024-05-18:
@josesimoes and @AdrianSoundy , I have tried to test it with the following result:

- All assemblies are up to date.

- **ESP32_PICO (1.9.1.147)** I2cDevice thrown ArgumentException for busId=2

- **ESP32_PICO (1.9.1.93)** The last working version.

- **ESP32_S3 (1.9.1.141, 138, 136)** Deployment Failed
     1>Deploying 35 assemblies to device... Total size in bytes is 360036.
     1>Deploy failed.
      ========== Deploy: 0 succeeded, 1 failed, 0 skipped ==========
      ========== Deploy completed at 2:44 PM and took 13.665 seconds ==========

- **ESP32_S3 (1.9.1.125, 116, 113, 109, 106, 103, 90)** I2cDevice thrown ArgumentException for busId=2

- **ESP32_S3 (1.9.1.69)** The last working version.

It looks like, that the overnight build was not published yet, also why the deploy failed for ESP32_S3 target with the latest firmware versions such as 1.9.1.141, 138 and 136?

AdrianSoundy, 2024-05-18:
@romankiss 
I don't think the pico firmware has been updated yet.  Will need to look at overnight builds.
I don't know why you are having deployment errors.  Have there been any changes in that area @josesimoes

alberk8, 2024-05-19:
@josesimoes ,  I have tested a simple program and it failed to deploy on ESP32 S3 starting with firmware 1.9.1.136.   Could it be y related to some changes of the [WireProtocol](https://github.com/nanoframework/nf-interpreter/commit/9b774ebd5207266ebde6ff19f9a61775af8236b4).

```
09:00:01.034 [Deploying 8128/49860 bytes.]
09:00:01.066 [Deploying 9144/49860 bytes.]
09:00:06.527 [Error writing 1016 bytes to device @ 0x001B1FC0. No reply from nanoDevice.]
09:00:06.527 [*** ERROR deploying assemblies to the device ***]
09:00:06.527 [*** ERROR: deployment failed ***]
```

romankiss, 2024-05-25:
I am confirming that my devices have been successfully tested for this issue with the following info:

- all nuget assemblies are up to date
- targets and firmware version: 
   - ESP32_PICO (1.9.1.162)
   - M5StickC (1.9.1.162)
   - ESP32_S3 (1.9.1.151)
   - ESP32_S3_BLE (1.9.1.162)
   - ESP32_S3_ALL (1.9.1.151)
   - AtomS3 (1.9.1.162)