nanoframework/Home #1487: System.Net.Sockets.SocketException at the target ESP32_S3_ALL

romankiss, 2024-05-25:
### Target name(s)

ESP32_S3_ALL

### Firmware version

1.9.1.151

### Was working before? On which version?

no

### Description

I have an IoT Application connected to the Azure IoT Central with all up-to-date nuget assemblies runs on the following targets:

- ESP32_S3 (1.9.1.151) - working very well
- ESP32_S3_BLE (1.9.1.162) - working very well
- **ESP32_S3_ALL (1.9.1.151)** - throwing the following exception:

++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (1) ++++
++++ Message: 
++++ System.Net.Security.SslNative::SecureClientInit [IP: 0000] ++++
++++ System.Net.Security.SslStream::Authenticate [IP: 0045] ++++
++++ System.Net.Security.SslStream::AuthenticateAsClient [IP: 000a] ++++
++++ nanoFramework.M2Mqtt.MqttNetworkChannel::Connect [IP: 0063] ++++
++++ nanoFramework.Azure.Devices.Client.DeviceClient::Open [IP: 00b6] ++++

Note, there is no graphics and BLE usage in the application, so it should be all three targets worked the same without this exception.

josesimoes, 2024-05-27:
Most likely not enough memory to deal with the TLS stuff. @AdrianSoundy is to look into improving the memory allocation in ESP32, so this should be working again after that.

AdrianSoundy, 2024-05-27:
What ESP32 board are you running this on? The ESP32_S3_ALL firmware is really for S3 with 8Mb spiram running in octal mode. If not capatible spiram then all allocation would be in ESP32 ram which is limited. The ESP2_S3 and ESP32_S3_BLE support quad mode spiram. Its not possible to have firmware with quad & octal mod spiram support.

romankiss, 2024-05-27:
@AdrianSoundy I have the following products built-in a screen with running a factory demo program (Arduino/MicroPython/ESP-IDF):

- [Waveshare ESP32-S3-GEEK](https://www.waveshare.com/esp32-s3-geek.htm) failed with a firmware ESP32_S3_ALL but is running very well without the screen using the targets ESP32_S3, ESP32_S3_BLE

- M5Stack products such as:
    - [AtomS3](https://docs.m5stack.com/en/core/AtomS3) - failed for ESP32_S3_ALL, ESP32_S3 and ESP32_S3_BLE is working
    - [M5Dial ](https://docs.m5stack.com/en/core/M5Dial)- failed for ESP32_S3_ALL, ESP32_S3 and ESP32_S3_BLE is working
    - [M5DimMeter ](https://docs.m5stack.com/en/core/M5DinMeter)- failed for ESP32_S3_ALL, ESP32_S3 and ESP32_S3_BLE is working

AdrianSoundy, 2024-05-27:
Yes all your devices use quad mode spiram so that's why they don't have enough memory when running ESP32_S3_ALL.
Is the issue that you want to use display as we can add the generic display driver to the other S3 firmware?

romankiss, 2024-05-28:
@AdrianSoundy thanks for explanation. Yes, we need a new firmware for QUAD mode **ESP32_S3_All** for these kinds of products, where are usage of screen (using a generic display driver) and BLE.

_Btw1._ for non-confusing naming of the firmware is better to keep names as following: for QUAD mode: ESP32_S3, ESP32_S3_BLE, ESP32_S3_ALL and for OCTAL mode: ESP32_S3_ALL_OCTAL 

_Btw2._ I have also a product [Unit-CamS3](https://docs.m5stack.com/en/unit/Unit-CamS3), so based on your info, the current firmware ESP32_S3_ALL should be work. I am going to test it soon.

Thank you so much.

romankiss, 2024-05-29:
@AdrianSoundy  I have just successfully finished my test with an IoT App connected to the Azure IoT Central using a target **ESP32_S3_ALL (1.9.1.151)** on the following products (8MB PSRAM):
   - [Unit-CamS3](https://docs.m5stack.com/en/unit/Unit-CamS3) 
   - [XIAO ESP32S3](https://www.seeedstudio.com/XIAO-ESP32S3-p-5627.html)
   - [M5Stack CoreS3](https://shop.m5stack.com/products/m5stack-cores3-esp32s3-lotdevelopment-kit)

Based on that, there is missing a S3 firmware image (included the AtomS3) for QUAD mode with a generic display driver, so I would like to ask, if we can have it ether a new image like we have for BLE or adding the generic display driver to the current one such as ESP32_S3.

Thank you so much for your help and support.

AdrianSoundy, 2024-05-30:
I'll shortly add the generic display driver to the ESP32_S3 and ESP32_S3_BLE firmware.
Probably need to review and optimize the naming of all firmware. But we can do that later.

romankiss, 2024-05-30:
@AdrianSoundy Thanks and I am looking forward for testing those ones with my IoT App.

AdrianSoundy, 2024-06-02:
Decided to add generic driver to just ESP32_S3_BLE and leave ESP32_S3 as minimum firmware

closed 2024-06-02 by AdrianSoundy

romankiss, 2024-06-02:
@AdrianSoundy  The following is a result of my testing on the S3:

- .NET nanoFramework Firmware Flasher v2.5.77+18c31baab4
- all nuget assemblies are up to date

- [M5StampS3 ](https://shop.m5stack.com/products/m5stamps3-with-1-27-header-pin), firmware ESP32_S3 (1.9.1.166), IoT Application (no scree, no BLE) thrown an SocketException at the IoT DPS process.

- [AtomS3](https://docs.m5stack.com/en/core/AtomS3) , firmware ESP32_S3_ALL (1.9.1.177) IoT App with a built-in screen (no BLE) is working well. Total Mem 297608, nF Mem 37452, I will test it again with a new overnight build ESP32_S3_BLE

- [Waveshare ESP32-S3-GEEK](https://www.waveshare.com/esp32-s3-geek.htm) - 2MB PSRAM, ESP32_S3_ALL (1.9.1.177) IoT App with a built-in screen (no BLE) is working well. Total Mem 297608, I will test it again with a new overnight build ESP32_S3_BLE

- [Unit-CamS3](https://docs.m5stack.com/en/unit/Unit-CamS3) - 8MB PSRAM, ESP32_S3_ALL (1.9.1.177) IoT App (no screen, no BLE) is working well. Total Mem 8718983, nF Mem 7115544.

- [Waveshare ESP32-S3-Zero ](https://www.waveshare.com/esp32-s3-zero.htm) - 2MB PSRAM, firmware ESP32_S3 (1.9.1.166) failed to connect with my MS VS2022, but when I use the older flasher such as version 2.5.67 everything is working well, Total Mem 2463447, nF Mem 1726296.

So, it looks like the target ESP32_S3 (1.9.1.166) has an issue with a firmware flasher and some board like M5StampS3 doesn't have enough an IDF configured memory when there is no additional PSRAM.