nanoframework/Home #1488: ESP32 S3 - Wifi AP not working anymore

Alex-111, 2024-05-29:
### Target name(s)

ESP32-S3-DevkitC1

### Firmware version

ESP32_S3-1.9.1.151.zip or ESP32_S3_BLE-1.9.1.162.zip

### Was working before? On which version?

It worked before with a version some months ago - do not remember the version

### Description

The sample application WifiAP makes the ESP32 not working anymore.

It seems that something goes wrong when the AP configuration is saved

 wapconf.SaveConfiguration();

After this the chip restarts but after the restart no connection via Visual Studio is possible anymore. Regardless what I try: Reboot, Reset, Power Off/On.....

On the raw serial output console it seems that the bootstraping is like before, but there is no single output from my own code anymore, even if the first line of my code is a Debug.Writeline or Console.Writeline, i.e. it seems no C# code is running at all. But also no error messages are appearing in the raw serial console...

I only can make the device work again if I flash the device with masserase flag set!

### How to reproduce

Run WifiAP sample
After AP configuration is saved by the SetWifiAp function the ESP32 does not work anymore

jazzman55, 2024-05-31:
On ESP32_REV3 it works on 1.9.1.93 and doesn't work since 1.9.1.123

AdrianSoundy, 2024-06-02:
Fixed on latest firmware 1.9.1.161

Alex-111, 2024-06-02:
Dows not work with ESP32_PICO 1.9.1.162. Will test with ESP32_S3 tomorrow...

AdrianSoundy, 2024-06-03:
The pico build wasn't upto date as it was failing in overnight builds. Fixed now on separate PR so should be available after next builds.

Alex-111, 2024-06-03:
@AdrianSoundy 
I just tested with ESP_S3 firmware 1.9.1.166 and it seems that it is still not fixed to 100%.
WIFIAP starts now and the device is also visible in Visual Studio, but the debugger cannot attach anymore after AP is configured:

Visual Studio logs:
07:01:14.351 [Attempting to connect the debugger engine (1/5)]
07:01:15.894 [Attempting to connect the debugger engine (2/5)]
07:01:17.929 [Attempting to connect the debugger engine (3/5)]
07:01:20.464 [Attempting to connect the debugger engine (4/5)]
07:01:23.490 [Attempting to connect the debugger engine (5/5)]

Please could you have a look at this, too? It sems related to AP configuration. Thanks!

AdrianSoundy, 2024-06-07:
I have isolated the problem and it will be fixed tomorrow. @Alex-111

Alex-111, 2024-06-08:
@AdrianSoundy Wow. I‘m glad that you found the issue.
Can and will test on Monday!

closed 2024-06-08 by AdrianSoundy

Alex-111, 2024-06-10:
Tested on S3 -> It is working now