nanoframework/Home #1489: Enable Bluetooth 5.0 Feature on ESP32 MCU

alberk8, 2024-05-30:
### Description

Enable Bluetooth 5.0 feature like Advanced Advertisement Extension for broadcast and scanning.

### How to solve the problem

Have to add feature in Native as well as manage code.

### Describe alternatives you've considered

none

networkfusion, 2025-03-07:
@alberk8 please can you specify some test targets that would be compatible with this feature.

alberk8, 2025-03-10:
> [@alberk8](https://github.com/alberk8) please can you specify some test targets that would be compatible with this feature.

ESP32 S3, C3 and C6

AdrianSoundy, 2026-09-11:
Add support for Extended Advertisements in to BluetoothLEAdvertisementPublisher via UseExtendedAdvertisement property.
Add support for selecting primary and secondary PHY on devices which support it.

Add support for receiving extended advertisements in BluetoothLEAdvertisementWatcher  and update event args