nanoframework/Home #1491: ESP32_S3_ALL does not recognize PSRAM on ESP32-S3-Zero device

JohnMasen, 2024-06-01:
### Target name(s)

ESP32-S3-ALL

### Firmware version

1.9.1.161

### Was working before? On which version?

N/A

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32_S3 built with ESP-IDF v5.1.3
  Target:   ESP32_S3_ALL
  Platform: ESP32

Firmware build Info:
  Date:        May 31 2024
  Type:        MinSizeRel build, chip rev. >= 0, support for PSRAM, support for BLE
  CLR Version: 1.9.1.161
  Compiler:    GNU ARM GCC v12.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  BasicFeature, 1.0.0.0
  nanoFramework.Hardware.Esp32, 1.6.15.0
  Iot.Device.Ws28xx.Esp32, 1.2.0.0
  nanoFramework.Runtime.Events, 1.11.18.0
  nanoFramework.Hardware.Esp32.Rmt, 2.0.10.0
  nanoFramework.Graphics.Core, 1.2.10.0
  mscorlib, 1.15.6.0

Native Assemblies:
  mscorlib v100.5.0.19, checksum 0x445C7AF9
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.10.0, checksum 0x6A20A689
  nanoFramework.Hardware.Esp32.Rmt v100.0.4.0, checksum 0x608C5658
  nanoFramework.Device.Bluetooth v100.0.5.0, checksum 0xD163A9D5
  nanoFramework.Graphics v100.0.0.10, checksum 0x6A3F90DD
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.System.IO.Hashing v100.0.0.1, checksum 0xEBD8ED20
  nanoFramework.System.Security.Cryptography v100.0.0.2, checksum 0xF4AEFE6C
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.1.0.0, checksum 0xCC556D24
  System.Math v100.0.5.5, checksum 0x9F9E2A7E
  System.Net v100.2.0.1, checksum 0xD82C1452
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.2, checksum 0xFA806D33
  System.Device.I2c.Slave v1.0.0.0, checksum 0x4238164B
  System.Device.I2s v100.0.0.1, checksum 0x478490FE
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Runtime.Serialization v100.0.0.0, checksum 0x0A066871
  System.Device.Wifi v100.0.6.4, checksum 0x00A058C6
  Windows.Storage v100.0.3.0, checksum 0xF0C37E1B

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3fcbbe88  0x0001bff8
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x1F0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x1F0000 (1984kB)   Deployment

Deployment Map
Empty

### Description

[ESP32-S3-Zero](https://www.waveshare.com/wiki/ESP32-S3-Zero) has 2MB Octal PSRAM, however ESP32-S3-ALL could not recognize them. 
```NativeMemory``` reports _297,608_ bytes total.

nanoff tool seems does not recognize the PSRAM( PSRAM=undetermined)
```
.NET nanoFramework Firmware Flasher v2.5.77+18c31baab4
Copyright (C) 2019 .NET Foundation and nanoFramework project contributors

Reading details from chip...OK

Connected to:
ESP32-S3 (ESP32-S3 (QFN56) (revision v0.2))
Features WiFi, BLE, Embedded Flash 4MB (XMC), Embedded PSRAM 2MB (AP_3v3)
Flash size 4MB unknown from ST (manufacturer 0x32 device 0x16406)
PSRAM: undetermined
Crystal 40MHz
MAC 34:B7:DA:5A:E6:E0

Extracting ESP32_S3_ALL-1.9.1.161.zip...OK

Updating to 1.9.1.161

Erasing flash...OK
Flashing firmware...OK
```

### How to reproduce

Flash the device with following command.
```console
nanoff --update --serialport COM12 --target ESP32_S3_ALL --masserase
```
Output
```
.NET nanoFramework Firmware Flasher v2.5.77+18c31baab4
Copyright (C) 2019 .NET Foundation and nanoFramework project contributors

Reading details from chip...OK

Connected to:
ESP32-S3 (ESP32-S3 (QFN56) (revision v0.2))
Features WiFi, BLE, Embedded Flash 4MB (XMC), Embedded PSRAM 2MB (AP_3v3)
Flash size 4MB unknown from ST (manufacturer 0x32 device 0x16406)
PSRAM: undetermined
Crystal 40MHz
MAC 34:B7:DA:5A:E6:E0

Extracting ESP32_S3_ALL-1.9.1.161.zip...OK

Updating to 1.9.1.161

Erasing flash...OK
Flashing firmware...OK

```

### Expected behaviour

Device able to boot with PSRAM

### Aditional information

**Boot log:**
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x29 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037a2d2
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1380
load:0x403c9700,len:0x4
load:0x403c9704,len:0xba4
load:0x403cc700,len:0x2c5c
SHA-256 comparison failed:
Calculated: a304a46714bc47e8fe394f0366d16ac41f04391c112aa7658818d34b71c8c6a6
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98f0
I (49) boot: .NET nanoFramework 2nd stage bootloader ESP-IDF v5.1.3
I (50) boot: build May 31 2024 13:52:02
I (50) boot: chip revision: v0.2
I (54) boot.esp32s3: Boot SPI Speed : 80MHz
I (58) boot.esp32s3: SPI Mode       : DIO
I (63) boot.esp32s3: SPI Flash Size : 4MB
I (68) boot: Enabling RNG early entropy source...
I (73) boot: Partition Table:
I (77) boot: ## Label            Usage          Type ST Offset   Length
I (84) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (92) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (99) boot:  2 factory          factory app      00 00 00010000 001a0000
I (107) boot:  3 deploy           Unknown data     01 84 001b0000 001f0000
I (114) boot:  4 config           Unknown data     01 82 003c0000 00040000
I (122) boot: End of partition table
I (126) esp_image: segment 0: paddr=00010020 vaddr=3c100020 size=2b328h (176936) map
I (154) esp_image: segment 1: paddr=0003b350 vaddr=3fc9d000 size=047b0h ( 18352) load
I (157) esp_image: segment 2: paddr=0003fb08 vaddr=40374000 size=00510h (  1296) load
I (160) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=f9b24h (1022756) map
I (283) esp_image: segment 4: paddr=00139b4c vaddr=40374510 size=18acch (101068) load
I (299) esp_image: segment 5: paddr=00152620 vaddr=600fe000 size=00064h (   100) load
I (309) boot: Loaded app from partition at offset 0x10000
I (310) boot: Disabling RNG early entropy source...
Hello from nanoFramework!
internal memory: total= 297608,free=52320,blob=31744
SPI memory: total= 0,free=0,blob=0

alberk8, 2024-06-01:
@JohnMasen Even the Waveshare documentation is wrong.  The Flash and PSRAM are Quad speed.  [ESP Doc](https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf)

[image]

JohnMasen, 2024-06-01:
@alberk8 Thanks for your info, I've contact their tech support for clarification. will update when I got their reply.
meaning while I tried to flash ESP_S3 to this borad, but it seems triggered another issue, I created #1492 for it.

AdrianSoundy, 2024-06-06:
The ESP32_S3_ALL is for devices using Octal psram. The ESP32_S3_Zero has 2Mb of Quad psram. 
You should use the ESP32_S3 or ESP32_S3_BLE firmware.

The ESP32_S3_BLE is now the same as ESP32_S3_ALL but for quad psram.

See newly updated readme
https://github.com/nanoframework/nf-interpreter#esp32_s3-boards

I will close issue.

closed 2024-06-06 by AdrianSoundy