nanoframework/Home #1493: WiFi AP crashes debugger

vtelenak, 2024-06-03:
### Target name(s)

ESP32_REV0, ESP_PSRAM_REV3

### Firmware version

1.9.1.181

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF e7771c7
  Target:   ESP32_REV0
  Platform: ESP32

Firmware build Info:
  Date:        Jun  3 2024
  Type:        MinSizeRel build, chip rev. >= 0, without support for PSRAM
  CLR Version: 1.9.1.181
  Compiler:    GNU ARM GCC v12.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  NFAppAPTest, 1.0.0.0
  mscorlib, 1.15.6.0
  System.Device.Gpio, 1.1.41.0
  Iot.Device.DhcpServer, 1.2.0.0
  nanoFramework.Runtime.Native, 1.6.12.0
  System.Threading, 1.1.32.63105
  nanoFramework.Runtime.Events, 1.11.18.0
  nanoFramework.System.Text, 1.2.54.0
  System.IO.Streams, 1.1.59.0
  System.Net, 1.10.79.0

Native Assemblies:
  mscorlib v100.5.0.19, checksum 0x445C7AF9
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.10.0, checksum 0x6A20A689
  nanoFramework.Hardware.Esp32.Rmt v100.0.4.0, checksum 0x608C5658
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.System.IO.Hashing v100.0.0.1, checksum 0xEBD8ED20
  nanoFramework.System.Security.Cryptography v100.0.0.2, checksum 0xF4AEFE6C
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.1.0.0, checksum 0xCC556D24
  System.Math v100.0.5.5, checksum 0x9F9E2A7E
  System.Net v100.2.0.1, checksum 0xD82C1452
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.2, checksum 0xFA806D33
  System.Device.I2c.Slave v1.0.0.0, checksum 0x4238164B
  System.Device.I2s v100.0.0.1, checksum 0x478490FE
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Runtime.Serialization v100.0.0.0, checksum 0x0A066871
  System.Device.Wifi v100.0.6.4, checksum 0x00A058C6
  Windows.Storage v100.0.3.0, checksum 0xF0C37E1B

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffe46bc  0x0001b000
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1B0000     nanoCLR
      1    0x001C0000       1      0x1E0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1B0000 (1728kB)   nanoCLR
  0x001C0000    0x1E0000 (1920kB)   Deployment

Deployment Map
Empty

### Description

After setting the WIFI AP and restarting the device, the debugger from Visual studio does not work. After turning off the AP, the debugger works again.

...
21:17:45.635 [Deploying 89408/89928 bytes.]
21:17:45.639 [Deploying 89928/89928 bytes.]
21:17:45.694 [Assemblies successfully deployed to the device]
21:17:45.716 [Launching debugger provider from v2022.3.0.78+8264a57302]
21:17:45.880 [Attempting to connect the debugger engine (1/5)]
21:17:45.880 [Assigning debug engine of target device]
21:17:45.880 [Device running nanoCLR, updating the debug flags...]
21:17:46.492 [Attempting to connect the debugger engine (1/5)]
21:17:46.492 [Device running nanoCLR, updating the debug flags...]
21:17:47.077 [Device is running nanoCLR, requesting a restart and pause of debugger (1/5).]
21:17:48.203 [nanoCLR restarted successfully.]
21:17:48.704 [Attempting to connect the debugger engine (1/5)]
21:17:48.704 [Device running nanoCLR, updating the debug flags...]
21:17:49.288 [Device is running the CLR in initialized state]
21:17:49.288 [The debugger engine is listening to debug events from the nanoDevice target]
21:17:49.879 [Event dispatch failed:Odkaz na objekt není nastaven na instanci objektu.]

### Aditional information

 public class Program {
     public static void Main() {
         string ssid = "TestX";
         string softApIP = "192.168.4.1";

         var gpioController = new GpioController();
         GpioPin pinD2 = gpioController.OpenPin(2, PinMode.Output);
         GpioPin pinD4 = gpioController.OpenPin(4, PinMode.Output);
         pinD2.Write(PinValue.Low);
         pinD4.Write(PinValue.Low);           

         NetworkInterface ni = GetInterface();
         WirelessAPConfiguration wapconf = GetConfiguration();

         //Check if already Enabled
         var wirelessAPConfigOptions = WirelessAPConfiguration.ConfigurationOptions.Enable | WirelessAPConfiguration.ConfigurationOptions.AutoStart;
         if (wapconf.Options != wirelessAPConfigOptions || ni.IPv4Address != softApIP) {

             // Set up IP address for Soft AP
             ni.EnableStaticIPv4(softApIP, "255.255.255.0", softApIP);

             wapconf.Options = wirelessAPConfigOptions;
             wapconf.Ssid = ssid;
             wapconf.MaxConnections = 1;
             wapconf.Authentication = System.Net.NetworkInformation.AuthenticationType.Open;
             wapconf.Password = "";
             // Save the configuration so on restart Access point will be running.
             wapconf.SaveConfiguration();

             // Reboot device to Activate Access Point on restart
             Debug.WriteLine($"Setup Soft AP, Rebooting device");
             pinD2.Write(PinValue.High);
             Power.RebootDevice();
         }

                   
         pinD4.Write(PinValue.High);

         Thread.Sleep(Timeout.Infinite);
     }

     public static NetworkInterface GetInterface() {
         NetworkInterface[] Interfaces = NetworkInterface.GetAllNetworkInterfaces();

         // Find WirelessAP interface
         foreach (NetworkInterface ni in Interfaces) {
             if (ni.NetworkInterfaceType == NetworkInterfaceType.WirelessAP) {
                 return ni;
             }
         }
         return null;
     }

     public static WirelessAPConfiguration GetConfiguration() {
         NetworkInterface ni = GetInterface();
         return WirelessAPConfiguration.GetAllWirelessAPConfigurations()[ni.SpecificConfigId];
     }
 }

networkfusion, 2024-06-03:
https://github.com/nanoframework/Home/issues/1335

Alex-111, 2024-06-05:
Can confirm this also with ESP32 PICO and S3.

Alex-111, 2024-06-10:
It is working now - tested with S3