nanoframework/Home #1498: Visual Studio 2022 Device Explorer does not find the device.

grallbring, 2024-06-08:
### Tool

Visual Studio extension

### Description

Hello, I have a `esp32-s3-touch-lcd-1.28` (https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-1.28) and I'm trying to develop something for it using the nanoFramework.
When trying to flash using `nanoff` (command: `nanoff --update --serialport COM4 --baud 115200 --verbosity diag`) the progress simply gets stuck at the `Flashing firmware...` step and nothing happenes (waited around 10 minutes).
When adding `--masserase` the flashing works with a simple green `Ok` after `Flashing firmware...` and I can also connect to it with putty and see the various booting messages.

Still, the device is neither found by the Device Explorer window in Visual Studio 2022 nor by the `nanoff --listdevices` command.

### Expected behaviour

The device should show up in the Device Explorer window and when using the `nanoff --listdevices` command.

### Aditional context

Extension Version: 2022.3.0.78
Visual Studio 2022 Version: 17.10.1
Visual Studio Output:
```
23:28:37.406 [NanoDevices: dropping device in exclusion list COM1]
23:28:37.406 [NanoDevices: new device arrival COM4]
23:28:37.416 [NanoDevices: candidate nano device COM4]
23:28:37.431 [NanoDevices: checking device  COM4 @ 921600]
23:28:38.445 [NanoDevices: checking device  COM4 @ 460800]
23:28:39.451 [NanoDevices: checking device  COM4 @ 115200]
23:28:42.438 [NanoDevices: checking device  COM4 *** 2nd attempt ***]
23:28:42.439 [NanoDevices: checking device  COM4 @ 921600]
23:28:44.445 [NanoDevices: checking device  COM4 @ 460800]
23:28:46.451 [NanoDevices: checking device  COM4 @ 115200]
23:28:48.975 [NanoDevices: quitting device COM4]
23:28:48.976 [NanoDevices: Serial device enumeration completed. Found 0 devices]
```

networkfusion, 2024-06-09:
Try `nanoff --platform esp32 --serialport COM4 --update`

AdrianSoundy, 2024-06-09:
The command @networkfusion menioned will upload the generic _ESP32_S3_ firmware
There is another which may be better which includes a generic display driver. Although it doesn't currently support your display type it should be able to with a lttle work looking at specs of device.

Use comand for this firmware
nanoff -target ESP32_S3_BLE --serialport COM4 --update

On first load after flashing the S3 can take upto 30 secs before its visible to Visual studio.

grallbring, 2024-06-09:
Thanks for your answers, I've tried `nanoff --platform esp32 --serialport COM4 --update` - this simply gets stuck at the `Flashing firmware...` message and nothing further happens (no additional output is shown when adding `--verbosity diag`).
I've also tried the command with `--target ESP32_S3_BLE` and `--target ESP32_S3_ALL`. Both only flash when adding the `--masserase` flag.
I also tried using `esptool` directly and erasing the flash before running the commands but it doesn't change anything.
In conclusion, the flashing of the firmware only seems to work when using the `--masserase` flag, but even then the device isn't detected by either `nanoff --listdevices` or by the Device Explorer in Visual Studio.

AdrianSoundy, 2024-06-09:
What device board are you using ?

grallbring, 2024-06-09:
As mentioned in the first post a `esp32-s3-touch-lcd-1.28` (link: https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-1.28).

AdrianSoundy, 2024-06-09:
I just looked at the device and I see the problem. The usb port is connected via a usb to serial chip to the uart on the ESP32_S3.   Our firmware expects the connection to use the direct USB connection.   A new firmware would need to be built to support this device.

Are you able to build your own firmware ?
See https://docs.nanoframework.net/content/building/using-dev-container.html

You would need to make a copy of the ESP32_S3_BLE config and turn off the option "ESP32_USB_CDC"