nanoframework/Home #1506: HttpClient / HttpWebRequest cannot download more than 2372B of file

vtelenak, 2024-06-17:
### Target name(s)

ESP32_REV0, ESP32_REV3

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF e7771c7
  Target:   ESP32_REV0
  Platform: ESP32

Firmware build Info:
  Date:        Jun 17 2024
  Type:        MinSizeRel build, chip rev. >= 0, without support for PSRAM
  CLR Version: 1.9.1.243
  Compiler:    GNU ARM GCC v12.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  NFAppAPTest, 1.0.0.0
  mscorlib, 1.15.6.0
  System.IO.FileSystem, 1.1.47.0
  nanoFramework.System.Collections, 1.5.31.0
  System.Device.Gpio, 1.1.41.0
  System.Net.Http, 1.5.138.0
  nanoFramework.Logging, 1.1.100.0
  nanoFramework.Runtime.Native, 1.6.12.0
  System.Threading, 1.1.32.63105
  nanoFramework.System.Runtime, 1.0.27.0
  nanoFramework.Runtime.Events, 1.11.18.0
  nanoFramework.System.Text, 1.2.54.0
  System.IO.Streams, 1.1.59.0
  System.Net, 1.10.79.0

Native Assemblies:
  mscorlib v100.5.0.19, checksum 0x445C7AF9
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.10.0, checksum 0x6A20A689
  nanoFramework.Hardware.Esp32.Rmt v100.0.4.0, checksum 0x608C5658
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.System.IO.Hashing v100.0.0.1, checksum 0xEBD8ED20
  nanoFramework.System.Security.Cryptography v100.0.0.2, checksum 0xF4AEFE6C
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.1.0.0, checksum 0xCC556D24
  System.Math v100.0.5.5, checksum 0x9F9E2A7E
  System.Net v100.2.0.1, checksum 0xD82C1452
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.2, checksum 0xFA806D33
  System.Device.I2c.Slave v1.0.0.0, checksum 0x4238164B
  System.Device.I2s v100.0.0.1, checksum 0x478490FE
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
[... 104 lines cut ...]
                do {
                    bytesRead = stream.Read(buffer, 0, buffer.Length);

                    if (bytesRead > 0) {
                        totalBytesRead += bytesRead;
                        //Debug.Write(Encoding.UTF8.GetString(buffer, 0, bytesRead));

                        fs.Write(buffer, 0, bytesRead);
                    }

                    Debug.WriteLine("bytes read " + bytesRead);
                }
                while (totalBytesRead < httpWebResponse.ContentLength);

                fs.Flush();
                fs.Close();
            }

            Debug.WriteLine(">>>>>>>>>>>>>");
            Debug.WriteLine("End of Http response");
            Debug.WriteLine($"Read {totalBytesRead} bytes");
        }
    }

}

Ellerbach, 2024-06-18:
You are in a limited environment, constraints with CPU and memory. You are most likely hitting the memory allocation for those kind of tasks. If you need to read a very large file, then read by chunks like you are doing but write by chunks as well, don't feel up the memory like you are currently doing.

vtelenak, 2024-06-19:
And how do I perceive memory now? It is read into a block that is 1KB and it is immediately stored on the Flash memory. First, it is an approach published in nanoframework samples. I need to do an OTA update and I am based on the article https://www.nanoframework.net/over-the-air-net-nanoframework-code-update-using-azure-iot/ I understand that I cannot work with large files, but over 2KB surely pe file with update will have.

Ellerbach, 2024-06-20:
Adjust how you read the file, download chunk of the file and save it, download the next part. Exactly as you would do when you have a long list and you break it into pages.

closed 2024-12-16 by networkfusion