nanoframework/Home #1509: Nanoff - flash ESP32-S3 hangs

torbacz, 2024-06-19:
### Tool

nanoff

### Description

When using command:
`nanoff --update --target ESP32_S3_ALL --serialport COM18 -v diag`

Nanoff is unable to flash device (or it take long time? I've waited only for 3 minutes)

Command with "masserase" works without problem (took like 30s)
`nanoff --masserase --update --target ESP32_S3_ALL --serialport COM18`

### Aditional context

```
PS C:\Users\Marcin> nanoff --update --target ESP32_S3_ALL --serialport COM18 -v diag
.NET nanoFramework Firmware Flasher v2.5.86+e504cd875d
Copyright (C) 2019 .NET Foundation and nanoFramework project contributors

Using COM18 @ 1500000 baud to connect to ESP32.
Reading details from chip...
Executing esptool with the following parameters:
'--port COM18  --chip auto   --after no_reset_stub flash_id'

OK

Connected to:
ESP32-S3 (ESP32-S3 (QFN56) (revision v0.2))
Features WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Flash size 8MB GD25Q64 from GIGADEVICE (manufacturer 0x200 device 0x16407)
PSRAM: undetermined
Crystal 40MHz
MAC 80:65:99:A7:04:00

Trying to find ESP32_S3_ALL in stable repository...OK
Extracting ESP32_S3_ALL-1.9.1.224.zip...OK

Updating to 1.9.1.224

Flashing firmware...
Executing esptool with the following parameters:
'--port COM18  --chip esp32s3   --after no_reset_stub read_flash 0x490000 0x200000 "askvm13k.bda"'
```

```
PS C:\Users\Marcin> nanoff --masserase --update --target ESP32_S3_ALL --serialport COM18
.NET nanoFramework Firmware Flasher v2.5.86+e504cd875d
Copyright (C) 2019 .NET Foundation and nanoFramework project contributors

Reading details from chip...OK

Connected to:
ESP32-S3 (ESP32-S3 (QFN56) (revision v0.2))
Features WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Flash size 8MB GD25Q64 from GIGADEVICE (manufacturer 0x200 device 0x16407)
PSRAM: undetermined
Crystal 40MHz
MAC 80:65:99:A7:04:00

Extracting ESP32_S3_ALL-1.9.1.224.zip...OK

Updating to 1.9.1.224

Erasing flash...OK
Flashing firmware...OK
```

CoryCharlton, 2024-06-19:
I've been seeing the same behavior. Waiting does result in success but it takes a very long time (will time it later).

josesimoes, 2024-06-19:
This is happenign because of the flash size.
If you don't `masserase` it will backup the config flash contents to reflash it after the general update (so you don't lose configs, wifi credentials, certs, etc)

Unless there is a better/quick way to perform these operations (which I'm not aware of) there is nothing we can do...
The alternative is to always ditch the config partition content on updates... 😞