nanoframework/Home #1511: NetworkHelper.SetupAnConnect() or WifiNetworkHelper.ConnectDhcp() do not properly reset if they fail with a Cancelation Token Timeout

muenchris, 2024-06-20:
### Library/API/IoT binding

nanoFramework.Networking

### Visual Studio version

VS2022.10.9.7

### .NET nanoFramework extension version

2022.7.8

### Target name(s)

ESP32_OLIMEX_WROVER

### Firmware version

.247

### Device capabilities

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

### Description

If the NetworkHelper cannot get an IP or fails otherwise (i.e. disconnected media), it fails to reset properly to "try again".

### How to reproduce

Upload a sketch to the ESP

Run the code below and dont plug in an ethernet cable.
Once it has failed, plug in the cable - it should try again but the cancellation token is not reset properly and the "SetupAndConnectNetwork" function fails right away.

### Expected behaviour

The tokes is "newed" again and should allow a second try. Since it fails right away, there might be something not working/resetting correctly in the SetupAndConnectNetwork helper.

### Sample project or code

```
        private static void InitNetworkAndComms()
        {
            bool success = false;
            int trialcounter = 0;
            while (!success)
            {

                    using (CancellationTokenSource cs = new(60000))
                    {
                        DebugWriteLine($"Trying Ethernet...");
                        success = NetworkHelper.SetupAndConnectNetwork(cs.Token);
                    }
                    if (success)
                        Debug.WriteLine("Ethernet Connected");
                    else
                        Debug.WriteLine($"Eth failed:{trialcounter}");
                
                if (!success)
                {
                    using (CancellationTokenSource cs2 = new(60000))
                    {
                        DebugWriteLine($"Trying Wifi...{WifiSSID}");
                        success = WifiNetworkHelper.ConnectDhcp(WifiSSID, WifiPWD, requiresDateTime: false, token: cs2.Token);
                        if (success)
                            Debug.WriteLine("Wifi Connected");
                        else
                            Debug.WriteLine($"wifi failed:{trialcounter}");
                    }
                }

                if (!success)
                {
                    Debug.WriteLine($"Can't get a proper IP address, error: {NetworkHelper.Status}. Trying again in 5 seconds");
                    if (NetworkHelper.HelperException != null)
                    {
                        Debug.WriteLine($"Exception: {NetworkHelper.HelperException}");
                    }
                    trialcounter++;
                    Thread.Sleep(5000);
                }
            }
        }
```

### Aditional information

The Wifi "ConnectDhcp" is never successful in this context even if the SSDI and PWD are correct

josesimoes, 2024-06-27:
Noted. That's a good suggestion for an improvement of the helper class!

josesimoes, 2024-06-27:
Happy to discuss the required changes and for a community member to submit a PR improving this. 😉

networkfusion, 2025-03-07:
Possibily related: https://github.com/nanoframework/nf-interpreter/pull/3033