nanoframework/Home #1514: VS extension no device detected

jxnkwlp, 2024-06-23:
### Tool

Visual Studio extension

### Description

VS extension and nanoff cli can't be detect the device. i try solution from other issues, but not working . 

The chip is esp32-s3 wroom n8r2.

### How to reproduce

1. flash command and result is ok.
```
nanoff --update --platform esp32 --target ESP32_S3 --serialport COM5 -v diag --baud 115200 --masserase
```
2. Check output when start.
```
---- Opened the serial port COM5 ----
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1380
load:0x403c9700,len:0x4
load:0x403c9704,len:0xba4
load:0x403cc700,len:0x2c5c
SHA-256 comparison failed:
Calculated: 355e2a6e0d4849dc88faff8966c6e20bcbf425d872672e90795b46c85cffdfdb
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98f0
I (45) boot: .NET nanoFramework 2nd stage bootloader ESP-IDF v5.1.3
I (45) boot: build Jun 16 2024 05:11:39
I (45) boot: chip revision: v0.2
I (49) boot.esp32s3: Boot SPI Speed : 80MHz
I (54) boot.esp32s3: SPI Mode       : DIO
I (59) boot.esp32s3: SPI Flash Size : 8MB
I (63) boot: Enabling RNG early entropy source...
I (69) boot: Partition Table:
I (72) boot: ## Label            Usage          Type ST Offset   Length
I (80) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (87) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (95) boot:  2 factory          factory app      00 00 00010000 001a0000
I (102) boot:  3 deploy           Unknown data     01 84 001b0000 002e0000
I (110) boot:  4 config           Unknown data     01 82 00490000 00200000
I (117) boot: End of partition table
I (121) esp_image: segment 0: paddr=00010020 vaddr=3c0d0020 size=24f78h (151416) map
I (147) esp_image: segment 1: paddr=00034fa0 vaddr=3fc99e00 size=03ea8h ( 16040) load
I (149) esp_image: segment 2: paddr=00038e50 vaddr=40374000 size=071c8h ( 29128) load
I (157) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=ce638h (845368) map
I (256) esp_image: segment 4: paddr=0010e660 vaddr=4037b1c8 size=0ebech ( 60396) load
I (265) esp_image: segment 5: paddr=0011d254 vaddr=600fe000 size=00064h (   100) load
I (275) boot: Loaded app from partition at offset 0x10000
I (275) boot: Disabling RNG early entropy source...

```
3. Check vs extensions & nano cli, can't be detect the device.

4. try deploy test project image 
``` cs
public static void Main()
{
    while (true)
    {
        Debug.WriteLine("Hello from nanoFramework!");

        Thread.Sleep(1500);
    }
}
```
deploy image
```
nanoff --deploy --platform esp32 --target ESP32_S3 --serialport COM5  -v diag --image "xxxxxxx\NFApp1\bin\Debug\NFApp1.bin"
```

check output

```
---- Opened the serial port COM5 ----
Hello from nanoFramework!
Hello from nanoFramework!
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1380
load:0x403c9700,len:0x4
load:0x403c9704,len:0xba4
load:0x403cc700,len:0x2c5c
SHA-256 comparison failed:
Calculated: 355e2a6e0d4849dc88faff8966c6e20bcbf425d872672e90795b46c85cffdfdb
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98f0
I (45) boot: .NETðESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1380
load:0x403c9700,len:0x4
load:0x403c9704,len:0xba4
load:0x403cc700,len:0x2c5c
SHA-256 comparison failed:
Calculated: 355e2a6e0d4849dc88faff8966c6e20bcbf425d872672e90795b46c85cffdfdb
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98f0
I (45) boot: .NET nanoFramework 2nd stage bootloader ESP-IDF v5.1.3
I (45) boot: build Jun 16 2024 05:11:39
I (45) boot: chip revision: v0.2
I (49) boot.esp32s3: Boot SPI Speed : 80MHz
I (54) boot.esp32s3: SPI Mode       : DIO
I (59) boot.esp32s3: SPI Flash Size : 8MB
I (63) boot: Enabling RNG early entropy source...
I (69) boot: Partition Table:
I (72) boot: ## Label            Usage          Type ST Offset   Length
I (80) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (87) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (95) boot:  2 factory          factory app      00 00 00010000 001a0000
I (102) boot:  3 deploy           Unknown data     01 84 001b0000 002e0000
I (110) boot:  4 config           Unknown data     01 82 00490000 00200000
I (117) boot: End of partition table
I (121) esp_image: segment 0: paddr=00010020 vaddr=3c0d0020 size=24f78h (151416) map
I (147) esp_image: segment 1: paddr=00034fa0 vaddr=3fc99e00 size=03ea8h ( 16040) load
I (149) esp_image: segment 2: paddr=00038e50 vaddr=40374000 size=071c8h ( 29128) load
I (157) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=ce638h (845368) map
I (256) esp_image: segment 4: paddr=0010e660 vaddr=4037b1c8 size=0ebech ( 60396) load
I (265) esp_image: segment 5: paddr=0011d254 vaddr=600fe000 size=00064h (   100) load
I (275) boot: Loaded app from partition at offset 0x10000
I (275) boot: Disabling RNG early entropy source...
Hello from nanoFramework!
Hello from nanoFramework!
Hello from nanoFramework!
Hello from nanoFramework!
Hello from nanoFramework!
Hello from nanoFramework!
Hello from nanoFramework!

```

5. Check vs extensions & nano cli again, can't detect the device.

alberk8, 2024-06-24:
@jxnkwlp , the nanoff command `nanoff --update --platform esp32 --target ESP32_S3 `  You either use `--platform esp32` for the nanoff to detect the chip/mcp or  -`-target ESP32_S3` for specific chip/mcu,  **don't use both**.   

One more thing, is your ESP32 S3 have one or two USB C?   If there is two then you need to plug in the one on the **left** for VS to detect.

jxnkwlp, 2024-06-24:
@alberk8   
1. I will try the command again.  
2. The board have one USB C. and connect by CH340K (with auto program circuit)

jxnkwlp, 2024-06-24:
> @jxnkwlp , the nanoff command `nanoff --update --platform esp32 --target ESP32_S3 ` You either use `--platform esp32` for the nanoff to detect the chip/mcp or -`-target ESP32_S3` for specific [...]

Use command flash again, not working. nanoff cli and vs extension can't detect the device.

btw, i add tx/rx LED on board, and can see the led flashed a few times when detecting

josesimoes, 2024-06-27:
This is a support question. Moving to Discussions.