nanoframework/Home #1518: Error E4000: Error executing esptool command. (Object reference not set to an instance of an object.) on deploy

TAMHAN, 2024-07-06:
### Tool

nanoff

### Description

Hello,
working with the ESP32-S3 DevKit C and try to deploy the firmware. Interestingly, if I run with --update, the program runs without errors, while the error shown in the title occurs whenever I run deploy. In both cases, I had the board in bootloader mode. Sadly, the .net nanoframework runtime did not work in any of the cases.

Here are some command line logs:

PS C:\Users\tamha> nanoff --update --target ESP32_S3 --serialport COM18
.NET nanoFramework Firmware Flasher v2.5.90+c66e018c48
Copyright (C) 2019 .NET Foundation and nanoFramework project contributors

Reading details from chip...OK

Connected to:
ESP32-S3 (ESP32-S3 (QFN56) (revision v0.1))
Features WiFi, BLE
Flash size 16MB GD25Q128 from GIGADEVICE (manufacturer 0x200 device 0x16408)
PSRAM: undetermined
Crystal 40MHz
MAC F4:12:FA:5B:44:F0

Extracting ESP32_S3-1.10.0.10.zip...OK

Updating to 1.10.0.10

PS C:\Users\tamha> nanoff --deploy --target ESP32_S3 --serialport COM18
.NET nanoFramework Firmware Flasher v2.5.90+c66e018c48
Copyright (C) 2019 .NET Foundation and nanoFramework project contributors

Reading details from chip...OK

Connected to:
ESP32-S3 (ESP32-S3 (QFN56) (revision v0.1))
Features WiFi, BLE
Flash size 16MB GD25Q128 from GIGADEVICE (manufacturer 0x200 device 0x16408)
PSRAM: undetermined
Crystal 40MHz
MAC F4:12:FA:5B:44:F0

Flashing deployment partition...Error E4000: Error executing esptool command. (Object reference not set to an instance of an object.)
PS C:\Users\tamha> nanoff --update --target ESP32_S3 --serialport COM18
.NET nanoFramework Firmware Flasher v2.5.90+c66e018c48
Copyright (C) 2019 .NET Foundation and nanoFramework project contributors

Reading details from chip...OK

Connected to:
ESP32-S3 (ESP32-S3 (QFN56) (revision v0.1))
Features WiFi, BLE
Flash size 16MB GD25Q128 from GIGADEVICE (manufacturer 0x200 device 0x16408)
PSRAM: undetermined
Crystal 40MHz
MAC F4:12:FA:5B:44:F0

Extracting ESP32_S3-1.10.0.10.zip...OK

Updating to 1.10.0.10

PS C:\Users\tamha> nanoff --update --target ESP32_S3 --serialport COM18 -v diag
.NET nanoFramework Firmware Flasher v2.5.90+c66e018c48
Copyright (C) 2019 .NET Foundation and nanoFramework project contributors

Using COM18 @ 1500000 baud to connect to ESP32.
Reading details from chip...
Executing esptool with the following parameters:
'--port COM18  --chip auto   --after no_reset_stub flash_id'

OK

Connected to:
ESP32-S3 (ESP32-S3 (QFN56) (revision v0.1))
Features WiFi, BLE
Flash size 16MB GD25Q128 from GIGADEVICE (manufacturer 0x200 device 0x16408)
PSRAM: undetermined
Crystal 40MHz
MAC F4:12:FA:5B:44:F0

Trying to find ESP32_S3 in stable repository...OK
Extracting ESP32_S3-1.10.0.10.zip...OK

Updating to 1.10.0.10

Backup configuration...
Executing esptool with the following parameters:
'--port COM18  --chip esp32s3   --after no_reset_stub read_flash 0x490000 0x300000 "iyyxp2jz.elz"'

Read 3145728 bytes at 0x00490000 in 282.8 seconds (89.0 kbit/s)...
Backup configuration...OK

Executing esptool with the following parameters:
'--port COM18 --baud 1500000 --chip esp32s3   --after hard_reset write_flash --flash_size 16MB 0x0 "C:\Users\tamha\.nanoFramework\fw_cache\ESP32_S3\bootloader.bin" 0x10000 "C:\Users\tamha\.nanoFramework\fw_cache\ESP32_S3\nanoCLR.bin" 0x8000 "C:\Users\tamha\.nanoFramework\fw_cache\ESP32_S3\partitions_16mb.bin" 0x490000 "iyyxp2jz.elz"'

Flashing firmware...OK
PS C:\Users\tamha> nanoff --listdevices
.NET nanoFramework Firmware Flasher v2.5.90+c66e018c48
Copyright (C) 2019 .NET Foundation and nanoFramework project contributors

No devices found
PS C:\Users\tamha> nanoff --listdevices -v diag
.NET nanoFramework Firmware Flasher v2.5.90+c66e018c48
Copyright (C) 2019 .NET Foundation and nanoFramework project contributors

No devices found
PS C:\Users\tamha> nanoff --deploy --target ESP32_S3 --serialport COM18 -v diag
.NET nanoFramework Firmware Flasher v2.5.90+c66e018c48
Copyright (C) 2019 .NET Foundation and nanoFramework project contributors

Using COM18 @ 1500000 baud to connect to ESP32.
Reading details from chip...
Executing esptool with the following parameters:
'--port COM18  --chip auto   --after no_reset_stub flash_id'

OK

Connected to:
ESP32-S3 (ESP32-S3 (QFN56) (revision v0.1))
Features WiFi, BLE
Flash size 16MB GD25Q128 from GIGADEVICE (manufacturer 0x200 device 0x16408)
PSRAM: undetermined
Crystal 40MHz
MAC F4:12:FA:5B:44:F0

Flashing deployment partition...Error E4000: Error executing esptool command. (Object reference not set to an instance of an object.)

alberk8, 2024-07-07:
You need to specify the fullpath of the app you are going to deploy.

`nanoff --target ESP32_S3 --serialport COM18 --deploy --image "E:\GitHub\nf-Samples\samples\Blinky\Blinky\bin\Debug\Blinky.bin"`

TAMHAN, 2024-07-09:
Hello,

thank you so much.

In fact, I have since been able to solve the problem. You need to switch
the USB port on the ESP32 S3 board after the firmware has deployed.

Sorry for causing grief.

Tam

With best regards
Tam HANNA

Enjoy electronics? Join 21k4 other followers by visiting the Crazy Electronics Lab athttps://www.instagram.com/tam.hanna/

closed 2024-07-09 by networkfusion