nanoframework/Home #1519: Unable to execute any command from the nanoFramework extension for VSCode on macOS

loicmorvan, 2024-07-08:
### Tool

Visual Studio Code extension

### Description

When trying to execute any command from the nanoFramework extension in VSCode, VSCode reports an command not found error. For example: when trying to execution "nanoFramework: Add a Project to existing Solution", it says `command 'vscode-nanoframework.nfadd' not found`.

### How to reproduce

1. Open a folder in VSCode.
2. Create a new solution in this folder with `dotnet new solution`.
3. Execute the command "nanoFramework: Add a Project to existing Solution".
4. An error popup shows up with the message `command 'vscode-nanoframework.nfadd' not found`.

### Expected behaviour

The command should be executed as expected.

### Screenshots

[image]

### Aditional context

macOS Monterey, version 12.7.5 (21H1222)
VSCode Version : 1.91.0 (Universal)
nanoFramework extension v1.0.165

josesimoes, 2024-07-17:
Fixed with nanoframework/nf-VSCodeExtension#307

closed 2024-07-17 by josesimoes

webo-de, 2024-07-29:
I still have this issue after installing latest version v1.0.176 on my intel mac with macos 14.2.1 (23C71):
**VSC->**
Version: 1.91.1 (Universal)
Commit: f1e16e1e6214d7c44d078b1f0607b2388f29d729
Date: 2024-07-09T22:07:54.982Z
Electron: 29.4.0
ElectronBuildId: 9728852
Chromium: 122.0.6261.156
Node.js: 20.9.0
V8: 12.2.281.27-electron.0
OS: Darwin x64 23.2.0