nanoframework/Home #1529: When reading text file with StreamReader, FileStream position resets to 0 periodically creating endless loop in the reader.

jazzman55, 2024-08-07:
### Library/API/IoT binding

System.IO.FileSystem, System.IO.Streams

### Visual Studio version

VS2022

### Target name(s)

ESP32_REV3

### Firmware version

1.12.0.32

### Description

Reading text file with StreamReader, the underlying FileStream resets its position to 0 every time StreamReader's buffer is filled up. This causes endless jumps to beginning of the file and thus the file can't be read to end.

### How to reproduce

1. Upload text file to ESP32 config partition. The file must be longer than StreamReader's buffer size (512B).
2. Open the file and read it line-by-line with StreamReader.ReadLine().

### Expected behaviour

File is correctly read line-by-line from start to end. When end of file is reached ReadLine() returns null.

### Screenshots

[image]

### Sample project or code

```
//The last line of loremipsum.txt is <END>
        [TestMethod]
        public void ReadText_StreamReader()
        {
            Debug.WriteLine(nameof(ReadText_StreamReader));
            string line;
            string text = "";
            using (var fs = new StreamReader(new FileStream(@"I:\loremipsum.txt", FileMode.Open, FileAccess.Read)))
            {
                line = fs.ReadLine();
                int i = 0;
                while (line != null && i++ < 50) //exit after reading 50 lines to avoid endless loop
                {
                    Debug.WriteLine(line);
            
                    line = fs.ReadLine();
                    if (line != null)
                        text = line;
                }
            }

            Assert.Equal("<END>", text);
        }
[loremipsum.txt](https://github.com/user-attachments/files/16523448/loremipsum.txt)

```

Flash to ESP32 with:
```
@echo off
set port=COM5
littlefs-python create ..\FileSystem %tmp%\config_image.bin --fs-size=262144 --block-size=4096
python %idf_path%\components\partition_table\parttool.py -p %port% write_partition -n config --input %tmp%\config_image.bin 
del %tmp%\config_image.bin 
```

### Aditional information

Workaround: 
Track stream's position in a separate variable and set the stream position before each read:
[https://github.com/CShark-Hub/Mako-IoT.Device.Services.FileStorage/blob/main/src/MakoIoT.Device.Services.FileStorage/MakoStreamReader.cs#L525](https://github.com/CShark-Hub/Mako-IoT.Device.Services.FileStorage/blob/main/src/MakoIoT.Device.Services.FileStorage/MakoStreamReader.cs#L525)

Note:
This was working OK with one of the previous versions of System.IO.FileSystem

jazzman55, 2024-08-07:
The test file
[loremipsum.txt](https://github.com/user-attachments/files/16523707/loremipsum.txt)

josesimoes, 2024-08-07:
The unit tests must not be covering this either because the content isn't large enough or the code there is missing this operation... 🤔 

Anyway, not surprising that it had a different behavior before, as the Stream has been introduced only recently.

alberk8, 2024-08-11:
@jazzman55,  
Can you try to compile the firmware with this option  `CONFIG_LITTLEFS_FLUSH_FILE_EVERY_WRITE=y ` in the esp SDKCONFIG file?

Akshit018, 2024-08-12:
Subject: Request to Work on Good First Issue

Hi jazzman55,

I noticed the issue with StreamReader and FileStream and believe it would be a great first issue for me to work on. I'm eager to contribute and would love the opportunity to tackle this.

May I take it on?

Thank you,
Akshit

josesimoes, 2024-08-12:
Hey @Akshit018!
That's great that you're willing to tackle this one. 🥳 
I'm assigning it to you now.

Please take a look at the contribution guide [here](https://docs.nanoframework.net/content/contributing/coding-contributions.html). This will typically require a PR to the Sys.IO.FileSystem repo. 
We 💜 PRs so feel free to start one as early as you want so you can have early feedback. 

If you have questions or want to discuss anything, suggest you head over to our Discord server and engage a conversation there.

josesimoes, 2024-10-28:
@Akshit018 curious if you had the time to start working on this one? 😉

jazzman55, 2024-12-17:
It seems that Peek() method is causing the issue. When I removed this call from ReadLine() method it works correctly:
original code:
```
                if (readLineBuff[curPos] == '\r')
                {
                    // If the next character is \n eat it
                    if (Peek() == '\n')
                    {
                        _ = Read();
                    }

                    return new string(readLineBuff, 0, curPos);
                }
```
replacement:
```
              if (readLineBuff[curPos] == '\r')
                {
                    continue;
                }
```

jazzman55, 2024-12-18:
OK, it looks like I narrowed it down to FileStream. It resets its Position after Length property is read. This occurs only when running the code on ESP32 hardware (_\<IsRealHardware>True\</IsRealHardware>_ in runsettings):

```
    [TestClass]
    public class FileStreamTest
    {
        [TestMethod]
        public void Read_should_keep_Position()
        {
            File.WriteAllBytes("I:\\loremipsum.txt", Encoding.UTF8.GetBytes(Lorem));

            using (var fs = new FileStream("I:\\loremipsum.txt", FileMode.Open))
            {
                var buffer = new byte[10];
                //read 10 bytes
                fs.Read(buffer, 0, 10);
                //get position
                Assert.AreEqual(10, fs.Position, "before getting stream length");
                //get length
                var length = fs.Length;
                //get position again
                Assert.AreEqual(10, fs.Position, "after getting stream length");
            }
        }

        private const string Lorem = @"Lorem ipsum dolor sit amet, consectetur adipiscing elit. Quisque lacinia pellentesque pharetra. Nulla a tortor in nunc facilisis cursus nec id felis.";
    }
```

Test result: 
Assert.AreEqual failed. Expected:<10>. Actual:<0>. after getting stream length

josesimoes, 2024-12-20:
@jazzman55 nice finding. 👍🏻  That looks like it.
The current workflow is the best option to get the file length. But, yes, it should be rewind to the last position not the begining of the file. Fix on the way!

closed 2024-12-20 by josesimoes