nanoframework/Home #1535: nanoCLR task not start correctly on ESP32-S3!

omidg2, 2024-08-24:
### Target name(s)

ESP32-S3

### Description

 I use ESP32-S3, I flash nanoframework image to it by this command line:
nanoff --masserase --update --target ESP32_S3_BLE --serialport COM5
the nanoframework image was program on ESP32-S3 correctly as you can see on the first attached screen shot...

but when I connect my device to putty with baudrate 115200 and start it, it seems nanoCLR not start correctly, as you can see
in second screen shot

I use nanoFramework on other version of ESP32 like ESP32-WROOM-32 and it works correctly

### Expected behaviour

nanoCLR task start correctly and we can see the device in Visual Studio Device Explorer...

### Screenshots

[image]
[image]

alberk8, 2024-08-24:
Which  ESP32 S3 do you have? From which manufacturer, make and model?

omidg2, 2024-08-24:
> Which ESP32 S3 do you have? From which manufacturer, make and model?

This is full part number:  ESP32-S3-WROOM-1U-N4
as I mentioned, nanoFramework Image pragrammed correctlly, (ESP32_S3_BLE 1.12.0.47)
but device not found in Visual Studio 22 Device Explorer, I check the problem and it seems nanoCLR task not start....

alberk8, 2024-08-25:
Looks good and based on the screen capture it looks like the boot process is good.  Are you sure that there are no other serial terminal app is opened when VS 2022 you plug in the ESP32 S3 and view it in Device Explorer?  Make sure that the Device Watcher is **enabled** in Device Explorer.

omidg2, 2024-08-25:
No, not good, if you see second screen capture, you can see that nanoCLR not started correctly, it stops in line
**_I (304) boot: Disabling RNG early entropy source ..._**
a good start up should be something like this screen capture:
[image]
this screen capture is from another board which works correctly, 
as you see here, many lines such as 
**_I (415) cpu_start: Multicore app_**

and lines after that should be exist,

alberk8, 2024-08-25:
If your S3 has 2 usb-c port then VS will only detect it on the other USB port (usually the left hand side usb c).

omidg2, 2024-08-25:
Dear Alber, I do not use USB, I use USART 0 on 36 and 37  of module, as you can see in the picture below:
[image]
my communications to module, fully are done by this USART port ...

alberk8, 2024-08-26:
The current S3 firmware expect to communicated with VS 2022 via USB, you could [compile ](https://github.com/nanoframework/nf-interpreter/blob/b777c0ee29475c6a53cbedb07604e5ed116370b5/targets/ESP32/CMakePresets.json#L358)with CDC off.

omidg2, 2024-08-26:
It means if I connect my PC to via USB port to Pin 19 and Pin 20 of module, like below diagram, it might work properly?
[image]

alberk8, 2024-08-27:
most likely will work

omidg2, 2024-08-27:
Thank you alberk8, I tested it and it didn't work, I'm probably doing something wrong somewhere,